openwrt/openwrt #17633: using status -> real time graphs -> connections, selecting use dns does not work

peterb99, 2025-01-16:
Lists of connections showing ip addresses of source and destinations, when selecting 'enable dns lookup' does not change IP addresses to related dns names

RHBH, 2025-01-24:
Is rpcd-mod-rrdns installed?

Cub0n, 2025-02-14:
@RHBH Yes it is.
I guess, it is a crucial combination of dnsmasq-full, local resolvers, adBlock, dnssec, ...

Openwrt asks the local resolver (normally 127.0.0.1:53) for reverse lookup the IP Address. This query can or will not be answered properly, so that OpenWrt fires a lot of UDP requests and no reverse lookup happens.
The same situation at Openwrt -> Network -> Diagnostics -> Nslookup: Enter the OpenWrt  server address (64.226.122.113) and the result is:
```
Server:		127.0.0.1
Address:	127.0.0.1:53

Non-authoritative answer:
```

I have not figured out why this situation happens.
The reverse lookup works properly from hosts behind Openwrt.