openwrt/openwrt #17635: Extreme Networks AP3935 (qcom-ipq8068-ap3935) lan1 port not working

plaes, 2025-01-16:
### Describe the bug

LAN1 port on Extreme Networks AP3935i-FCC board is not working (detecting carrier):
```
[   20.213617] ipq806x-gmac-dwmac 37000000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
[   20.296872] ipq806x-gmac-dwmac 37000000.ethernet eth0: PHY [37000000.mdio-mii:01] driver [Qualcomm Atheros AR8035] (irq=POLL)
[   20.501160] ipq806x-gmac-dwmac 37000000.ethernet: Failed to reset the dma
[   20.501196] ipq806x-gmac-dwmac 37000000.ethernet eth0: stmmac_hw_setup: DMA engine initialization failed
[   20.507023] ipq806x-gmac-dwmac 37000000.ethernet eth0: __stmmac_open: Hw setup failed
[   43.885997] ipq806x-gmac-dwmac 37000000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
[   43.987080] ipq806x-gmac-dwmac 37000000.ethernet eth0: PHY [37000000.mdio-mii:01] driver [Qualcomm Atheros AR8035] (irq=POLL)
[   44.189637] ipq806x-gmac-dwmac 37000000.ethernet: Failed to reset the dma
[   44.189682] ipq806x-gmac-dwmac 37000000.ethernet eth0: stmmac_hw_setup: DMA engine initialization failed
[   44.195527] ipq806x-gmac-dwmac 37000000.ethernet eth0: __stmmac_open: Hw setup failed
[   44.217904] ipq806x-gmac-dwmac 37000000.ethernet eth0: entered allmulticast mode
[   44.223332] ipq806x-gmac-dwmac 37000000.ethernet eth0: entered promiscuous mode
```

### OpenWrt version

r28304-6dacba30a7

### OpenWrt release

24.10.0-rc5

### OpenWrt target/subtarget

ipq806x/generic

### Device

Extreme Networks AP3935

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2025-01-17:
@Ansuel Got any ideas?

plaes, 2025-01-17:
Seems like the DMA engine isn't culprit, as network seemingly worked to do the bootstrap into Openwrt:
1. Fetch `extreme_ap3935-initramfs-uImage` via tftpboot (from internal network) and boot it from memory
2. Fetch the `extreme_ap3935-squashfs-nand-sysupgrade.bin` from tftpbooted system and do the sysupgrade

But once OpenWRT had been flashed booted from nand, I was unable to get LAN1 to recognize that something is connected to ethernet port.

Ansuel, 2025-01-17:
> But once OpenWRT had been flashed and boots from nand, I am unable to get

Just to make sure U-Boot tftp doesn't have the problem but a flashed
firmware triggers it?

I think it's he bootloader doing magic when booting from flash. It's not
the first time actually it's pretty common on ipq806x. A test would be try
manually loading the firmware from nand without using the default command.

plaes, 2025-01-19:
Ok, seems like when I boot the device without actively connected cable, the PHY is not correctly powered up:
```
# mdio 37000000.mdio-mii 0x1
BMCR(0x00): 0x1800
  flags: -reset -loopback +aneg-enable +power-down -isolate -aneg-restart
         -collision-test
  speed: 10-half
```

Now, when I attempt to bring the interface up manually (still with no cable connected), I got following:
```
# ip link set eth0 up
[  214.090760] ipq806x-gmac-dwmac 37000000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
[  214.177230] ipq806x-gmac-dwmac 37000000.ethernet eth0: PHY [37000000.mdio-mii:01] driver [Qualcomm Atheros AR8035] (irq=POLL)
[  214.379475] ipq806x-gmac-dwmac 37000000.ethernet: Failed to reset the dma
[  214.379511] ipq806x-gmac-dwmac 37000000.ethernet eth0: stmmac_hw_setup: DMA engine initialization failed
[  214.385352] ipq806x-gmac-dwmac 37000000.ethernet eth0: __stmmac_open: Hw setup failed
ip: SIOCSIFFLAGS: Operation timed out
```

Then, after connecting cable I still got 4 failures (same log as previously), until on 5th attempt it worked:
```
ip link set eth0 up
[  363.910397] ipq806x-gmac-dwmac 37000000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
[  363.997238] ipq806x-gmac-dwmac 37000000.ethernet eth0: PHY [37000000.mdio-mii:01] driver [Qualcomm Atheros AR8035] (irq=POLL)
[  364.189454] dwmac1000: Master AXI performs any burst length
[  364.189487] ipq806x-gmac-dwmac 37000000.ethernet eth0: No Safety Features support found
[  364.203967] ipq806x-gmac-dwmac 37000000.ethernet eth0: IEEE 1588-2008 Advanced Timestamp supported
[  364.204152] ipq806x-gmac-dwmac 37000000.ethernet eth0: registered PTP clock
[  364.212616] ipq806x-gmac-dwmac 37000000.ethernet eth0: configuring for phy/rgmii-id link mode
[  370.499837] ipq806x-gmac-dwmac 37000000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[  370.499933] br-lan: port 1(eth0) entered blocking state
[  370.507778] br-lan: port 1(eth0) entered forwarding state

# mdio 37000000.mdio-mii 0x1
BMCR(0x00): 0x1000
  flags: -reset -loopback +aneg-enable -power-down -isolate -aneg-restart
         -collision-test
  speed: 10-half

BMSR(0x01): 0x796d
  capabilities: -100-t4 +100-tx-f +100-tx-h +10-t-f +10-t-h -100-t2-f -100-t2-h
  flags:        +ext-status +aneg-complete -remote-fault +aneg-capable +link
                -jabber +ext-register

ID(0x02/0x03): 0x004dd072

ESTATUS(0x0F): 0xa000
  capabilities: +1000-x-f -1000-x-h +1000-t-f -1000-t-h
```

Now, after these operations whenever I disconnect/reconnect cable link status gets properly detected again. And the culprit seems that we initially start with `+power-down` flag enabled in the BMCR register:
* `BMCR(0x00): 0x1800` -> not working
* `BMCR(0x00): 0x1000` -> working

Also, LAN2 minor issues:
1. It takes ~40 seconds to detect link up after I connect cable (disconnect is detected immediately)
2. Link is detected as 100MBit (laptop on the other side has gigabit port)
```
$ mdio 37000000.mdio-mii 2
BMCR(0x00): 0x1000
  flags: -reset -loopback +aneg-enable -power-down -isolate -aneg-restart
         -collision-test
  speed: 10-half

BMSR(0x01): 0x796d
  capabilities: -100-t4 +100-tx-f +100-tx-h +10-t-f +10-t-h -100-t2-f -100-t2-h
  flags:        +ext-status +aneg-complete -remote-fault +aneg-capable +link
                -jabber +ext-register

ID(0x02/0x03): 0x004dd074

ESTATUS(0x0F): 0xa000
  capabilities: +1000-x-f -1000-x-h +1000-t-f -1000-t-h
```

stangri, 2025-01-21:
Not sure if it's helpful, but here LAN1 is working fine on AP3935/OpenWrt 24.10.0-rc5, **when there's PoE on LAN1**. Without PoE supplied, I have the same issue with LAN1 being flakey.

robimarko, 2025-01-21:
I guess that there is a reset GPIO attached to the PHY

plaes, 2025-01-21:
> I guess that there is a reset GPIO attached to the PHY

Reset seems OK, (though I tested it only with multimeter) being 3.2V in both working and non-working cases.

[image]

Though, one thing I can confirm is whenever I ping a working IP address from u-boot, link-detection starts working properly in Linux and DMA reset issue disappears.

plaes, 2025-01-28:
I dug into AR8035 datasheet (March 2011) and found following snippet (verbatim) in the description of the POWER_DOWN bit of the control register:
> IEEE power down shuts down the chip except for the MAC interface if 16.3 is set to 1. If 16.3 is set to 0, then the MAC interface also shuts down. Power-downalos has no effect on the 125clk output if 16.4 is set to 0.

The `16.3` syntax seems to represent register address in decimal and the bit, so it's Function Control register 0x10 which has bits 3 and 4 defined as reserved.

Now, power on default for 0x10 defaults to `0x0862` which means that bits 3 and 4 are set to 0.

dmascord, 2025-03-21:
My workaround is:
```
     set bootcmd 'ping 192.168.1.1; run boot_flash'
     saveenv
```

maurerle, 2025-06-19:
I also have this problem on AP3935.
Only the ports which are used when starting the network come up correctly.

For me, it helps to run `service network restart` with a lan cable connected to LAN1 to get LAN1 to work.
After that, unplugging and replugging is no problem.
Alternatively, rebooting with the plugged cable helps as well to get the port to work.

So if you ignore the second port and use this device as a single port PoE device, you should not be running into this problem.

Sidenote: I did not get LAN1 or LAN2 to work at all on 24.10.1 release. Snapshot has the issues as described, but works fine otherwise.

grische, 2025-07-12:
> For me, it helps to run `service network restart` with a lan cable connected to LAN1 to get LAN1 to work. After that, unplugging and replugging is no problem. Alternatively, rebooting with the plugged cable helps as well to get the port to work.

I can confirm that this workaround works like a charm.

grische, 2026-04-25:
This change fixed the problem for me:

```diff
diff --git a/target/linux/ipq806x/files-6.6/arch/arm/boot/dts/qcom/qcom-ipq8068-ap3935.dts b/target/linux/ipq806x/files-6.6/arch/arm/boot/dts/qcom/qcom-ipq806
index 9219e34a92..80cd500c7f 100644
--- a/target/linux/ipq806x/files-6.6/arch/arm/boot/dts/qcom/qcom-ipq8068-ap3935.dts
+++ b/target/linux/ipq806x/files-6.6/arch/arm/boot/dts/qcom/qcom-ipq8068-ap3935.dts
@@ -318,6 +318,7 @@

        phy1: ethernet-phy@1 {
                reg = <1>;
+               qca,disable-hibernation-mode;
        };

        phy2: ethernet-phy@2 {
```

I'll try to create a PR, but first I want to understand what consequences it has to disable hibernation.