openwrt/openwrt #17641: Log: mtk_soc_eth 1e100000.ethernet eth0: transmit queue timed out

getreu, 2025-01-17:
### Describe the bug

# Problem description

I observed the following with

* OpenWrt: 23.5.5, 24.10rc4 and 24.10rc5
* Netgear WNDR3700v2, Netgear WNDR3700v4, Miaomi Mi 3G

The router connects to the provider, but after 7 minutes loses the connection.

A typical cycle looks like this:

```
Thu Jan 16 11:52:36 2025 daemon.notice netifd: Interface 'wan' is setting up now
Thu Jan 16 11:52:36 2025 daemon.info pppd[11134]: Plugin pppoe.so loaded.
Thu Jan 16 11:52:36 2025 daemon.info pppd[11134]: PPPoE plugin from pppd 2.5.1
Thu Jan 16 11:52:36 2025 daemon.notice pppd[11134]: pppd 2.5.1 started by root, uid 0
Thu Jan 16 11:52:37 2025 kern.crit kernel: [ 9701.965836] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 3: transmit queue 4 timed out 9540250 ms
Thu Jan 16 11:52:42 2025 kern.crit kernel: [ 9707.005788] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 3: transmit queue 4 timed out 9545290 ms
Thu Jan 16 11:52:42 2025 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) 00:21:b7:bf:68:c7
Thu Jan 16 11:52:42 2025 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.1.172 00:21:b7:bf:68:c7
Thu Jan 16 11:52:42 2025 daemon.warn dnsmasq-dhcp[1]: Error sending DHCP packet to 192.168.1.172: Resource temporarily unavailable
Thu Jan 16 11:52:47 2025 kern.crit kernel: [ 9711.965783] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 3: transmit queue 4 timed out 9550250 ms
Thu Jan 16 11:52:52 2025 kern.crit kernel: [ 9717.005804] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 3: transmit queue 4 timed out 9555290 ms
Thu Jan 16 11:52:57 2025 kern.crit kernel: [ 9721.965777] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 3: transmit queue 4 timed out 9560250 ms
Thu Jan 16 11:53:02 2025 kern.crit kernel: [ 9727.005790] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 3: transmit queue 4 timed out 9565290 ms
Thu Jan 16 11:53:07 2025 kern.crit kernel: [ 9731.965773] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 3: transmit queue 4 timed out 9570250 ms
Thu Jan 16 11:53:11 2025 daemon.warn pppd[11134]: Timeout waiting for PADO packets
Thu Jan 16 11:53:11 2025 daemon.err pppd[11134]: Unable to complete PPPoE Discovery phase 1
Thu Jan 16 11:53:11 2025 daemon.info pppd[11134]: Exit.
Thu Jan 16 11:53:11 2025 daemon.notice netifd: Interface 'wan' is now down
```

# Workaround

Execute in a router shell:

```sh
ethtool -K eth0 tso off gso off gro off
ethtool -K br-lan tso off gso off gro off
ethtool -K br-lan64 tso off gso off gro off
```

See also:

* [Redmi AX6000 random crash (NETDEV WATCHDOG: eth0 (mtk_soc_eth): transmit queue 7 timed out) · Issue #13262 · openwrt/openwrt · GitHub](https://github.com/openwrt/openwrt/issues/13262)
* [Lan, br-lan and eth0 across different devices - Installing and Using OpenWrt / Network and Wireless Configuration - OpenWrt Forum](https://forum.openwrt.org/t/lan-br-lan-and-eth0-across-different-devices/167505)

# Make the workaround reboot persistent

[How to make ethtool setting persistent on br-lan - Installing and Using OpenWrt / Network and Wireless Configuration - OpenWrt Forum](https://forum.openwrt.org/t/how-to-make-ethtool-setting-persistent-on-br-lan/6433/3)

    cd /etc/hotplug.d/iface
    vi 10-ethtool
    

in that file add

```sh
[ ifup = "$ACTION" ] || exit 0
[ -n "$DEVICE" ] || exit 0

if [ "$DEVICE" == eth0 ] || [ "$DEVICE" == br-lan ] || [ "$DEVICE" == br-lan64 ] ; then
      ethtool -K "$DEVICE" tso off gso off gro off
fi
```

Remark:
or maybe you don't want/need the check for

if [ "$DEVICE" == eth0 ] || [ "$DEVICE" == br-lan ] || [ "$DEVICE" == br-lan64 ] ; then

# Make the workaround upgrade persistent

Persist it across `sysupgrades`, add file to `/etc/sysupgrade.conf` through the CLI or Luci UI (`System - Backup / Flash Firmware- Configuration`) 

```sh
cat << EOF >> /etc/sysupgrade.conf
/etc/hotplug.d/iface/10-ethtool
EOF
```

### OpenWrt version

DISTRIB_REVISION='r24106-10cc5fcd00'

### OpenWrt release

DISTRIB_RELEASE='23.05.5'

### OpenWrt target/subtarget

DISTRIB_TARGET='ath79/nand'

### Device

Netgear WNDR3700 v4

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-01-17:
wow, if you search you will see dozen affected devices. Can you try harder flipping one parameter a day, they all mean performance due to bigger bunch of packets being DMA-d to/from netcard.

raenye, 2025-01-18:
BTW, I'd go for `case`, e.g.,
```
case "$DEVICE" in
  eth0|br-lan|br-lan64)
    ethtool ...
    ;;
esac
```

getreu, 2025-01-18:
Yes, indeed!

getreu, 2025-03-16:
Addendum: the above workaround did not work at the long run.
Some users suggest to install the `swconfig` and `kmod-swconfig` packages. This did not work either.
The way out I found so far is to take on older router and to downgrade OpenWrt to version OpenWrt 21.02.7.

Since all my routers are affected by the problem, this could be a general problem with the (new) Ethernet stack. This is contradicted by the fact that relatively few users report similar problems. It is possible that my ISP is not particularly stable. I am running out of ideas.

brada4, 2025-03-16:
```
ethtool -k wan | grep -v fixed
```

Disable half of options , run for a day, then other half, and so on bisect to a parameter that malfunctions?

ikcalB, 2025-07-28:
I can report similar problems on a Linksys AX23 v1.
> I highly suspect I hit this issue before with at least 1 other model - but as I bought them used, cheap I did not bother testing further an ordered a recent AP.

**100% reproducible with v24.10.1.**  
Takes between hours and days, ALWAYS happens. High amount of traffic, particularly youtube 4k seems to reliably trigger this issue within <3hours.

Also: #16419 seems to reference this issue, bug was closed without a solution (only a commit, that _possibly_ fixes things)

I've tried disabling all options for all available interfaces (as mentioned above): no difference.
(now trying v24.10.2)

typical cycle for me looks like:
(starting from line 11)
```
Mon Jul 28 10:06:43 2025 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 10.0.1.180 76:8d:b6:c4:74:a7
Mon Jul 28 10:06:43 2025 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 10.0.1.180 76:8d:b6:c4:74:a7 A55
Mon Jul 28 10:07:20 2025 daemon.notice hostapd: phy1-ap0: AP-STA-DISCONNECTED 76:8d:b6:c4:74:a7
Mon Jul 28 10:07:20 2025 daemon.info hostapd: phy0-ap0: STA e2:fd:01:ab:d0:72 IEEE 802.11: authenticated
Mon Jul 28 10:07:20 2025 daemon.info hostapd: phy0-ap0: STA e2:fd:01:ab:d0:72 IEEE 802.11: associated (aid 3)
Mon Jul 28 10:07:20 2025 daemon.notice hostapd: phy0-ap0: AP-STA-CONNECTED e2:fd:01:ab:d0:72 auth_alg=open
Mon Jul 28 10:07:20 2025 daemon.info hostapd: phy0-ap0: STA e2:fd:01:ab:d0:72 WPA: pairwise key handshake completed (RSN)
Mon Jul 28 10:07:20 2025 daemon.notice hostapd: phy0-ap0: EAPOL-4WAY-HS-COMPLETED e2:fd:01:ab:d0:72
Mon Jul 28 10:07:20 2025 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 10.0.1.231 e2:fd:01:ab:d0:72
Mon Jul 28 10:07:20 2025 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 10.0.1.231 e2:fd:01:ab:d0:72 A55
Mon Jul 28 10:14:10 2025 kern.crit kernel: [ 6708.951375] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 3: transmit queue 3 timed out 5250 ms
Mon Jul 28 10:14:15 2025 kern.crit kernel: [ 6713.911343] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 3: transmit queue 3 timed out 10210 ms
Mon Jul 28 10:14:20 2025 kern.crit kernel: [ 6718.951305] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 3: transmit queue 3 timed out 15250 ms
Mon Jul 28 10:14:25 2025 kern.crit kernel: [ 6723.911286] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 3: transmit queue 3 timed out 20210 ms
Mon Jul 28 10:14:30 2025 kern.crit kernel: [ 6728.951255] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 3: transmit queue 3 timed out 25250 ms
Mon Jul 28 10:14:35 2025 kern.crit kernel: [ 6733.911199] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 3: transmit queue 3 timed out 30210 ms
Mon Jul 28 10:14:40 2025 kern.crit kernel: [ 6738.951186] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 3: transmit queue 3 timed out 35250 ms
Mon Jul 28 10:14:45 2025 kern.crit kernel: [ 6743.911128] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 3: transmit queue 3 timed out 40210 ms
Mon Jul 28 10:14:49 2025 daemon.notice hostapd: phy0-ap0: AP-STA-DISCONNECTED e2:fd:01:ab:d0:72
Mon Jul 28 10:14:50 2025 kern.crit kernel: [ 6748.951127] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 3: transmit queue 3 timed out 45250 ms
Mon Jul 28 10:14:55 2025 kern.crit kernel: [ 6753.911065] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 3: transmit queue 3 timed out 50210 ms
Mon Jul 28 10:15:00 2025 kern.crit kernel: [ 6758.951035] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 3: transmit queue 3 timed out 55250 ms
Mon Jul 28 10:15:05 2025 kern.crit kernel: [ 6763.910989] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 3: transmit queue 3 timed out 60210 ms
Mon Jul 28 10:15:10 2025 kern.crit kernel: [ 6768.950953] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 3: transmit queue 3 timed out 65250 ms
Mon Jul 28 10:15:15 2025 kern.crit kernel: [ 6773.910926] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 3: transmit queue 3 timed out 70210 ms
Mon Jul 28 10:15:20 2025 kern.crit kernel: [ 6778.950897] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 3: transmit queue 3 timed out 75250 ms
Mon Jul 28 10:15:25 2025 kern.crit kernel: [ 6783.910857] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 3: transmit queue 3 timed out 80210 ms
Mon Jul 28 10:15:30 2025 kern.crit kernel: [ 6788.950816] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 3: transmit queue 3 timed out 85250 ms
Mon Jul 28 10:15:35 2025 kern.crit kernel: [ 6793.910798] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 3: transmit queue 3 timed out 90210 ms
Mon Jul 28 10:15:40 2025 kern.crit kernel: [ 6798.950742] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 3: transmit queue 3 timed out 95250 ms
Mon Jul 28 10:15:45 2025 kern.crit kernel: [ 6803.910713] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 3: transmit queue 3 timed out 100210 ms
Mon Jul 28 10:15:50 2025 kern.crit kernel: [ 6808.950669] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 3: transmit queue 3 timed out 105250 ms
Mon Jul 28 10:15:55 2025 kern.crit kernel: [ 6813.910641] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 3: transmit queue 3 timed out 110210 ms
Mon Jul 28 10:16:00 2025 kern.crit kernel: [ 6818.950606] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 3: transmit queue 3 timed out 115250 ms
[... cut ...]
Mon Jul 28 10:24:10 2025 kern.crit kernel: [ 7308.947344] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 3: transmit queue 3 timed out 605250 ms
Mon Jul 28 10:24:15 2025 kern.crit kernel: [ 7313.907308] mtk_soc_eth 1e100000.ethernet eth0: NETDEV WATCHDOG: CPU: 3: transmit queue 3 timed out 610210 ms
```

@getreu I can understand your frustration - as an embedded developer I know, that those kind of bugs are possible to happen, with best developer intentions.

I just hope that someone with more knowledge will look into this, b/c that could be a real dealbreaker for many.  
(and openwrt reputation)

ikcalB, 2025-08-07:
@getreu my current setup seems to be fixed with x24.10.2 (running 1w so far).

Hope for the best.
Br

ikcalB, 2025-08-13:
@getreu I had some intermittent connection timeouts.
Checking the logs, the same failure seems to happen, only the device is able to recover without a restart.

Would you mind checking your own logs?

getreu, 2025-08-13:
I would start by enabling a more verbose logging to get more context, e.g. what happened before the time out.