openwrt/openwrt #17651: HTTPS-DNS-Proxy not starting properly

ob45, 2025-01-17:
### Describe the bug

After restarting the router the HTTPS-DNS-Proxy service does not start and work properly despite it being set to "Enabled".

It does show up as "Enabled" and the service appears to be running but the service does not actually work.

I have to manually press the button "Restart" to make it run.

It is not possible that this is a RAM issue because enough RAM is available.

[image]

### OpenWrt version

r28304-6dacba30a7

### OpenWrt release

24.10.0-rc5

### OpenWrt target/subtarget

ramips/mt76x8

### Device

Xiaomi Mi Router 4A (100M International Edition V2)

### Image kind

Official downloaded image

### Steps to reproduce

1. Install HTTPS-DNS-Proxy
2. Configure the https-DNS servers in the service tab
3. Restart the router
4. Look at the HTTPS DNS Proxy tab and check the U.I
5. Check if HTTPS over DNS is working (it is not)

I've tried waiting up to half an hour after the restart because "maybe it just takes some time" but the only thing which actually makes it work is restarting the service manually - after which all blocked sites become accessible.

### Actual behaviour

The service appears to be running in the U.I but is extremely buggy.
I have to manually restart the service after restarting the router because the service does not automatically start properly.

Stopping the service can lead to nothing happening and the U.I keep showing that it is running.

[image]

### Expected behaviour

The service should start and work automatically after a router restart and route all DNS requests to the configured DNS servers and use DNS over HTTPS when the service is set as "Enabled".

The U.I should show th ecorrect status of the service.

### Additional info

This is a 64MB device but as shown in the screenshot enough RAM is available.

For me it is quite easy to check if it works or not because my ISP DNS blocks certain sites, which I cannot access without using DNS over HTTPS so I can just try to hit the site after a router restart and easily check it that way too.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-01-18:
Any log entries at startup like via `logread -e https-`

ob45, 2025-01-19:
> Any log entries at startup like via `logread -e https-`

```

Sun Jan 19 17:37:24 2025 user.notice https-dns-proxy: Starting service instances on_boot ✓✓✓
Sun Jan 19 17:37:24 2025 user.notice https-dns-proxy: Updating dnsmasq config ✓
Sun Jan 19 17:37:32 2025 user.notice https-dns-proxy: Restarting dnsmasq on_config_update ✓
Sun Jan 19 17:37:32 2025 user.notice https-dns-proxy: Setting service raw_trigger for 'interface.*.up'✓
Sun Jan 19 17:37:46 2025 user.notice https-dns-proxy: Stopping service on_failed_health_check ✓
Sun Jan 19 17:37:46 2025 user.notice https-dns-proxy: Setting service raw_trigger for 'interface.*.up'✓
```

Shine-, 2025-01-19:
Check with `uci show dhcp.@dnsmasq[0].server` whether your '127.0.0.1#505x' entries are there after a reboot. If they're missing, your issue might be related to the one I reported in openwrt/packages#23469.

Also, you might want to close this issue and re-open one in the proper repository [openwrt/packages](https://github.com/openwrt/packages/) to get more attention.

ob45, 2025-01-20:
> Check with `uci show dhcp.@dnsmasq[0].server` whether your `127.0.0.1#505x` entries are there after a reboot. If they're missing, your issue might be related to the one I reported in [openwrt/packages#23469](https://github.com/openwrt/packages/issues/23469).

I just tested it and that is very weird... idk what to make of this.

During normal operation:

```
root@OpenWrt:~# uci show dhcp.@dnsmasq[0].server
dhcp.cfg01411c.server='127.0.0.1#5053' '127.0.0.1#5054' '/mask.icloud.com/' '/mask-h2.icloud.com/' '/use-application-dns.net/' '127.0.0.1#5050' '127.0.0.1#5051' '127.0.0.1#5052'
```

After a reboot:

```
root@OpenWrt:~# uci show dhcp.@dnsmasq[0].server
dhcp.cfg01411c.server='127.0.0.1#5053' '127.0.0.1#5054' '/mask.icloud.com/' '/mask-h2.icloud.com/' '/use-application-dns.net/'
```

Looks like **_something_** gets loaded but not the entire / my config.

After restarting the service it looks normal again:

```
root@OpenWrt:~# uci show dhcp.@dnsmasq[0].server
dhcp.cfg01411c.server='127.0.0.1#5053' '127.0.0.1#5054' '/mask.icloud.com/' '/mask-h2.icloud.com/' '/use-application-dns.net/' '127.0.0.1#5050' '127.0.0.1#5051' '127.0.0.1#5052'
```

Idk what to make of this. The op in https://github.com/openwrt/packages/issues/23469 did say he got nothing at all... meanwhile I do get something but not my entire valid config....

ob45, 2025-01-21:
@Shine- 

Makes sense to me.

I'm closing this issue  because I opened a new one in https://github.com/openwrt/packages/issues/25797

Thanks for the help

closed 2025-01-21 by ob45