openwrt/openwrt #17655: Error 503 Backend unavailable, connection timeout

1junting, 2025-01-18:
### Describe the bug

Error 503 Backend unavailable, connection timeout
Backend unavailable, connection timeout

Error 54113
Details: cache-nrt-rjtf7700037-NRT 1737184134 953693898

Varnish cache server

### OpenWrt version

23

### OpenWrt release

23

### OpenWrt target/subtarget

23

### Device

x86

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

PunkFleet, 2025-01-18:
When i visit this https://downloads.openwrt.org/releases/

[image]

unr3al1089, 2025-01-18:
Same issue here

phiberoptick, 2025-01-18:
It's back up.

closed 2025-01-18 by zorun