openwrt/openwrt #17660: fstools: block-mount is missing necessary dependencies

muink, 2025-01-19:
### Describe the bug

I noticed when building the 24.10-rc5 x86_64 of the image using IB, the following error was reported:
``` shell
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency libuci20130104 for block-mount
 * pkg_hash_fetch_best_installation_candidate: Packages for block-mount found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for blockd:
 * 	libuci20130104
 * opkg_install_cmd: Cannot install package blockd.
```
https://github.com/fantastic-packages/packages/actions/runs/12845956359/job/35820591587#step:19:3834

I looked at the reason and it seems to be `libuci20130104_2023.08.10~5781664d-r1_x86_64` in https://downloads.openwrt.org has been upgraded to [libuci20250117_2025.01.17~fb3c2343-r1_x86_64](https://downloads.openwrt.org/releases/24.10.0-rc5/packages/x86_64/base/libuci20250117_2025.01.17~fb3c2343-r1_x86_64.ipk)

But `block-mount_2024.07.14~408c2cc4-r1_x86_64` still depends on `libuci20130104_2023.08.10~5781664d-r1_x86_64`
[image]

### OpenWrt version

r28211-d55754ce0d

### OpenWrt release

24.10.0-rc4

### OpenWrt target/subtarget

x86/64

### Device

QEMU Standard PC (Q35 + ICH9, 2009)

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

nerdhardest, 2025-01-19:
I'm having exactly the same problem; I'm using imagebuilder for gl-ar750s.
The device is supported; the repository was clean.

mic-user, 2025-01-19:
Same here.
 for Linksys WRT32X / Linksys Venom
 (mvebu/cortexa9)

`Signature check passed.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency libuci20130104 for block-mount
 * pkg_hash_fetch_best_installation_candidate: Packages for block-mount found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package block-mount.
make[2]: *** [Makefile:234: package_install] Error 255
make[1]: *** [Makefile:171: _call_manifest] Error 2
make: *** [Makefile:349: manifest] Error 2`

efahl, 2025-01-19:
I think this was a build issue.  Resolved now?

```
$ date
Sun Jan 19 09:39:18 PST 2025

$ opkg install block-mount
Installing block-mount (2024.07.14~408c2cc4-r1) to root...
Downloading https://downloads.openwrt.org/releases/24.10.0-rc5/targets/x86/64/packages/block-mount_2024.07.14~408c2cc4-r1_x86_64.ipk
Configuring block-mount.
```

nerdhardest, 2025-01-19:
Re: "Resolved now?"

Nope.  imagebuilder still fails in exactly the same way for gl-ar750s.

Loafter, 2025-01-19:
x86 release have same issues,
Image builder fails
With error:
 pkg_hash_check_unresolved: cannot find dependency libuci20130104 for block-mount

PedrosBle, 2025-01-19:
Dynalink DL-WRX36 24.10.0 rc5 firmware builder same error
pkg_hash_check_unresolved: cannot find dependency libuci20130104 for block-mount
pkg_hash_fetch_best_installation_candidate: Packages for block-mount found, but incompatible with the architectures configured

agevlakh, 2025-01-20:
Hello! Same problem... Banana Pi R2. 24.10-rc5

[image]

agevlakh, 2025-01-21:
I think developers needs to change the dependency to a more recent package from the repository: libuci20250117

closed 2025-01-22 by hauke