openwrt/openwrt #17661: Clients are unable to connect to the wireless network

grifos, 2025-01-19:
### Describe the bug

Since the first 24.10.0 RC releases, wireless clients could not establish a successful network connection. The OpenWrt logs were showing clients connected then disconnected after about 30 seconds. The clients showed similar logs but never had any actual network connectivity at all.

`Tue Jan  7 13:08:29 2025 daemon.info hostapd: wl1.101: STA aa:bb:cc:01:23:45 IEEE 802.11: authenticated`
`Tue Jan  7 13:08:29 2025 daemon.info hostapd: wl1.101: STA aa:bb:cc:01:23:45 IEEE 802.11: associated (aid 1)`
`Tue Jan  7 13:08:29 2025 daemon.notice hostapd: wl1.101: AP-STA-CONNECTED aa:bb:cc:01:23:45 auth_alg=open`
`Tue Jan  7 13:08:29 2025 daemon.info hostapd: wl1.101: STA aa:bb:cc:01:23:45 RADIUS: starting accounting session 7D707FBCB13EEE03`
`Tue Jan  7 13:08:29 2025 daemon.info hostapd: wl1.101: STA aa:bb:cc:01:23:45 WPA: pairwise key handshake completed (RSN)`
`Tue Jan  7 13:08:29 2025 daemon.notice hostapd: wl1.101: EAPOL-4WAY-HS-COMPLETED aa:bb:cc:01:23:45`
`Tue Jan  7 13:09:00 2025 daemon.notice hostapd: wl1.101: AP-STA-DISCONNECTED aa:bb:cc:01:23:45`
`Tue Jan  7 13:09:23 2025 daemon.info hostapd: wl1.101: STA aa:bb:cc:01:23:45 IEEE 802.11: authenticated`
`Tue Jan  7 13:09:23 2025 daemon.info hostapd: wl1.101: STA aa:bb:cc:01:23:45 IEEE 802.11: associated (aid 1)`
`Tue Jan  7 13:09:23 2025 daemon.notice hostapd: wl1.101: AP-STA-CONNECTED aa:bb:cc:01:23:45 auth_alg=open`
`Tue Jan  7 13:09:23 2025 daemon.info hostapd: wl1.101: STA aa:bb:cc:01:23:45 RADIUS: starting accounting session BB50B2A50135313A`
`Tue Jan  7 13:09:23 2025 daemon.info hostapd: wl1.101: STA aa:bb:cc:01:23:45 WPA: pairwise key handshake completed (RSN)`
`Tue Jan  7 13:09:23 2025 daemon.notice hostapd: wl1.101: EAPOL-4WAY-HS-COMPLETED aa:bb:cc:01:23:45`
`Tue Jan  7 13:10:08 2025 daemon.notice hostapd: wl1.101: AP-STA-DISCONNECTED aa:bb:cc:01:23:45`
`Tue Jan  7 13:10:10 2025 daemon.info hostapd: wl1.101: STA aa:bb:cc:01:23:45 IEEE 802.11: authenticated`
`Tue Jan  7 13:10:10 2025 daemon.info hostapd: wl1.101: STA aa:bb:cc:01:23:45 IEEE 802.11: associated (aid 1)`
`Tue Jan  7 13:10:10 2025 daemon.notice hostapd: wl1.101: AP-STA-CONNECTED aa:bb:cc:01:23:45 auth_alg=open`
`Tue Jan  7 13:10:10 2025 daemon.info hostapd: wl1.101: STA aa:bb:cc:01:23:45 RADIUS: starting accounting session 60129E9B594E6574`
`Tue Jan  7 13:10:10 2025 daemon.info hostapd: wl1.101: STA aa:bb:cc:01:23:45 WPA: pairwise key handshake completed (RSN)`
`Tue Jan  7 13:10:10 2025 daemon.notice hostapd: wl1.101: EAPOL-4WAY-HS-COMPLETED aa:bb:cc:01:23:45`
`Tue Jan  7 13:10:16 2025 daemon.notice hostapd: wl1.101: AP-STA-DISCONNECTED aa:bb:cc:01:23:45`

### OpenWrt version

r28304-6dacba30a7

### OpenWrt release

24.10.0-rc5

### OpenWrt target/subtarget

ramips/mt7621

### Device

Netgear WAX202

### Image kind

Official downloaded image

### Steps to reproduce

1- Configure a wireless network. Use a custom interface name, for example: `wl0.101`
2- On Network > Interfaces > Devices, configure the corresponding wireless device `wl0.101` with default settings
3- The wireless clients can no longer connect to that SSID. Restarting the wireless device seems to resolve but the problem reappears after a reboot.

### Actual behaviour

I reconfigured the AP from scratch and it looks like the WiFi fails when I configure the wireless network devices under Network > Interfaces > Devices.

I'm using custom wireless interface names and as long as the corresponding device is left unconfigured the WiFi works fine, while if I configure it, even if leaving everything default (like with the code below), the WiFi starts failing. Restarting the wireless device (on the Wireless page) seems to fix it, but after a reboot the problem reappears.

`uci add network device # =cfg1f0f15`
`uci set network.@device[-1].name='wl1.101'`

I then unconfigured the wireless devices on my other MT7621+MT7915E devices and now they're also working fine under 24.10.0-rc5.

### Expected behaviour

Clients should connect successfully to the wireless network even when the wireless device is configured on the OpenWrt device.

### Additional info

Thank you for the work you do guys, OpenWrt is magnificent.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-01-19:
In your log clients connect and authenticate, but do not reach dhcp. What is set in `option network`  for particular ap in etc/config/wireless

grifos, 2025-01-20:
The DHCP server is a Belkin RT1800 router running OpenWrt, the WAX202 is running as a dumb AP. The problem disappears if the AP runs 23.05 so it's not a problem on the DHCP server (running either 23.05 or 24.10).

The traffic isn't passing within the AP when running 24.10 and having the wireless devices configured. The RT1800 had the same problem (its own wireless connections failing) with the same version and configuration.

brada4, 2025-01-20:
Maybe you share "configurationn" that causes the issue.

grifos, 2025-01-20:
By "configuration" there I simply meant the wireless device being configured vs. unconfigured in Network > Interfaces > Devices. For the rest the configurations between the router (RT1800) and the AP (WAX202) are different as the WAX202 is running as a dumb AP.

It doesn't really look like a configuration issue, otherwise it wouldn't work under 23.05 either. Anyway, below is some configuration from the WAX202 AP, running 24.10.0-rc5:

`/etc/config/wireless` snippet for the SSID:

`config wifi-iface 'default_radio1'`
	`option device 'radio1'`
	`option network 'lan'`
	`option mode 'ap'`
	`option key 'redacted'`
	`option ssid 'redacted'`
	`option encryption 'psk2+ccmp'`
	`option ieee80211r '1'`
	`option ft_over_ds '0'`
	`option ft_psk_generate_local '1'`
	`option ifname 'wl1.101'`

`/etc/config/network` snippets for the `lan` device and interface and for the wireless device (with default settings), which when configured (eg. when it's present in `/etc/config/network`) causes the problem:

`config device`
	`option name 'br-lan.101'`
	`option type '8021q'`
	`option ifname 'br-lan'`
	`option vid '101'`
	`option ipv6 '0'`
	`option multicast '0'`

`config interface 'lan'`
	`option device 'br-lan.101'`
	`option proto 'static'`
	`option ipaddr 'redacted, management IP'`
	`option netmask '255.255.255.0'`
	`option delegate '0'`
	`option gateway 'redacted'`
	`list dns 'redacted'`
	`list dns_search 'redacted'`

`config device`
	`option name 'wl1.101'`

brada4, 2025-01-20:
What do you see in brctl show about bridge members?

grifos, 2025-01-20:
`brctl show` after a reboot, with the wireless device `wl1.101` configured and the wireless connection not establishing:

`bridge name	   bridge id		        STP enabled	    interfaces`
`br-lan		   3fdc.redacted                no		    lan2`
 `                                                                    wan`
 `							             wl1.101`
 `						                     lan3`
 `							             lan1` 

It is exactly the same as with the `wl1.101` unconfigured and successful network connectivity.

I suspect we may be wasting our time going back and forth, and it would be more productive if a developer with one of these MT7621+MT7915E devices at hand (they're extremely widespread after all) could reproduce the issue at their end and investigate it directly.

brada4, 2025-01-20:
Is there interface named `wl1` - the dot  in this case signifies vlan, which will not work with wifi. (colon would signify alias interface with different IP) Just rename interface with dash or underscore or anything else...

grifos, 2025-01-20:
That's not the case, I already replied to you about this last night [on the forum thread](https://forum.openwrt.org/t/openwrt-24-10-0-rc5-fifth-release-candidate/220854/140?u=grifo) and tried using a dash.

brada4, 2025-01-20:
You give very brief responses. Does base interface for suspected vlan exist or not?
I have similar dumb-ap setup and it works, have not tried renaming wifis.
Run tcpdump on wif,i on uplink port then on downlink port at other router-dhcp server. Then see how far reply returns.

grifos, 2025-01-29:
I've tested running 24.10.0-rc7 and the problem is not present if I call the wireless interface/device `qwerty` and configure it, so the interface syntax does make the difference, however using `wl1.101` was working fine under 23.05 so this looks like a regression, and it should be possible to use that interface naming syntax.

To answer your question, if for the base interface for the VLAN you mean the bridge subinterface `br-lan.101` yes it exists. No other related interfaces show up on `ifconfig`

I'm not going to run `tcpdump` as that doesn't seem useful to me here, this seems more like a bridging issue internal to the AP.

What I think is needed is a developer who can replicate the issue and look at it directly, if this is deemed to be worthy of fixing, else we can just close it but it may pop up again in future if someone else tries to use that interface naming syntax.

brada4, 2025-01-29:
Ckeck ../netifd changelog. Using corner naming scheme yields problem sooner or later.

grifos, 2025-01-29:
I wouldn't call `wl1.101` corner naming as it helps quickly identifying the interface and subinterface involved, just like the `br-lan.101` does without causing any problems. That should be the point of using custom wireless interface names, to recognize them quickly in logs. To me that was a useful feature in 22.03 and 23.05, but if it's no longer available in 24.10 then no problem.

Should I close the issue then or is there any chance that this would be tackled at some point even as a very low priority?

brada4, 2025-01-29:
Well, now it broke. It was always for vlans by default, obviously you are permitted to rename interfaces to unicode cats and dogs.
https://wiki.archlinux.org/title/VLAN#Create_the_VLAN_device

grifos, 2025-01-30:
The thing is, the problem doesn't seem to be the `.101` part as, like I mentioned above, I tried with a dash, eg. with `wl1-101`, and the issue was still there. In theory a dash should be okay.

```
Mon Jan 20 01:52:16 2025 daemon.notice hostapd: wl1-101: AP-STA-DISCONNECTED aa:bb:cc:01:23:45
Mon Jan 20 01:52:41 2025 daemon.info hostapd: wl1-101: STA aa:bb:cc:01:23:45 IEEE 802.11: authenticated
Mon Jan 20 01:52:41 2025 daemon.info hostapd: wl1-101: STA aa:bb:cc:01:23:45 IEEE 802.11: associated (aid 2)
Mon Jan 20 01:52:41 2025 daemon.notice hostapd: wl1-101: AP-STA-CONNECTED aa:bb:cc:01:23:45 auth_alg=open
Mon Jan 20 01:52:41 2025 daemon.info hostapd: wl1-101: STA aa:bb:cc:01:23:45 RADIUS: starting accounting session F4686DD41FDD70FD
Mon Jan 20 01:52:41 2025 daemon.info hostapd: wl1-101: STA aa:bb:cc:01:23:45 WPA: pairwise key handshake completed (RSN)
Mon Jan 20 01:52:41 2025 daemon.notice hostapd: wl1-101: EAPOL-4WAY-HS-COMPLETED aa:bb:cc:01:23:45
Mon Jan 20 01:53:26 2025 daemon.notice hostapd: wl1-101: AP-STA-DISCONNECTED aa:bb:cc:01:23:45
```

When I have some time I'll do some more tests to see at which character exactly this fails.

brada4, 2025-01-30:
Check with tcpdump if dhcp lands on the bridge where your dhcp server is running. Because wifi connects and authenticates, then no sign of dhcp.

grifos, 2025-01-30:
Yes the router+DHCP server side is fine, as I already mentioned, if the AP is running 23.05, with the router+DHCP server running either 23.05 or 24.10,  everything is fine, so the problem isn't on the router/DHCP server.

On the other hand, if the router runs 24.10, the wireless clients connecting directly to its wireless network have the same problem.

It looks like a bridging issue between the wireless and the wired subinterfaces, triggered by that naming syntax on the wireless subinterface.

brada4, 2025-01-30:
Alternative view is from ip-bridge and look at bridge vlan, may give better idea if something is linked together wrong or looks right but does not work right.

grifos, 2025-02-02:
The `bridge fdb` command shows something interesting.

The below is after I configure the interface as `wl1.101` and the device with default settings, then restart the wireless device from the Wireless page. As I noted previously, doing that gets it working but the problem reappears after a reboot. The MAC address is of the AP itself.

```
dd:ee:ff:56:78:90 dev wl1.101 vlan 101 master br-lan permanent
dd:ee:ff:56:78:90 dev wl1.101 master br-lan permanent
```

The below is after a reboot and the wireless clients unable to connect. Now interface `wl.101` is mapped to `vlan 1`, which is not used on the AP, and no longer to `vlan101` like above. This explains why this isn't working.

```
dd:ee:ff:56:78:90 dev wl1.101 vlan 1 master br-lan permanent
dd:ee:ff:56:78:90 dev wl1.101 master br-lan permanent
```

brada4, 2025-02-03:
I am not denying possibility of bug... Is there any difference in bridge or brctl between working and not?

grifos, 2025-02-03:
`brctl show` doesn't show any differences between working and not working, as I already posted above.

Other `bridge` commands are consistent with the previous ones:

Working:

```
root@a1:~# bridge vlan | grep 101
wan               101 PVID Egress Untagged
lan1              101 PVID Egress Untagged
lan2              101 PVID Egress Untagged
lan3              101
br-lan            101
wl1.101           101 PVID Egress Untagged
root@a1:~# 
```

Not working:

```
root@a1:~# bridge vlan | grep 101
wan               101 PVID Egress Untagged
lan1              101 PVID Egress Untagged
lan2              101 PVID Egress Untagged
lan3              101
br-lan            101
wl1.101           1 PVID Egress Untagged
root@a1:~# 
```

brada4, 2025-02-03:
Next:
save var/run/hostap*conf and compare bad to good. Ie if scripts generated wrong conf for hostap or hostap itself raced  to half-started bridge.

grifos, 2025-02-03:
`hostapd-phy1.conf` below, they look the same.

Working:

```
driver=nl80211
logger_syslog=127
logger_syslog_level=2
logger_stdout=127
logger_stdout_level=2
country_code=IT
ieee80211d=1
ieee80211h=1
hw_mode=a
beacon_int=100
stationary_ap=1
channel=149
chanlist=149

tx_queue_data2_burst=2.0

#num_global_macaddr=1
#macaddr_base=
ieee80211n=1
ht_coex=0
ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935]
ieee80211ac=1
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=155
vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][SU-BEAMFORMER][SU-BEAMFORMEE][MU-BEAMFORMER][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][SOUNDING-DIMENSION-2][BF-ANTENNA-2][MAX-MPDU-7991][MAX-A-MPDU-LEN-EXP7]

interface=wl1.101
ctrl_interface=/var/run/hostapd
ap_isolate=1
bss_load_update_period=60
chan_util_avg_period=600
disassoc_low_ack=1
skip_inactivity_poll=0
preamble=1
wmm_enabled=1
ignore_broadcast_ssid=0
uapsd_advertisement_enabled=1
utf8_ssid=1
multi_ap=0
wpa_passphrase=redacted
wpa_psk_file=/var/run/hostapd-wl1.101.psk
auth_algs=1
wpa=2
wpa_pairwise=CCMP
ssid=redacted
bridge=br-lan
wds_bridge=
snoop_iface=br-lan.101
wpa_disable_eapol_key_retries=0
wpa_key_mgmt=WPA-PSK FT-PSK
ft_iface=br-lan.101
mobility_domain=6405
ft_psk_generate_local=1
ft_over_ds=0
reassociation_deadline=1000
okc=0
disable_pmksa_caching=1
dynamic_vlan=0
vlan_naming=1
vlan_no_bridge=1
vlan_file=/var/run/hostapd-wl1.101.vlan
qos_map_set=0,0,2,16,1,1,255,255,18,22,24,38,40,40,44,46,48,56
bssid=dd:ee:ff:56:78:90
#default_macaddr
```

Not working:

```
driver=nl80211
logger_syslog=127
logger_syslog_level=2
logger_stdout=127
logger_stdout_level=2
country_code=IT
ieee80211d=1
ieee80211h=1
hw_mode=a
beacon_int=100
stationary_ap=1
channel=149
chanlist=149

tx_queue_data2_burst=2.0

#num_global_macaddr=1
#macaddr_base=
ieee80211n=1
ht_coex=0
ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935]
ieee80211ac=1
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=155
vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][SU-BEAMFORMER][SU-BEAMFORMEE][MU-BEAMFORMER][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][SOUNDING-DIMENSION-2][BF-ANTENNA-2][MAX-MPDU-7991][MAX-A-MPDU-LEN-EXP7]

interface=wl1.101
ctrl_interface=/var/run/hostapd
ap_isolate=1
bss_load_update_period=60
chan_util_avg_period=600
disassoc_low_ack=1
skip_inactivity_poll=0
preamble=1
wmm_enabled=1
ignore_broadcast_ssid=0
uapsd_advertisement_enabled=1
utf8_ssid=1
multi_ap=0
wpa_passphrase=redacted
wpa_psk_file=/var/run/hostapd-wl1.101.psk
auth_algs=1
wpa=2
wpa_pairwise=CCMP
ssid=redacted
bridge=br-lan
wds_bridge=
snoop_iface=br-lan.101
wpa_disable_eapol_key_retries=0
wpa_key_mgmt=WPA-PSK FT-PSK
ft_iface=br-lan.101
mobility_domain=6405
ft_psk_generate_local=1
ft_over_ds=0
reassociation_deadline=1000
okc=0
disable_pmksa_caching=1
dynamic_vlan=0
vlan_naming=1
vlan_no_bridge=1
vlan_file=/var/run/hostapd-wl1.101.vlan
qos_map_set=0,0,2,16,1,1,255,255,18,22,24,38,40,40,44,46,48,56
bssid=dd:ee:ff:56:78:90
#default_macaddr
```

brada4, 2025-02-03:
Summary Hostap bound to wrong bridge interface during early boot.

grifos, 2025-02-03:
No idea how to fix it myself, I hope you do :smiley:

brada4, 2025-02-03:
We should ask @rany2 @ynezz

rany2, 2025-02-03:
@grifos As a workaround, doing something like this should work:

1. Make a backup of your current configuration.
2. Create a new bridge `br-lan101` and add `br-lan.101` to it.
3. Reassign any interface for `br-lan.101` to `br-lan101`. For example, you'll need to change the `lan` interface from `br-lan.101` to `br-lan101`.
4. Run `/etc/init.d/network restart` or reboot.

grifos, 2025-02-04:
Hi @rany2, thank you. I've tested the workaround with a guest network on `vlan 502` and it works. `bridge` still shows the device mapped to `vlan 1`, but clients can connect.

In any case, as long as you've identified the bug is for me it's fine, for now I'll just leave the devices unconfigured.

Thanks again and to @brada4 

```
34:aa:bb:aa:bb:cc dev br-lan vlan 502 master br-lan permanent

34:aa:bb:aa:bb:cc dev br-lan.502 vlan 1 master br-lan502 permanent
34:aa:bb:aa:bb:cc dev br-lan.502 master br-lan502 permanent

36:aa:bb:aa:bb:ff dev wl1.502 vlan 1 master br-lan502 permanent
36:aa:bb:aa:bb:ff dev wl1.502 master br-lan502 permanent
```

```
root@a1:~# bridge vlan | grep 502
                  502
                  502
br-lan502         1 PVID Egress Untagged
br-lan.502        1 PVID Egress Untagged
wl1.502           1 PVID Egress Untagged
root@a1:~# 
```

e9hack, 2025-02-04:
> `brctl show` doesn't show any differences between working and not working, as I already posted above.
>
> Other `bridge` commands are consistent with the previous ones:
>
> Working:
>
> ```
> root@a1:~# bridge [...]

I see the same issue. To solve it, I'm using a hotplug script to change the vlan id. 
/etc/hotplug.d/net/20-bridge-vlans:
```
#!/bin/sh

check_pivd() {
	local IF=$1
	local wanted_pvid=${IF##*.}
	[ -n "$wanted_pvid" ] && {
		local pvid="$(bridge vlan ls dev "$IF" | sed -nre "s@$IF[[:space:]]+([0-9]+) PVID Egress Untagged\$@\1@ p")"
		[ -z "$pvid" ] && pvid="$(bridge vlan ls dev "$INTERFACE" | sed -nre "s@$INTERFACE[[:space:]]+([0-9]+)\$@\1@ p")"
		[ -n "$pvid" ] && {
			if [ "$pvid" != "$wanted_pvid" ]; then
				logger -t 20-bridge-vlans "$IF: pvid=$pvid, correcting it to $wanted_pvid"
				bridge vlan del vid $pvid dev "$IF"
				bridge vlan add vid $wanted_pvid dev "$IF" pvid untagged
			fi

		}
	}
}

[ "$ACTION" = add -a "$SUBSYSTEM" = net -a "$DEVTYPE" = wlan ] && {
	usleep 500000

	REG_EXPR="^phy[0-9]-ap([0-9]|[1-9][0-9]*).([0-9]|[1-9][0-9]*)$"
	[[ "$INTERFACE" =~ $REG_EXPR ]] && check_pivd "$INTERFACE"

	REG_EXPR="^[25]G-ap([0-9]|[1-9][0-9]*).([0-9]|[1-9][0-9]*)$"
	[[ "$INTERFACE" =~ $REG_EXPR ]] && check_pivd "$INTERFACE"

	REG_EXPR="^eth([0-9]|[1-9][0-9]*).([0-9]|[1-9][0-9]*)$"
	[[ "$INTERFACE" =~ $REG_EXPR ]] && check_pivd "$INTERFACE"

}
```
I see the following messages in the log:
`Tue Feb  4 18:16:36 2025 user.notice 20-bridge-vlans: 2G-ap2.201: pvid=1, correcting it to 201
`

The usleep delay at the beginning must be adapt to the router, because the interface is generated after triggering of the hotplug event.

brada4, 2025-02-04:
Either way it should not (attempt to} bridge unrelated vlans.

grifos, 2025-02-05:
Yep, clever workarounds, that I wouldn't have been able to come up with my relatively short OpenWrt experience - though I'm learning every day and lovin' it 😉  

I'll wait for the bug to be fixed as really at present I have no compelling reason to configure those devices, I did in my OpenWrt beginnings and then it became a habit more than a necessity.

I can still use the `wl1.101` like naming scheme for the wireless interfaces to show up on the logs (that is indeed useful) without needing to configure the device. In fact, I opened the issue more to flag the bug for the project's sake than for my own.

Kveri, 2025-04-22:
I think I hit the same bug. I have device named br-lan.10 as VLAN device (using vlan tag 10). Then I have interface br-lan.10 - protocol unmanaged. I don't need any IP on that because this is just an AP (it should just bridge wireless to wire). Under Wireless > MY-SSID in General Setup the field 'Network' I'm unable to fully select br-lan.10. I can actually click on it and GUI selects it, but Save button does nothing. In other words it seems I can't bridge the wireless SSID to respective interface. Therefore the client can authenticate but disconnects few seconds after auth because no DHCP reply. DHCP server is on a different router.