openwrt/openwrt #17664: UCI ISSUES X86/64 24.10.0-RC5

unr3al1089, 2025-01-19:
### Describe the bug

root@openwrt.lan:~# opkg upgrade uci
Upgrading uci on root from 2023.08.10~5781664d-r1 to 2025.01.17~fb3c2343-r1...
Downloading https://downloads.openwrt.org/releases/24.10.0-rc5/packages/x86_64/base/uci_2025.01.17~fb3c2343-r1_x86_64.ipk
Installing libuci20250117 (2025.01.17~fb3c2343-r1) to root...
Downloading https://downloads.openwrt.org/releases/24.10.0-rc5/packages/x86_64/base/libuci20250117_2025.01.17~fb3c2343-r1_x86_64.ipk
Collected errors:
 * check_data_file_clashes: Package libuci20250117 wants to install file /lib/libuci.so
        But that file is already provided by package  * libuci20130104

UCI Shows under available upgrades but errors, even tried a fresh install of the firmware.
:~# opkg list-upgradable
uci - 2023.08.10~5781664d-r1 - 2025.01.17~fb3c2343-r1

:~# opkg upgrade uci
Upgrading uci on root from 2023.08.10~5781664d-r1 to 2025.01.17~fb3c2343-r1...
Downloading https://downloads.openwrt.org/releases/24.10.0-rc5/packages/x86_64/base/uci_2025.01.17~fb3c2343-r1_x86_64.ipk
Installing libuci20250117 (2025.01.17~fb3c2343-r1) to root...
Downloading https://downloads.openwrt.org/releases/24.10.0-rc5/packages/x86_64/base/libuci20250117_2025.01.17~fb3c2343-r1_x86_64.ipk
Collected errors:
 * check_data_file_clashes: Package libuci20250117 wants to install file /lib/libuci.so
        But that file is already provided by package  * libuci20130104
:~# 

### OpenWrt version

24.10.0-RC5

### OpenWrt release

OpenWrt 24.10.0-rc5 r28304-6dacba30a7

### OpenWrt target/subtarget

UCI 

### Device

Micro-Star International Co., Ltd. MS-7C95

### Image kind

Official downloaded image

### Steps to reproduce

opkg update && opkg list-upgradable 

UCI shows in the list, when trying to upgrade/update it it fails. and qosmate displays UCI Parse errors, tested back 23.05.5 to make sure i didnt mess anything up but its not on my end ;p

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

efahl, 2025-01-19:
Using `opkg upgrade` is not recommended for many reasons (see https://openwrt.org/docs/guide-user/additional-software/opkg#upgrading_packages).  You should do a sysupgrade using a full image to avoid both the reported ABI problem, and any issues as described on the wiki. (Use one of the Attended Sysupgrade tools if you wish to retain any optional packages you may have installed, see LuCI ASU app, owut or the Firmware Selector.)

Synchronos, 2025-01-19:
I am seeing the same issue with OpenWrt 24.10.0-rc5 r28304-6dacba30a7 / LuCI openwrt-24.10 branch 25.014.55016~7046a1c. This appears to be a dependency chain maintenance problem.

Tynba, 2025-01-20:
I also encountered this on the A53 version

MichelHanzen, 2025-01-20:
Same problem with Firmware Version:
OpenWrt 24.10.0-rc5 r28304-6dacba30a7 / LuCI openwrt-24.10 branch 25.014.55016~7046a1c

smirkybg, 2025-01-20:
```
Upgrading uci on root from 2023.08.10~5781664d-r1 to 2025.01.17~fb3c2343-r1...
Downloading https://downloads.openwrt.org/releases/24.10.0-rc5/packages/aarch64_cortex-a53/base/uci_2025.01.17~fb3c2343-r1_aarch64_
Installing libuci20250117 (2025.01.17~fb3c2343-r1) to root...
Downloading https://downloads.openwrt.org/releases/24.10.0-rc5/packages/aarch64_cortex-a53/base/libuci20250117_2025.01.17~fb3c2343-
Collected errors:
 * check_data_file_clashes: Package libuci20250117 wants to install file /lib/libuci.so
        But that file is already provided by package  * libuci20130104
```

Pretty much the same on Linksys 8450 with 24.10.0-rc5.

pety3bi, 2025-01-20:
`Collected errors:
 * check_data_file_clashes: Package libuci20250117 wants to install file /lib/libuci.so
	But that file is already provided by package  * libuci20130104`

BPI-R4
OpenWrt 24.10.0-rc5 r28304-6dacba30a7 / LuCI openwrt-24.10 branch 25.014.55016~7046a1c

updated from OpenWrt 24.10.0-rc4 via owut

same issue

andrey-le, 2025-01-21:
same error, just tried to install dawn package

agevlakh, 2025-01-21:
Same problem with block-mount
https://github.com/openwrt/openwrt/issues/17660

nbd168, 2025-01-21:
Pushed another uci update in f9df8a90c8a44814df78e25bbda254a79f9d3785, which changes the library name to avoid that conflict for future builds.

closed 2025-01-21 by nbd168

MichelHanzen, 2025-01-21:
So it will be corrected in 24.10.0-rc6 or the 24.10.0 stable release?

iBug, 2025-01-23:
Just to note that upgrading isn't the only way to trigger this issue. Download the Image Builder tarball (`.tar.zst`), extract it and run `make image PROFILE=qihoo_360t7` produced the same for me:

```text
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency libuci20130104 for block-mount
 * pkg_hash_fetch_best_installation_candidate: Packages for block-mount found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package block-mount.
[...]
```

josteink, 2025-03-06:
I'm still having lots of opkg related issues, both on router and with imagebuilder. Common for both is the inability to obtain the packages I want/need.

When running image-builder I'm getting errors like this:

```
 Collected errors:
 * opkg_install_pkg: Checksum or size mismatch for package busybox. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package base-files.
 * opkg_install_pkg: Checksum or size mismatch for package busybox. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package busybox.
 * opkg_install_cmd: Cannot install package cgi-io.
 * opkg_install_cmd: Cannot install package curl.
 * opkg_install_cmd: Cannot install package ddns-scripts.
 * opkg_install_cmd: Cannot install package ddns-scripts-cloudflare.
 * opkg_install_cmd: Cannot install package ddns-scripts-services.
 * opkg_install_cmd: Cannot install package fping.
 * opkg_install_cmd: Cannot install package gzip.
 * opkg_install_pkg: Checksum or size mismatch for package hostapd-common. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package hostapd-common.
 * opkg_install_cmd: Cannot install package htop.
 * opkg_install_cmd: Cannot install package i2c-tools.
 * opkg_install_cmd: Cannot install package i2csfp.
 * opkg_install_cmd: Cannot install package iperf.
 * opkg_install_cmd: Cannot install package iperf3.
 * opkg_install_pkg: Checksum or size mismatch for package hostapd-common. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package kmod-mac80211.
 * opkg_install_cmd: Cannot install package libcap-ng.
 * opkg_install_cmd: Cannot install package libcurl.
 * opkg_install_cmd: Cannot install package libevdev.
 * opkg_install_cmd: Cannot install package libgnutls.
 * opkg_install_cmd: Cannot install package libi2c.
 * opkg_install_cmd: Cannot install package libimobiledevice.
 * opkg_install_cmd: Cannot install package libimobiledevice-glue.
 * opkg_install_cmd: Cannot install package libiperf3.
 * opkg_install_cmd: Cannot install package libkmod.
 * opkg_install_cmd: Cannot install package libminiupnpc.
 * opkg_install_cmd: Cannot install package libnghttp2.
 * opkg_install_pkg: Checksum or size mismatch for package libopenssl3. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package libopenssl.
 * opkg_install_cmd: Cannot install package libpci.
 * opkg_install_cmd: Cannot install package libplist.
 * opkg_install_cmd: Cannot install package libsensors.
 * opkg_install_cmd: Cannot install package libudev-zero.
 * opkg_install_cmd: Cannot install package libusbmuxd.
 * opkg_install_pkg: Checksum or size mismatch for package lldpd. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package lldpd.
 * opkg_install_cmd: Cannot install package lm-sensors.
 * opkg_install_cmd: Cannot install package luasocket.
 * pkg_hash_check_unresolved: cannot find dependency cgi-io for luci-base
 * pkg_hash_fetch_best_installation_candidate: Packages for luci-base found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci:
 * 	cgi-io
 * opkg_install_cmd: Cannot install package luci.
 * pkg_hash_check_unresolved: cannot find dependency attendedsysupgrade-common for luci-app-attendedsysupgrade
 * pkg_hash_check_unresolved: cannot find dependency cgi-io for luci-app-attendedsysupgrade
 * pkg_hash_fetch_best_installation_candidate: Packages for luci-app-attendedsysupgrade found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package luci-app-attendedsysupgrade.
 * pkg_hash_check_unresolved: cannot find dependency ddns-scripts for luci-app-ddns
 * pkg_hash_fetch_best_installation_candidate: Packages for luci-app-ddns found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package luci-app-ddns.
 * opkg_install_pkg: Checksum or size mismatch for package lldpd. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package luci-app-lldpd.
 * pkg_hash_check_unresolved: cannot find dependency nlbwmon for luci-app-nlbwmon
 * pkg_hash_fetch_best_installation_candidate: Packages for luci-app-nlbwmon found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package luci-app-nlbwmon.
 * pkg_hash_check_unresolved: cannot find dependency miniupnpd for luci-app-upnp
 * pkg_hash_fetch_best_installation_candidate: Packages for luci-app-upnp found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package luci-app-upnp.
 * opkg_install_cmd: Cannot install package miniupnpd-nftables.
 * opkg_install_cmd: Cannot install package msmtp.
[... 32 lines cut ...]
Signature check passed.
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/aarch64_cortex-a53/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/aarch64_cortex-a53/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/aarch64_cortex-a53/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
[... cut ...]
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/aarch64_cortex-a53/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/aarch64_cortex-a53/telephony/Packages.sig
Signature check failed.
Remove wrong Signature file.
```

LIke @iBug , I'm also getting this for target `mediatek-filogic` (BananaPi R4 to be specific).

Have the opkg repos mirros been replicated incorrectly?