openwrt/openwrt #17666: MT-6000: kernel crash

systemcrash, 2025-01-19:
### Describe the bug

I cannot be certain but I previously had a config that contained a VLAN, and it seemed to precipitate a similar or identical crash to this. 

So MT-6000 and VLANs seem to trigger this crash. This may or may not have been resolved in master already.

What seems unfortunate is that eth0 comes up again at 2.5G, but no device connected handles 2.5G, so that link is effectively severed.

### OpenWrt version

r24106-10cc5fcd00

### OpenWrt release

23.05.5

### OpenWrt target/subtarget

mediatek/filogic

### Device

GL.iNet GL-MT6000

### Image kind

Official downloaded image

### Steps to reproduce

unknown.

### Actual behaviour

kernel crash. What seems unfortunate is that eth0 comes up again at 2.5G, but no device connected handles 2.5G, so that link is effectively severed.

```
[83117.527362] lightbridge: port 1(phy1-ap2) entered disabled state
[83117.535670] device phy1-ap2 left promiscuous mode
[83117.540493] lightbridge: port 1(phy1-ap2) entered disabled state
[83117.731203] lightbridge: port 1(phy1-ap2) entered blocking state
[83117.737264] lightbridge: port 1(phy1-ap2) entered disabled state
[83117.743493] device phy1-ap2 entered promiscuous mode
[83117.750178] lightbridge: port 1(phy1-ap2) entered blocking state
[83117.756182] lightbridge: port 1(phy1-ap2) entered forwarding state
[759130.052736] ------------[ cut here ]------------
[759130.057445] NETDEV WATCHDOG: eth1 (mtk_soc_eth): transmit queue 1 timed out
[759130.064508] WARNING: CPU: 0 PID: 0 at 0xffffffc0086c6464
[759130.069890] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet wireguard pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack_netlink nf_conntrack mt7915e mt76_connac_lib mt76 mac80211 libchacha20poly1305 chacha_neon cfg80211 usblp slhc poly1305_neon nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcurve25519_generic libcrc32c libchacha compat crypto_safexcel dummy sit tunnel4 ip_tunnel veth vxlan udp_tunnel ip6_udp_tunnel sha1_generic seqiv md5 des_generic libdes authencesn authenc leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd gpio_button_hotplug usbcore usb_common aquantia crc_ccitt
[759130.070027] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 5.15.167 #0
[759130.070033] Hardware name: GL.iNet GL-MT6000 (DT)
[759130.070036] pstate: 40400005 (nZcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[759130.070041] pc : 0xffffffc0086c6464
[759130.070043] lr : 0xffffffc0086c6464
[759130.070046] sp : ffffffc008003db0
[759130.070048] x29: ffffffc008003db0 x28: 0000000000000140 x27: 00000000ffffffff
[759130.070054] x26: 0000000000000000 x25: 0000000000000000 x24: ffffff8000ccd4c0
[759130.070061] x23: 0000000000000000 x22: 0000000000000001 x21: ffffffc008a46000
[759130.070067] x20: ffffff8000ccd000 x19: 0000000000000001 x18: ffffffc008a5a2a8
[759130.070074] x17: ffffffc03735f000 x16: ffffffc008000000 x15: 00000000000003d8
[759130.070080] x14: 0000000000000148 x13: ffffffc008003ae8 x12: ffffffc008ab22a8
[759130.070086] x11: 712074696d736e61 x10: ffffffc008ab22a8 x9 : 0000000000000000
[759130.070093] x8 : ffffffc008a5a258 x7 : ffffffc008a5a2a8 x6 : 0000000000000001
[759130.070099] x5 : 0000000000000000 x4 : 0000000000000000 x3 : 0000000000000000
[759130.070105] x2 : ffffff803fd8d090 x1 : ffffffc03735f000 x0 : 000000000000003f
[759130.070112] Call trace:
[759130.070115]  0xffffffc0086c6464
[759130.070118]  0xffffffc0080a5b50
[759130.070120]  0xffffffc0080a63c8
[759130.070123]  0xffffffc0080a6918
[759130.070125]  0xffffffc00803b04c
[759130.070127]  0xffffffc00803b404
[759130.070130]  0xffffffc00803b5d0
[759130.070133]  0xffffffc008089454
[759130.070135]  0xffffffc0080100e0
[759130.070138]  0xffffffc00801584c
[759130.070140]  0xffffffc00801609c
[759130.070143]  0xffffffc008835d60
[759130.070145]  0xffffffc008836514
[759130.070148]  0xffffffc008011374
[759130.070150]  0xffffffc008836af4
[759130.070152]  0xffffffc008069350
[759130.070155]  0xffffffc008069560
[759130.070157]  0xffffffc008836e94
[759130.070160]  0xffffffc0089d0954
[759130.070162]  0xffffffc0089d0f2c
[759130.070165]  0xffffffc0089d034c
[759130.070167] ---[ end trace 7e078d408e8f6139 ]---
[759130.074868] mtk_soc_eth 15100000.ethernet eth1: transmit timed out
[759130.250105] mtk_soc_eth 15100000.ethernet eth0: Link is Down
[759130.255891] mtk_soc_eth 15100000.ethernet eth1: Link is Down
[759130.912910] mtk_soc_eth 15100000.ethernet: warm reset failed
[759130.933906] mtk_soc_eth 15100000.ethernet eth0: configuring for fixed/2500base-x link mode
[759130.942416] mtk_soc_eth 15100000.ethernet eth0: Link is Up - 2.5Gbps/Full - flow control rx/tx
[759131.261582] mtk_soc_eth 15100000.ethernet eth1: PHY [mdio-bus:01] driver [RTL8221B-VB-CG 2.5Gbps PHY] (irq=62)
[759131.271916] mtk_soc_eth 15100000.ethernet eth1: configuring for phy/sgmii link mode
[759134.169142] mtk_soc_eth 15100000.ethernet eth1: Link is Up - 1Gbps/Full - flow control rx/tx
[759191.043490] page_pool_release_retry() stalled pool shutdown 6 inflight 60 sec
[759251.326192] page_pool_release_retry() stalled pool shutdown 6 inflight 120 sec
[759311.613289] page_pool_release_retry() stalled pool shutdown 6 inflight 180 sec
[759371.912090] page_pool_release_retry() stalled pool shutdown 6 inflight 241 sec
[759432.191519] page_pool_release_retry() stalled pool shutdown 6 inflight 301 sec
[759492.481222] page_pool_release_retry() stalled pool shutdown 6 inflight 361 sec
[759552.771014] page_pool_release_retry() stalled pool shutdown 6 inflight 422 sec
[759613.060844] page_pool_release_retry() stalled pool shutdown 6 inflight 482 sec
[759673.350686] page_pool_release_retry() stalled pool shutdown 6 inflight 542 sec
[759733.650531] page_pool_release_retry() stalled pool shutdown 6 inflight 603 sec
[759793.940385] page_pool_release_retry() stalled pool shutdown 6 inflight 663 sec
[759854.220250] page_pool_release_retry() stalled pool shutdown 6 inflight 723 sec
[759914.510096] page_pool_release_retry() stalled pool shutdown 6 inflight 784 sec
[759974.800049] page_pool_release_retry() stalled pool shutdown 6 inflight 844 sec
[760035.089848] page_pool_release_retry() stalled pool shutdown 6 inflight 904 sec
[760095.379668] page_pool_release_retry() stalled pool shutdown 6 inflight 965 sec
[760155.669514] page_pool_release_retry() stalled pool shutdown 6 inflight 1025 sec
[760215.959385] page_pool_release_retry() stalled pool shutdown 6 inflight 1085 sec
...
[828403.785619] page_pool_release_retry() stalled pool shutdown 6 inflight 69307 sec
```

logs flood continuously with `page_pool_release_retry() stalled pool shutdown 6 inflight xxx sec` every >= 60ish seconds.

### Diffconfig

```text
`-ash: ./scripts/diffconfig.sh: not found`
```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

zhaojh329, 2025-01-20:
Please use OpenWrt-24.10.

systemcrash, 2025-01-22:
Happy to move up to the release version when it exists. Logged this crash in case others encounter it and for posterity.

xirixiz, 2025-01-24:
If have been using 23.05.5 with a complex setup, but without any crashes. I now moved to 24.10-rc4 and above releases, and also I`m not experiencing any issues like you encounter. However, I try to keep things as vanilla as possible, meaning I only install and configure packages I really need. 

This could be related to bad config, but then you need to share more logs and config.

You could also create a backup, then upgrade to 24.10-rc6 and see if that solves the issue. If not, then it's almost bad config for sure.

hayzamjs, 2025-01-27:
@xirixiz This is a kernel panic. Regardless of what a user does in userland, it should never result in a kernel panic under any circumstances. While updating to 24 might be a potential solution, we need to acknowledge that 23.05 is still officially supported, and this issue should be addressed.

hayzamjs, 2025-01-27:
Ah, you're right @xirixiz, it's my bad skimming the logs, I've had instability with the same machine where the system does go into a full blown panic right after the driver-level failure occurs, I will open a new thread for that since it seems it's unrelated to this (atleast from first glance)

xirixiz, 2025-01-28:
Ah no worries, it's not about right or wrong, it's about understanding and trying to solve the issue together 😉 . But intereseting that your device goes into a kernel panic.