openwrt/openwrt #17673: Banana Pi R4 SFP Not Connecting

RonWilliam547, 2025-01-20:
### Describe the bug

I apologize in advance if this is not the correct place to put this and for all I know, this might be more user error than a bug but I have tried 2 different SFP to RJ-45 converters and neither one of them will ever give a connection. I have tried a 10G XiCom and a 1G ipolex and neither one of them connects. From what I can tell in the logs, the XiCom is closer to working but I could absolutely be wrong as I'm not familiar with SFP other than the little bit of research I've done in the past few days trying to get this to work. I have tried turning off autoneg and setting the speed manually as well as setting full duplex but nothing seems to work. I have included my logs below. I would appreciate any guidance you can provide or any ideas that you might have.

**XiCom**

**ethtool -m sfp-lan output**

root@OpenWrt:~# ethtool -m sfp-lan
	Identifier                                : 0x03 (SFP)
	Extended identifier                       : 0x04 (GBIC/SFP defined by 2-wire interface ID)
	Connector                                 : 0x07 (LC)
	Transceiver codes                         : 0x10 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
	Transceiver type                          : 10G Ethernet: 10G Base-SR
	Encoding                                  : 0x06 (64B/66B)
	BR, Nominal                               : 10300MBd
	Rate identifier                           : 0x00 (unspecified)
	Length (SMF,km)                           : 0km
	Length (SMF)                              : 0m
	Length (50um)                             : 80m
	Length (62.5um)                           : 20m
	Length (Copper)                           : 0m
	Length (OM3)                              : 300m
	Laser wavelength                          : 850nm
	Vendor name                               : OEM
	Vendor OUI                                : 00:00:00
	Vendor PN                                 : SFP+-T30
	Vendor rev                                : 1
	Option values                             : 0x00 0x1a
	Option                                    : RX_LOS implemented
	Option                                    : TX_FAULT implemented
	Option                                    : TX_DISABLE implemented
	BR margin, max                            : 0%
	BR margin, min                            : 0%
	Vendor SN                                 : C202412310055
	Date code                                 : 250102
	Optical diagnostics support               : Yes
	Laser bias current                        : 6.000 mA
	Laser output power                        : 0.5000 mW / -3.01 dBm
	Receiver signal average optical power     : 0.4000 mW / -3.98 dBm
	Module temperature                        : 51.08 degrees C / 123.94 degrees F
	Module voltage                            : 3.3054 V
	Alarm/warning flags implemented           : Yes
	Laser bias current high alarm             : Off
	Laser bias current low alarm              : Off
	Laser bias current high warning           : Off
	Laser bias current low warning            : Off
	Laser output power high alarm             : Off
	Laser output power low alarm              : Off
	Laser output power high warning           : Off
	Laser output power low warning            : Off
	Module temperature high alarm             : Off
	Module temperature low alarm              : Off
	Module temperature high warning           : Off
	Module temperature low warning            : Off
	Module voltage high alarm                 : Off
	Module voltage low alarm                  : Off
	Module voltage high warning               : Off
	Module voltage low warning                : Off
	Laser rx power high alarm                 : Off
	Laser rx power low alarm                  : Off
	Laser rx power high warning               : Off
	Laser rx power low warning                : Off
	Laser bias current high alarm threshold   : 15.000 mA
	Laser bias current low alarm threshold    : 1.000 mA
	Laser bias current high warning threshold : 13.000 mA
	Laser bias current low warning threshold  : 2.000 mA
	Laser output power high alarm threshold   : 1.9952 mW / 3.00 dBm
	Laser output power low alarm threshold    : 0.1584 mW / -8.00 dBm
	Laser output power high warning threshold : 1.7782 mW / 2.50 dBm
	Laser output power low warning threshold  : 0.1778 mW / -7.50 dBm
	Module temperature high alarm threshold   : 95.00 degrees C / 203.00 degrees F
	Module temperature low alarm threshold    : -50.00 degrees C / -58.00 degrees F
	Module temperature high warning threshold : 90.00 degrees C / 194.00 degrees F
	Module temperature low warning threshold  : -45.00 degrees C / -49.00 degrees F
	Module voltage high alarm threshold       : 3.6000 V
	Module voltage low alarm threshold        : 3.0000 V
	Module voltage high warning threshold     : 3.5000 V
	Module voltage low warning threshold      : 3.1000 V
	Laser rx power high alarm threshold       : 1.5849 mW / 2.00 dBm
	Laser rx power low alarm threshold        : 0.0199 mW / -17.01 dBm
	Laser rx power high warning threshold     : 1.2589 mW / 1.00 dBm
	Laser rx power low warning threshold      : 0.0251 mW / -16.00 dBm
root@OpenWrt:~# 

**dmesg | grep sfp-lan output**

[image]

**ethtool sfp-lan output**

[image]

**Ipolex**

**ethtool -m sfp-lan output**

root@OpenWrt:~# ethtool -m sfp-lan
	Identifier                                : 0x03 (SFP)
	Extended identifier                       : 0x04 (GBIC/SFP defined by 2-wire interface ID)
	Connector                                 : 0x00 (unknown or unspecified)
	Transceiver codes                         : 0x00 0x00 0x00 0x08 0x00 0x00 0x00 0x00 0x00
	Transceiver type                          : Ethernet: 1000BASE-T
	Encoding                                  : 0x01 (8B/10B)
	BR, Nominal                               : 1300MBd
	Rate identifier                           : 0x00 (unspecified)
	Length (SMF,km)                           : 0km
	Length (SMF)                              : 0m
	Length (50um)                             : 0m
	Length (62.5um)                           : 0m
	Length (Copper)                           : 100m
	Length (OM3)                              : 0m
	Laser wavelength                          : 0nm
	Vendor name                               : OEM
	Vendor OUI                                : 00:00:00
	Vendor PN                                 : SFP-GE-T
	Vendor rev                                : 
	Option values                             : 0x00 0x1a
	Option                                    : RX_LOS implemented
	Option                                    : TX_FAULT implemented
	Option                                    : TX_DISABLE implemented
	BR margin, max                            : 0%
	BR margin, min                            : 0%
	Vendor SN                                 : CSGE2O59225
	Date code                                 : 24052701
root@OpenWrt:~#

**dmesg | grep sfp-lan output**

[image]

**ethtool sfp-lan output**

[image]

### OpenWrt version

r28628-0f8a5a5fd6

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R4

### Image kind

Official downloaded image

### Actual behaviour

Computer can't establish a connection via the SFP to RJ-45 adapter

### Expected behaviour

Computer to get a connection via the adapter

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

dangowrt, 2025-01-31:
Probably you need to install `kmod-phy-marvell` for this module to work.

RonWilliam547, 2025-01-31:
> Probably you need to install `kmod-phy-marvell` for the 1G `SFP-GE-T` module to work. For the 10G XiCom module we may need a quirk to enable PHY access over I2C similar to other 10G RJ-45 modules, e.g. FS SFP-10G-T (see `drivers/net/phy/sfp.c` in Linux kernel source). And maybe `kmod-phy-marvell-10g` should the 10G PHY be using a Marvell PHY, or `kmod-phy-aquantia` should it be using Aquantia (I haven't seen anything but those two in 10G RJ45 SFPs)

Okay, thank you. I'll try these. From my understanding, the XiCom is supposed to be very universal and from the logs, looks to be the closest to working but I absolutely could be wrong about that. Thank you again for the suggestions, I'll let you know how it goes. 😊

RonWilliam547, 2025-02-05:
> Probably you need to install `kmod-phy-marvell` for the 1G `SFP-GE-T` module to work. For the 10G XiCom module we may need a quirk to enable PHY access over I2C similar to other 10G RJ-45 modules, [...]

Hey, so I finally tried to get this working and whenever I try to install kmod-phy-marvell or kmod-phy-marvell-10g it says "kmod-phy-marvell (no such package)"
required by: world(kmod-phy-marvell)

blunden, 2025-11-17:
@RonWilliam547 Which specific Xicom 10GBASE-T transceiver is that? Is it the Marvell/Aquantia based one rated for 30 meters or the Broadcom based one rated for 80 meters?

Looking at the product name (SFP+-T30) in the EEPROM data you posted, it sounds like you have the Aquantia/Marvell AQR113C based model so that `kmod-phy-aquantia` package might be needed. Perhaps the issue that caused you to run a snapshot build has been fixed in the stable builds so that you can run a build where package installation is easier? 🙂  

I have the Broadcom based one ("SFP+-T80" written on the label and "SFP-10G-T100" in the EEPROM data) and it works fine in the BPI-R4 with 24.10.x, at least with the EEPROM data they configured it with when I ordered it.

I also have two other transceivers using the same Broadcom chip, one from FlexOptix (T.C96.04.KF) and one from ZYOPM (rated for 100 meters). They also work fine at 10 Gbps, seemingly without loading a PHY driver at all (dmesg claims "no PHY detected"). Support for lower speeds is not guaranteed though.

@Nico-Hard Many 10GBASE-T SFP+ transceivers only officially support 10 Gbps, none of the lower speeds.

mhalano, 2026-01-03:
I think I have the same problem as you. I also have a module like that, but it just announces 10GbE, even it should announce 1GbE and 2.5GbE.
```
	Identifier                                : 0x03 (SFP)
	Extended identifier                       : 0x04 (GBIC/SFP defined by 2-wire interface ID)
	Connector                                 : 0x07 (LC)
	Transceiver codes                         : 0x00 0x00 0x00 0x01 0x20 0x40 0x0c 0x05 0x00
	Transceiver type                          : FC: 100 MBytes/sec
	Encoding                                  : 0x01 (8B/10B)
	BR Nominal                                : 1300MBd
	Rate identifier                           : 0x00 (unspecified)
	Length (SMF)                              : 0km
	Length (OM2)                              : 550m
	Length (OM1)                              : 270m
	Length (Copper or Active cable)           : 0m
	Length (OM3)                              : 0m
	Laser wavelength                          : 850nm
	Vendor name                               : OEM
	Vendor OUI                                : 00:1b:21
	Vendor PN                                 : SFP-10G-T
	Vendor rev                                : 1
	Option values                             : 0x00 0x1a
	Option                                    : TX_DISABLE implemented
	BR margin max                             : 0%
	BR margin min                             : 0%
	Vendor SN                                 : CI202511040410
	Date code                                 : 251104
	Optical diagnostics support               : Yes
	Laser bias current                        : 6.000 mA
	Laser output power                        : 0.5000 mW / -3.01 dBm
	Receiver signal average optical power     : 0.0000 mW / -inf dBm
	Module temperature                        : 46.80 degrees C / 116.23 degrees F
	Module voltage                            : 3.3573 V
	Alarm/warning flags implemented           : Yes
	Laser bias current high alarm             : Off
	Laser bias current low alarm              : Off
	Laser bias current high warning           : Off
	Laser bias current low warning            : Off
	Laser output power high alarm             : Off
	Laser output power low alarm              : Off
	Laser output power high warning           : Off
	Laser output power low warning            : Off
	Module temperature high alarm             : Off
	Module temperature low alarm              : Off
	Module temperature high warning           : Off
	Module temperature low warning            : Off
	Module voltage high alarm                 : Off
	Module voltage low alarm                  : Off
	Module voltage high warning               : Off
	Module voltage low warning                : Off
	Laser rx power high alarm                 : Off
	Laser rx power low alarm                  : On
	Laser rx power high warning               : Off
	Laser rx power low warning                : On
	Laser bias current high alarm threshold   : 15.000 mA
	Laser bias current low alarm threshold    : 1.000 mA
	Laser bias current high warning threshold : 13.000 mA
	Laser bias current low warning threshold  : 2.000 mA
	Laser output power high alarm threshold   : 1.9952 mW / 3.00 dBm
	Laser output power low alarm threshold    : 0.1584 mW / -8.00 dBm
	Laser output power high warning threshold : 1.7782 mW / 2.50 dBm
	Laser output power low warning threshold  : 0.1778 mW / -7.50 dBm
	Module temperature high alarm threshold   : 95.00 degrees C / 203.00 degrees F
	Module temperature low alarm threshold    : -50.00 degrees C / -58.00 degrees F
	Module temperature high warning threshold : 90.00 degrees C / 194.00 degrees F
	Module temperature low warning threshold  : -45.00 degrees C / -49.00 degrees F
	Module voltage high alarm threshold       : 3.6000 V
	Module voltage low alarm threshold        : 3.0000 V
	Module voltage high warning threshold     : 3.5000 V
	Module voltage low warning threshold      : 3.1000 V
	Laser rx power high alarm threshold       : 1.5849 mW / 2.00 dBm
	Laser rx power low alarm threshold        : 0.0199 mW / -17.01 dBm
	Laser rx power high warning threshold     : 1.2589 mW / 1.00 dBm
	Laser rx power low warning threshold      : 0.0251 mW / -16.00 dBm

```

danpawlik, 2026-01-03:
@mhalano could you send output:
- `apk list -I | grep kmod`
- `dmesg |grep -i sfp`

mhalano, 2026-01-03:
Give some minutes. I added the kernel module for marvell and I will try it. I already had for Aquantia, and the module was loaded, but I don't know if there is some kind of mismatch recognizing the device.
EDIT: Here it is. Fresh boot with Marvell drivers installed:
```
root@router:~# dmesg|grep -i sfp
[    0.000000] Machine model: Banana Pi BPI-R4 (2x SFP+)
[    6.431532] mtk_soc_eth 15100000.ethernet sfp-lan: renamed from eth1
[    6.439531] mtk_soc_eth 15100000.ethernet sfp-wan: renamed from eth2
[   12.600050] sfp sfp1: Host maximum power 3.0W
[   12.604960] sfp sfp2: Host maximum power 3.0W
[   12.922120] sfp sfp2: module OEM              DAC-10G-5M       rev A0   sn C202507070011    dc 250707
[   12.931566] mtk_soc_eth 15100000.ethernet sfp-lan: switched to inband/10gbase-r link mode
[   12.951054] sfp sfp1: module OEM              SFP-10G-T        rev 1    sn CI202511040410   dc 251104
[   16.878365] mtk_soc_eth 15100000.ethernet sfp-lan: configuring for inband/10gbase-r link mode
[   16.902880] br-management: port 1(sfp-lan.1) entered blocking state
[   16.909170] br-management: port 1(sfp-lan.1) entered disabled state
[   16.915510] sfp-lan.1: entered allmulticast mode
[   16.920128] mtk_soc_eth 15100000.ethernet sfp-lan: entered allmulticast mode
[   16.927354] sfp-lan.1: entered promiscuous mode
[   16.931920] mtk_soc_eth 15100000.ethernet sfp-lan: entered promiscuous mode
[   16.942600] br-servers: port 1(sfp-lan.20) entered blocking state
[   16.948709] br-servers: port 1(sfp-lan.20) entered disabled state
[   16.954925] sfp-lan.20: entered allmulticast mode
[   16.959795] sfp-lan.20: entered promiscuous mode
[   16.983454] br-iot: port 1(sfp-lan.10) entered blocking state
[   16.989213] br-iot: port 1(sfp-lan.10) entered disabled state
[   16.995145] sfp-lan.10: entered allmulticast mode
[   17.000046] sfp-lan.10: entered promiscuous mode
[   17.002844] mtk_soc_eth 15100000.ethernet sfp-lan: Link is Up - 10Gbps/Full - flow control off
[   17.015888] br-management: port 1(sfp-lan.1) entered blocking state
[   17.022185] br-management: port 1(sfp-lan.1) entered forwarding state
[   17.029377] br-servers: port 1(sfp-lan.20) entered blocking state
[   17.035528] br-servers: port 1(sfp-lan.20) entered forwarding state
[   17.043588] br-iot: port 1(sfp-lan.10) entered blocking state
[   17.049356] br-iot: port 1(sfp-lan.10) entered forwarding state
[   17.197135] br-lan: port 4(sfp-lan.15) entered blocking state
[   17.202948] br-lan: port 4(sfp-lan.15) entered disabled state
[   17.208714] sfp-lan.15: entered allmulticast mode
[   17.213604] sfp-lan.15: entered promiscuous mode
[   17.220201] mtk_soc_eth 15100000.ethernet sfp-wan: configuring for inband/10gbase-r link mode
[   17.242845] br-lan: port 5(sfp-wan) entered blocking state
[   17.248348] br-lan: port 5(sfp-wan) entered disabled state
[   17.253900] mtk_soc_eth 15100000.ethernet sfp-wan: entered allmulticast mode
[   17.261116] mtk_soc_eth 15100000.ethernet sfp-wan: entered promiscuous mode
[   17.269343] br-lan: port 4(sfp-lan.15) entered blocking state
[   17.275130] br-lan: port 4(sfp-lan.15) entered forwarding state
[   24.932405] mtk_soc_eth 15100000.ethernet sfp-wan: validation with support 00,00000000,00000000,00000000 failed: -EINVAL
[   24.943346] sfp sfp1: sfp_add_phy failed: -EINVAL
```
```
root@router:~# apk list -I|grep kmod
kmod-cfg80211-6.12.63.6.18-r1 aarch64_cortex-a53 {feeds/base/kernel/mac80211} (GPL-2.0-only) [installed]
kmod-crypto-aead-6.12.63-r1 aarch64_cortex-a53 {feeds/base/kernel/linux} (GPL-2.0-only) [installed]
kmod-crypto-authenc-6.12.63-r1 aarch64_cortex-a53 {feeds/base/kernel/linux} (GPL-2.0-only) [installed]
kmod-crypto-ccm-6.12.63-r1 aarch64_cortex-a53 {feeds/base/kernel/linux} (GPL-2.0-only) [installed]
kmod-crypto-cmac-6.12.63-r1 aarch64_cortex-a53 {feeds/base/kernel/linux} (GPL-2.0-only) [installed]
kmod-crypto-crc32c-6.12.63-r1 aarch64_cortex-a53 {feeds/base/kernel/linux} (GPL-2.0-only) [installed]
kmod-crypto-ctr-6.12.63-r1 aarch64_cortex-a53 {feeds/base/kernel/linux} (GPL-2.0-only) [installed]
kmod-crypto-des-6.12.63-r1 aarch64_cortex-a53 {feeds/base/kernel/linux} (GPL-2.0-only) [installed]
kmod-crypto-gcm-6.12.63-r1 aarch64_cortex-a53 {feeds/base/kernel/linux} (GPL-2.0-only) [installed]
kmod-crypto-geniv-6.12.63-r1 aarch64_cortex-a53 {feeds/base/kernel/linux} (GPL-2.0-only) [installed]
kmod-crypto-gf128-6.12.63-r1 aarch64_cortex-a53 {feeds/base/kernel/linux} (GPL-2.0-only) [installed]
kmod-crypto-ghash-6.12.63-r1 aarch64_cortex-a53 {feeds/base/kernel/linux} (GPL-2.0-only) [installed]
kmod-crypto-hash-6.12.63-r1 aarch64_cortex-a53 {feeds/base/kernel/linux} (GPL-2.0-only) [installed]
kmod-crypto-hmac-6.12.63-r1 aarch64_cortex-a53 {feeds/base/kernel/linux} (GPL-2.0-only) [installed]
kmod-crypto-hw-safexcel-6.12.63-r1 aarch64_cortex-a53 {feeds/base/kernel/linux} (GPL-2.0-only) [installed]
kmod-crypto-manager-6.12.63-r1 aarch64_cortex-a53 {feeds/base/kernel/linux} (GPL-2.0-only) [installed]
kmod-crypto-md5-6.12.63-r1 aarch64_cortex-a53 {feeds/base/kernel/linux} (GPL-2.0-only) [installed]
kmod-crypto-null-6.12.63-r1 aarch64_cortex-a53 {feeds/base/kernel/linux} (GPL-2.0-only) [installed]
kmod-crypto-rng-6.12.63-r1 aarch64_cortex-a53 {feeds/base/kernel/linux} (GPL-2.0-only) [installed]
kmod-crypto-seqiv-6.12.63-r1 aarch64_cortex-a53 {feeds/base/kernel/linux} (GPL-2.0-only) [installed]
[... 48 lines cut ...]
[... cut ...]
kmod-slhc-6.12.63-r1 aarch64_cortex-a53 {feeds/base/kernel/linux} (GPL-2.0-only) [installed]
kmod-usb-common-6.12.63-r1 aarch64_cortex-a53 {feeds/base/kernel/linux} (GPL-2.0-only) [installed]
kmod-usb-core-6.12.63-r1 aarch64_cortex-a53 {feeds/base/kernel/linux} (GPL-2.0-only) [installed]
kmod-usb-xhci-hcd-6.12.63-r1 aarch64_cortex-a53 {feeds/base/kernel/linux} (GPL-2.0-only) [installed]
kmod-usb-xhci-mtk-6.12.63-r1 aarch64_cortex-a53 {feeds/base/kernel/linux} (GPL-2.0-only) [installed]
kmod-usb3-6.12.63-r1 aarch64_cortex-a53 {feeds/base/kernel/linux} (GPL-2.0-only) [installed]
```

dangowrt, 2026-01-03:
The PHY inside that copper SFP is not being recognized 😒 
Maybe try fiddling with `i2csfp` tool to find out which access method to use, so we can add a quirk (which won't be easy as the name "OEM", "SFP-10G-T" is already in use)

Edit: also `mdio-tools` can be useful to figure out if the (currently assumed) RollBall method shows you a PHY, but just not Aquantia or Marvell, but maybe RealTek, Motorcom or something else.

mhalano, 2026-01-03:
@dangowrt Do you have any link to a documentation where I can learn about how to do that and what I should be looking for?

danpawlik, 2026-01-04:
@mhalano tbh it looks like you have a module with rtl8261be not N - https://github.com/openwrt/openwrt/issues/20294
You should try patch https://github.com/openwrt/openwrt/pull/20450

mhalano, 2026-01-12:
@danpawlik @dangowrt I think my problem could be solved with this three patches: https://patchwork.kernel.org/project/netdevbpf/cover/aLSHmddAqiCISeK3@shell.armlinux.org.uk/. I think that because I started getting a new error:
```
[71120.468592] mtk_soc_eth 15100000.ethernet sfp-wan: validation with support 00,00000000,00000000,00000000 failed: -EINVAL
[71120.479582] sfp sfp1: sfp_add_phy failed: -EINVAL
[71533.651645] sfp sfp1: Unable to ascertain link mode
[71533.656534] mtk_soc_eth 15100000.ethernet sfp-wan: selection of interface failed, advertisement 00,00000000,00000000,00002040
[71840.009380] sfp sfp1: Unable to ascertain link mode
[71840.014337] mtk_soc_eth 15100000.ethernet sfp-wan: selection of interface failed, advertisement 00,00000000,00000000,00002000
```
What do you think?