openwrt/openwrt #17674: luci won't install on NIGHTLY

bittin, 2025-01-20:
### Describe the bug

root@rainbowdash:~# apk add luci
ERROR: unable to select packages:
  libuci20250117-2025.01.17~fb3c2343-r1:
    conflicts: libuci20130104-2024.11.26~10f7996e-r1[libuci=2025.01.17~fb3c2343-r1]
    satisfies: world[libuci20250117] libiwinfo20230701-2024.10.20~b94f066e-r1[libuci20250117]
               netifd-2024.12.17~ea01ed41-r1[libuci20250117] odhcpd-ipv6only-2024.05.08~a2988231-r1[libuci20250117]
               procd-ujail-2024.12.22~42d39376-r1[libuci20250117] ubox-2024.04.26~85f10530-r1[libuci20250117]
               uci-2025.01.17~fb3c2343-r1[libuci20250117] ucode-mod-uci-2024.12.06~209f041f-r1[libuci20250117]
  libuci20130104-2024.11.26~10f7996e-r1:
    conflicts: libuci20250117-2025.01.17~fb3c2343-r1[libuci=2024.11.26~10f7996e-r1]
    satisfies: rpcd-2024.12.02~cc9a471c-r1[libuci20130104]

### OpenWrt version

OpenWrt SNAPSHOT r28632-9454331b7f

### OpenWrt release

SNAPSHOT r28632-9454331b7f

### OpenWrt target/subtarget

mvebu/cortexa9

### Device

Linksys WRT1900ACS

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

bittin, 2025-01-21:
works now again thx @hnyman

bittin, 2025-01-23:
and now it does not work again

hnyman, 2025-01-23:
The error message in your email looks like APK database corruption(?) on your router?

```
( 7/25) Installing rpcd-mod-ucode (2024.12.02~cc9a471c-r1)
ERROR: rpcd-mod-ucode-2024.12.02~cc9a471c-r1: ADB integrity error
( 8/25) Installing ucode-mod-html (1)
ERROR: ucode-mod-html-1: ADB integrity error
( 9/25) Installing ucode-mod-math (2024.12.06~209f041f-r1)
ERROR: ucode-mod-math-2024.12.06~209f041f-r1: ADB integrity error
(10/25) Installing luci-base (25.021.30653~d889c27)
ERROR: luci-base-25.021.30653~d889c27: remote server returned error
(try 'apk update')
(11/25) Installing luci-app-package-manager (25.021.30653~d889c27)
ERROR: luci-app-package-manager-25.021.30653~d889c27: remote server
returned error (try 'apk update')
```

Did you follow the advice of running "APK update"?

bittin, 2025-01-23:
This is fixed now again

closed 2025-01-23 by bittin