openwrt/openwrt #17676: WAN Interface loses connection

pppranik, 2025-01-20:
### Describe the bug

My x86 router based on juniper mag2600 is periodically lost connection on wan
```Mon Jan 20 16:37:00 2025 daemon.notice netifd: Network device 'eth1' link is up
Mon Jan 20 16:37:00 2025 daemon.notice netifd: Interface 'wan' has link connectivity
Mon Jan 20 16:37:00 2025 daemon.notice netifd: Interface 'wan' is setting up now
Mon Jan 20 16:37:00 2025 kern.info kernel: [  231.641901] e1000e 0000:02:00.0 eth1: NIC Link is Up 1000 Mbps Full Duplex, Flow Control: None
Mon Jan 20 16:37:00 2025 daemon.notice netifd: wan (7083): udhcpc: started, v1.36.1
Mon Jan 20 16:37:00 2025 daemon.notice netifd: wan (7083): udhcpc: broadcasting discover
Mon Jan 20 16:37:03 2025 daemon.notice netifd: Network device 'eth1' link is down
Mon Jan 20 16:37:03 2025 daemon.notice netifd: Interface 'wan' has link connectivity loss
Mon Jan 20 16:37:03 2025 kern.info kernel: [  234.800993] e1000e 0000:02:00.0 eth1: NIC Link is Down
Mon Jan 20 16:37:03 2025 daemon.notice netifd: wan (7083): udhcpc: broadcasting discover
Mon Jan 20 16:37:03 2025 daemon.notice netifd: wan (7083): udhcpc: received SIGTERM
Mon Jan 20 16:37:03 2025 daemon.notice netifd: wan (7083): udhcpc: entering released state
Mon Jan 20 16:37:03 2025 daemon.notice netifd: wan (7083): Command failed: ubus call network.interface notify_proto { "action": 0, "link-up": false, "keep": false, "interface": "wan" } (Permission denied)
Mon Jan 20 16:37:03 2025 daemon.notice netifd: Interface 'wan' is now down```

Reconnecting the wan cable solves the problem for a while, but not for long

### OpenWrt version

r24106-10cc5fcd00

### OpenWrt release

23.05.5

### OpenWrt target/subtarget

x86/generic

### Device

Juniper Networks MAG2600

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

aimacintyre, 2025-01-21:
Is EEE enabled on the WAN interface?  If so try disabling it (e.g. with `ethtool`).

BUG-reporter, 2025-02-04:
It looks the same as mine report: https://github.com/openwrt/openwrt/issues/17711 but it has been invalidated wrongly.
I would also recommend to try my workaround: https://github.com/openwrt/openwrt/issues/17711#issuecomment-2631992772