openwrt/openwrt #17677: 24.10.0-rc5: Asus RT-N56U A1 fails to boot

goosnarrggh, 2025-01-20:
### Describe the bug

Router fails to boot into any of the 24.10.0-rc* images that have been released to date.

### OpenWrt version

24.10.0-rc5

### OpenWrt release

unable to ssh into system to determine this information

### OpenWrt target/subtarget

rampis/rt3883

### Device

Asus RT-N56U A1

### Image kind

Official downloaded image

### Steps to reproduce

Any one of the following will trigger the bug:

- Upload and install image as an upgrade from 23.05.5 via LuCI
- Upload and install image as an upgrade from 23.05.5 via scp, ssh, and sysupgrade
- Install image using Asus' firmware restoration utility.

### Actual behaviour

Router appears to fail to boot. No network interfaces are active.

### Expected behaviour

Router should successfully boot, with clear signs of life on its various network interfaces.

### Additional info

I have attempted to install the image from an existing 23.05.5 system using LuCI, and again using Asus' firmware restoration tool. In both cases, the router appears to reboot, but does not come back up. The LEDs on all RJ45 jacks and the wifi annunciation LEDs go out. There is no responsiveness on any network interfaces. It remains in this state indefinitely. Power cycling does not improve the situation.

Recovery is possible by using Asus' firmware restoration utility to revert back to 23.05.5.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

goosnarrggh, 2025-01-20:
The Target/Subtarget field was filled in using the equivalent command from a running 23.05.5 system.
I think the information I put in the Version field should have gone into the Release field instead.
I do not know how to extract the correct string to input in the Version field because I cannot boot into the system to obtain it.

hauke, 2025-01-20:
Maybe the image or kernel is getting too big. Could you connect a UART and post the error message please.

udanfacy22, 2025-01-21:
On the unofficial /r/openwrt subreddit, where they [originally posted](https://www.reddit.com/r/openwrt/comments/1i5eqwl/the_last_2410_release_candidate_to_be_tagged_on/m8372gx/) about this, they said they physically fried one of two identical devices while attempting to get debugging information over serial. Given that, it's unclear if they're willing to risk it again with the surviving device.

Also, in this bug report currently, they say *all* 24.10 release candidates so far have this issue, but on Reddit they only mention rc2 to rc5. It's unclear if they've tested rc1.

goosnarrggh, 2025-01-22:
The history I included in this bug report is accurate; the forum post was an overstatement. I have not attempted to boot rc1.

I have updated the original submission to include the correct string for the target/subtarget and what I have been told is the accurate version string for the 24.10.0-rc5 builds.

@hauke I attempted to gather boot logs from one of two units I have at my disposal, but did not find the right pull-up resistor to establish correct serial port TX voltage levels. During my attempts to do so, a multimeter probe slipped and damaged at least one IC. So far, I've been reluctant to retry the investigation with the remaining unit since it's currently in service. I will look for an opportunity to swap it out so that I can experiment with it more freely.

Lucky1openwrt, 2025-01-27:
FYI DIR-645 also won't boot same subtarget v24 or master

error from serial 
[    0.000000] Couldn't register fixed clock 3
[    0.000000] Couldn't register fixed clocks

v23 
[    0.000000] CPU Clock: 500MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041786 ns
[    0.000002] sched_clock: 32 bits at 250MHz, resolution 4ns, wraps every 8589934590ns

https://forum.openwrt.org/t/openwrt-24-10-0-rc5-fifth-release-candidate/220854/117

goosnarrggh, 2025-01-27:
I note many of the ralink MIPS family of SoCs went through a clock driver overhaul: https://lore.kernel.org/lkml/20230619040941.1340372-5-sergio.paracuellos@gmail.com/T/#m14ab194413c3ab1dcfc62600f406cbd808349091

The contributor stated that they had tested their changes in real silicon using boards based on RT5350, MT7620A, and MT7628AM; however, for RT2880 and RT3883 targets they claimed, at the time, to have stopped at the point of proving that they compiled cleanly.

There was also talk of further dts changes being needed for most of the impacted SoCs platforms, but such changes were not included in that patch set. The author noted that they were planning on coordinating with the OpenWRT team to phase in those changes, but I don't know if such collaboration ever actually occurred.

Certainly there have been no recent commits to the upstream arch/mips/boot/dts/ralink/rt3883.dtsi file within the time-frame in question.

goosnarrggh, 2025-01-27:
Upstream, **rt3883_clks_base** has two entries, "cpu" and "bus".

In OpenWRT's kernel 6.6 patches, **rt3883_clks_base** grows to three entries, "xtal", "cpu", and "bus".

Based my reading of **mtmips_register_fixed_clocks**, with a base clocks array of size 3, if we are seeing the error message "Couldn't register fixed clock 3", then the problem is being encountered when we try to initialize the first entry in the fixed clock array.

Upstream, **rt3883_fixed_clocks** goes by a different name, **rt305x_fixed_clocks**, with a single entry "xtal".

In OpenWRT's kernel 6.6 patches, rt3883_fixed_clocks has two entries, "xtal" and "periph".

So, it looks like at least in @Lucky1openwrt 's case, something is going wrong when trying to initialize the fixed clock named "xtal". Is it valid to register a base clock **and** a fixed clock **both** going by the same name "xtal"?

namiltd, 2025-02-01:
Similat issue: https://github.com/openwrt/openwrt/issues/17810

namiltd, 2025-02-01:
Please upload the [999-rt3883_clock_init.patch](https://github.com/user-attachments/files/18628704/999-rt3883_clock_init.patch) file to target/linux/generic/hack-6.6 generate images and try to run.

Lucky1openwrt, 2025-02-01:
I Tested this on the DIR-645 this is on the master branch
mixed results it boot sort of 
[    0.000000] CPU Clock: 40MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 95563022313 ns
[    0.000032] sched_clock: 32 bits at 20MHz, resolution 50ns, wraps every 107374182375ns
[    0.198643] Calibrating delay loop... 19.94 BogoMIPS (lpj=99712)

but serial console drops out after 
[    5.948626] Serial: 8250/16550 driver, 2 ports, IRQ sharing disabled
[    6.139795] printk: console [ttyS0] disabled
▒▒x▒▒▒▒▒▒xx▒▒▒▒x▒x▒▒▒▒xx▒▒▒▒▒▒▒▒▒x▒▒▒▒▒▒▒▒▒▒▒▒▒▒x▒▒▒x▒▒▒▒▒x▒xx▒x▒▒▒▒▒▒▒▒x▒▒▒▒x▒▒x▒▒▒▒▒▒xx▒▒▒▒x▒x▒▒▒▒xx▒▒

nariox, 2025-02-01:
Attaching the entire bootlog for TEW-692GR (also RT3883)

[bootlog_afterpatch.txt](https://github.com/user-attachments/files/18630388/bootlog_afterpatch.txt)

namiltd, 2025-02-01:
CPU clock is 40MHz and should be 500MHz

namiltd, 2025-02-01:
Try new version: [999-rt3883_clock_init.patch](https://github.com/user-attachments/files/18630596/999-rt3883_clock_init.patch)

Lucky1openwrt, 2025-02-02:
HI Thanks 
Early results are good with V24 and master are good but serial port speed is bad

unit boots up and seem to work fine via web interface

V23
[    0.000000] NR_IRQS: 256
[    0.000000] CPU Clock: 500MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041786 ns
[    0.000002] sched_clock: 32 bits at 250MHz, resolution 4ns, wraps every 8589934590ns
[    0.015588] Calibrating delay loop... 249.44 BogoMIPS (lpj=1247232)
[    0.097834] pid_max: default: 32768 minimum: 301

Main Branch
[    0.000000] CPU Clock: 500MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041786 ns
[    0.000002] sched_clock: 32 bits at 250MHz, resolution 4ns, wraps every 8589934590ns
[    0.015546] Calibrating delay loop... 249.44 BogoMIPS (lpj=1247232)
[    0.097811] pid_max: default: 32768 minimum: 301
[    0.116229] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)

but something wrong with clock and the serial port speed

V23
[    0.548720] printk: console [ttyS0] disabled
[    0.557282] 10000c00.uartlite: ttyS0 at MMIO 0x10000c00 (irq = 20, base_baud = 2500000) is a Palmchip BK-3103
[    0.576955] printk: console [ttyS0] enabled
[    0.576955] printk: console [ttyS0] enabled
[    0.593516] printk: bootconsole [early0] disabled
[    0.593516] printk: bootconsole [early0] disabled
[    0.640822] spi spi0.0: force spi mode3

Main Branch
[    0.546050] printk: console [ttyS0] disabled
[    0.555377] 10000c00.uartlite: ttyS0 at MMIO 0x10000c00 (irq = 20, base_baud = 10375000) is a Palmchip BK-3103
▒▒x▒▒▒▒▒▒xx▒▒▒x▒x▒x▒xx▒▒▒▒▒▒▒▒▒x▒▒▒▒▒▒▒▒▒▒▒▒▒▒x▒▒▒x▒▒▒▒▒x▒xx▒x▒▒▒▒▒▒▒▒x▒▒▒▒x▒▒x▒▒▒

DragonBluep, 2025-02-02:
The fix: https://github.com/openwrt/openwrt/pull/17830

namiltd, 2025-02-02:
@Lucky1openwrt  Please test the final patch https://github.com/openwrt/openwrt/pull/17830

closed 2025-02-02 by openwrt-bot

nariox, 2025-02-03:
Works beautifully. Thank you all

goosnarrggh, 2025-02-05:
Confirmed, this fix works on the RT-N56U from the original report.