openwrt/openwrt #17681: Sysupgrade from snapshot to 24.10.0-rc5 not working

tskarthikkumar, 2025-01-21:
### Describe the bug

I'm trying to upgrade Linksys MX5300 from firmware (OpenWrt SNAPSHOT r28602-c9ebd4fb30-> fresh install) to openwrt-24.10.0-rc5-1a36e83bede4.

When I run "sysupgrade -F openwrt-24.10.0-rc5-1a36e83bede4-qualcommax-ipq807x-linksys_mx5300-squashfs-sysupgrade.bin", getting below error.

root@OpenWrt:/tmp# sysupgrade -F openwrt-24.10.0-rc5-1a36e83bede4-qualcommax-ipq807x-linksys_mx5300-squashfs-sysupgrade.bin
Tue Jan 21 01:22:24 GMT 2025 upgrade: Saving config files...
Tue Jan 21 01:22:25 GMT 2025 upgrade: Commencing upgrade. Closing all shell sessions.
Command failed: Connection failed

[image]

with verbose:

[image]

I tried different images, from openwrt official site/firmware selector image/firmware selector custom build/luci-attended-sysupgrade. all resulting in same issue.

Device goes to softbrick.  Have to hard reset couple of times to get it working again.

### OpenWrt version

r28602-c9ebd4fb30

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Linksys MX5300

### Image kind

Official downloaded image

### Steps to reproduce

Install snapshot version in Linksys MX5300
Download sysupgrade image from [firmware selector](https://firmware-selector.openwrt.org/?version=24.10.0-rc5&target=qualcommax%2Fipq807x&id=linksys_mx5300) 
run sysupgrade -v -F openwrt-24.10.0-rc5-sysupgrade.bin

### Actual behaviour

Device goes to softbrick.
After rebooting, it still stays in snapshot.

### Expected behaviour

Upgrade the router to openwrt-24.10.0-rc5 version

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

aimacintyre, 2025-01-21:
Try installing without preserving configuration (`sysupgrade -n`)?

tskarthikkumar, 2025-01-21:
> Try installing without preserving configuration (`sysupgrade -n`)?

still the same with -n option as well.

root@OpenWrt-LS-2:/tmp# sysupgrade -F -n -v openwrt-24.10.0-rc5-qualcommax-ipq807x-linksys_mx5300-squashfs-sysupgrade.bin 
Mon Jan 20 23:17:41 CST 2025 upgrade: Commencing upgrade. Closing all shell sessions.
Command failed: Connection failed

robimarko, 2025-01-21:
Why are you forcing the upgrade with `-F`?

@testuser7 I think you have one of these

tskarthikkumar, 2025-01-22:
I did with and without -F but results are same. After I'm able to connect back to the router, I installed luci-app-advanced-reboot and rebooted to linksys partition (1) since the router has 2 partitions, still the same result, led is flashing solid blue, router is not reachable via ip. I then switched partition by turning router on and off three times to boot to openwrt again.

Then flashed the kernel and alt_kernel with OEM image file using mtd command, which allowed me to enter the linksys firmware.  Then I used openwrt 24.10.0-rc5 factory image from linksys firmware upgrade section, now I'm finally upgraded to 24.10.0-rc5. Additionally, I'm now able to dualboot between linksys oem and openwrt.

This is not an ideal way to upgrade but my issue is solved and this can be closed, unless if someone is looking into it and need any info from me, I can provide. Hope this will be helpful for someone who is facing this issue.

testuser7, 2025-01-25:
@tskarthikkumar Did you have OpenWrt installed on both partitions before upgrading? Installation via sysupgrade does not work if there is OEM software on the second partition.

@robimarko This commit should eliminate issues like these in the future: https://github.com/openwrt/openwrt/pull/14720/commits/bedaf55af3bcbc6338165274c73c3ae7cf2130e5

tskarthikkumar, 2025-01-26:
> @tskarthikkumar Did you have OpenWrt installed on both partitions before upgrading? Installation via sysupgrade does not work if there is OEM software on the second partition.
>
> @robimarko This [...]

@testuser7  i didn't have openwrt in both partitions. I think first partition was OEM, im not sure though which partition it was on. 

After upgrade as per my previous note, it currently shows openwrt in partition 1 and oem in partition 2.

counterpoke, 2025-01-27:
@tskarthikkumar I was in the same state you were in with the failed sysupgrade with the same router. I then ran the steps listed here to install openwrt in both partitions (https://forum.openwrt.org/t/openwrt-support-for-linksys-mx4200/86477/1644 ). After doing so, I was able to upgrade to rc6 without any issues.

tskarthikkumar, 2025-01-27:
@counterpoke thanks for the link, I saw that already, but having OEM in 1 partition is nice to have, because why not?

If the commit which @testuser7 user mentioned fixes the sysupgrade issue on openwrt partition when having dual firmwares, then it will be a great fix!

@

Ansuel, 2025-02-06:
@testuser7 any clue why the OEM partition needs to be wiped? Can we have a log of what the bootloader do after the sysupgrade with the OEM partition?

I assume the bootloader check both partition... Doesn't find some kind of metadata in the first one, but found some in the second partition and boots the second partition?

Could also be that we are not resetting some kind of counter ?

testuser7, 2025-02-06:
@Ansuel the problem is resolved by this commit: https://github.com/openwrt/openwrt/commit/df1f6e1e186a99b2180abac5da87f071b7f6b3fa
I need to prepare backports to the 24.10 branch.

Ansuel, 2025-02-06:
@testuser7 more extensive info on this? what is the OEM ubi volume? Is it recreated when original firmware is restored?

testuser7, 2025-02-06:
> [@testuser7](https://github.com/testuser7) more extensive info on this?

I don't have any logs from a failed installation via sysupgrade at this time.
The update works if we remove the OEM volume before installation.

> what is the OEM ubi volume?

How partitions are used in OEM software can be found in this post: https://forum.openwrt.org/t/adding-openwrt-support-for-linksys-mx5300/92667/25
There is only one large volume on UBI partition `rootfs`/`alt_rootfs` named:
- `ubifs` (MX5300/MX8500)
- `squashfs` (All the rest)
The name refers to the FS used.

> Is it recreated when original firmware is restored?

OEM firmware restore is done through mtd (same as OEM firmware update) so volumes on UBI are recreated as they are restored.

testuser7, 2025-02-24:
@Ansuel Logs from an unsuccessful attempt to perform a sysupgrade on the second partition with OEM firmware:
```
Sun Feb 23 11:49:42 UTC 2025 upgrade: Performing system upgrade...
verifying sysupgrade tar file integrity
Unlocking alt_kernel ...

Writing from <stdin> to alt_kernel ...     
[  261.629912] ubi1: attaching mtd24
[  262.245293] ubi1: scanning is finished
[  262.270951] ubi1: volume 0 ("ubifs") re-sized from 603 to 1068 LEBs
[  262.271352] ubi1: attached mtd24 (name "alt_rootfs", size 144 MiB)
[  262.276026] ubi1: PEB size: 131072 bytes (128 KiB), LEB size: 126976 bytes
[  262.282301] ubi1: min./max. I/O unit sizes: 2048/2048, sub-page size 2048
[  262.289142] ubi1: VID header offset: 2048 (aligned 2048), data offset: 4096
[  262.295991] ubi1: good PEBs: 1152, bad PEBs: 0, corrupted PEBs: 0
[  262.302764] ubi1: user volume: 1, internal volumes: 1, max. volumes count: 128
[  262.309019] ubi1: max/mean erase counter: 1/0, WL threshold: 4096, image sequence number: 1404302797
[  262.316135] ubi1: available PEBs: 0, total reserved PEBs: 1152, PEBs reserved for bad PEB handling: 80
[  262.325435] ubi1: background thread "ubi_bgt1d" started, PID 4792
UBI device number 1, total 1152 LEBs (146276352 bytes, 139.5 MiB), available 0 LEBs (0 bytes), LEB size 126976 bytes (124.0 KiB)
ubimkvol: error!: UBI device does not have free logical eraseblocks
cannot create rootfs volume
sysupgrade failed
umount: can't unmount /dev: Resource busy
umount: can't unmount /tmp: Resource busy
[  262.862464] remoteproc remoteproc0: stopped remote processor cd00000.q6v5_wcss
[  263.397103] reboot: Restarting system
```

After such an unsuccessful update, the beginning of the kernel partition is empty:
```
NAND read: device 0 offset 0xa680000, size 0x600000
 6291456 bytes read: OK
Wrong Image Format for bootm command
ERROR: can't get kernel image!
```

And here are the logs from a successful update after deleting the `ubifs` volume before the update:
```
Sun Feb 23 12:12:35 UTC 2025 upgrade: Performing system upgrade...
[ 1630.915946] ubi1: attaching mtd24
[ 1631.741233] ubi1: scanning is finished
[ 1631.749337] ubi1: attached mtd24 (name "alt_rootfs", size 144 MiB)
[ 1631.749381] ubi1: PEB size: 131072 bytes (128 KiB), LEB size: 126976 bytes
[ 1631.754420] ubi1: min./max. I/O unit sizes: 2048/2048, sub-page size 2048
[ 1631.761301] ubi1: VID header offset: 2048 (aligned 2048), data offset: 4096
[ 1631.768142] ubi1: good PEBs: 1152, bad PEBs: 0, corrupted PEBs: 0
[ 1631.774904] ubi1: user volume: 1, internal volumes: 1, max. volumes count: 128
[ 1631.781164] ubi1: max/mean erase counter: 1/0, WL threshold: 4096, image sequence number: 1404302797
[ 1631.788282] ubi1: available PEBs: 0, total reserved PEBs: 1152, PEBs reserved for bad PEB handling: 80
[ 1631.797578] ubi1: background thread "ubi_bgt1d" started, PID 4786
UBI device number 1, total 1152 LEBs (146276352 bytes, 139.5 MiB), available 0 LEBs (0 bytes), LEB size 126976 bytes (124.0 KiB)
verifying sysupgrade tar file integrity
Unlocking alt_kernel ...

Writing from <stdin> to alt_kernel ...     
[ 1632.911343] block ubiblock1_0: created from ubi1:0(rootfs)
Volume ID 0, size 77 LEBs (9777152 bytes, 9.3 MiB), LEB size 126976 bytes (124.0 KiB), dynamic, name "rootfs", alignment 1
Set volume size to 125833216
Volume ID 1, size 991 LEBs (125833216 bytes, 120.0 MiB), LEB size 126976 bytes (124.0 KiB), dynamic, name "rootfs_data", alignment 1
Unlocking alt_kernel ...

Writing from <stdin> to alt_kernel ...     
[ 1637.271564] UBIFS (ubi1:1): default file-system created
[ 1637.272213] UBIFS (ubi1:1): Mounting in unauthenticated mode
[ 1637.275696] UBIFS (ubi1:1): background thread "ubifs_bgt1_1" started, PID 4913
[ 1637.335195] UBIFS (ubi1:1): UBIFS: mounted UBI device 1, volume 1, name "rootfs_data"
[ 1637.335241] UBIFS (ubi1:1): LEB size: 126976 bytes (124 KiB), min./max. I/O unit sizes: 2048 bytes/2048 bytes
[ 1637.342053] UBIFS (ubi1:1): FS size: 124436480 bytes (118 MiB, 980 LEBs), max 991 LEBs, journal size 6221824 bytes (5 MiB, 49 LEBs)
[ 1637.351932] UBIFS (ubi1:1): reserved for root: 4952683 bytes (4836 KiB)
[ 1637.363546] UBIFS (ubi1:1): media format: w5/r0 (latest is w5/r0), UUID 581DA9BB-42DC-4412-88B9-8E29A951F3AC, small LPT model
[ 1637.383408] UBIFS (ubi1:1): un-mount UBI device 1
[ 1637.383459] UBIFS (ubi1:1): background thread "ubifs_bgt1_1" stops
configuration saved
sysupgrade successful
umount: can't unmount /dev: Resource busy
umount: can't unmount /tmp: Resource busy
[ 1637.891675] remoteproc remoteproc0: stopped remote processor cd00000.q6v5_wcss
[ 1638.386870] reboot: Restarting system
```

xuefer, 2025-07-26:
this happens too for phicomm k2p trunk (and maybe since long ago, because it never work for me)```
sysupgrade -n openwrt-ramips-mt7621-phicomm_k2p-squashfs-sysupgrade.bin
Sat Jul 26 12:28:16 UTC 2025 upgrade: Commencing upgrade. Closing all shell sessions.
Command failed: Connection failed
```

testuser7, 2025-10-16:
The issue is already resolved in the associated PR, can be closed.