openwrt/openwrt #17683: MacOSX15.2.sdk/usr/include/hfs/hfs_format.h:808:2: error: unknown type name 'uuid_string_t'

mvpbang, 2025-01-21:
### Describe the bug

time: tools/e2fsprogs/compile#0.40#0.35#0.90
[5/10] Building CXX object Source/CMakeFiles/CPackLib.dir/CPack/cmCPackDragNDropGenerator.cxx.o
FAILED: Source/CMakeFiles/CPackLib.dir/CPack/cmCPackDragNDropGenerator.cxx.o 
/Volumes/OpenWrt/openwrt-24.10.0-rc5/staging_dir/host/bin/g++ -DLIBARCHIVE_STATIC -I/Volumes/OpenWrt/openwrt-24.10.0-rc5/build_dir/host/cmake-3.30.5/Source/CPack -I/Volumes/OpenWrt/openwrt-24.10.0-rc5/build_dir/host/cmake-3.30.5/Source -I/Volumes/OpenWrt/openwrt-24.10.0-rc5/build_dir/host/cmake-3.30.5/Source/LexerParser -isystem /Volumes/OpenWrt/openwrt-24.10.0-rc5/build_dir/host/cmake-3.30.5/Utilities/std -isystem /Volumes/OpenWrt/openwrt-24.10.0-rc5/build_dir/host/cmake-3.30.5/Utilities -O2 -I/Volumes/OpenWrt/openwrt-24.10.0-rc5/staging_dir/host/include -O3 -DNDEBUG -std=c++17 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX15.2.sdk -MD -MT Source/CMakeFiles/CPackLib.dir/CPack/cmCPackDragNDropGenerator.cxx.o -MF Source/CMakeFiles/CPackLib.dir/CPack/cmCPackDragNDropGenerator.cxx.o.d -o Source/CMakeFiles/CPackLib.dir/CPack/cmCPackDragNDropGenerator.cxx.o -c /Volumes/OpenWrt/openwrt-24.10.0-rc5/build_dir/host/cmake-3.30.5/Source/CPack/cmCPackDragNDropGenerator.cxx
In file included from /Volumes/OpenWrt/openwrt-24.10.0-rc5/build_dir/host/cmake-3.30.5/Source/CPack/cmCPackDragNDropGenerator.cxx:35:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX15.2.sdk/System/Library/Frameworks/CoreServices.framework/Headers/CoreServices.h:23:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX15.2.sdk/System/Library/Frameworks/CoreServices.framework/Frameworks/AE.framework/Headers/AE.h:20:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX15.2.sdk/System/Library/Frameworks/CoreServices.framework/Frameworks/CarbonCore.framework/Headers/CarbonCore.h:208:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX15.2.sdk/System/Library/Frameworks/CoreServices.framework/Frameworks/CarbonCore.framework/Headers/HFSVolumes.h:25:
/Library/Developer/CommandLineTools/SDKs/MacOSX15.2.sdk/usr/include/hfs/hfs_format.h:808:2: error: unknown type name 'uuid_string_t'; did you mean 'io_string_t'?
  808 |         uuid_string_t   ext_jnl_uuid;
      |         ^
/Library/Developer/CommandLineTools/SDKs/MacOSX15.2.sdk/usr/include/device/device_types.h:89:33: note: 'io_string_t' declared here
   89 | typedef char                    io_string_t[512];
      |                                 ^
In file included from /Volumes/OpenWrt/openwrt-24.10.0-rc5/build_dir/host/cmake-3.30.5/Source/CPack/cmCPackDragNDropGenerator.cxx:35:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX15.2.sdk/System/Library/Frameworks/CoreServices.framework/Headers/CoreServices.h:23:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX15.2.sdk/System/Library/Frameworks/CoreServices.framework/Frameworks/AE.framework/Headers/AE.h:20:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX15.2.sdk/System/Library/Frameworks/CoreServices.framework/Frameworks/CarbonCore.framework/Headers/CarbonCore.h:208:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX15.2.sdk/System/Library/Frameworks/CoreServices.framework/Frameworks/CarbonCore.framework/Headers/HFSVolumes.h:25:
/Library/Developer/CommandLineTools/SDKs/MacOSX15.2.sdk/usr/include/hfs/hfs_format.h:810:20: error: unknown type name 'uuid_string_t'; did you mean 'io_string_t'?
  810 |         char            reserved[JIB_RESERVED_SIZE];
      |                                  ^
/Library/Developer/CommandLineTools/SDKs/MacOSX15.2.sdk/usr/include/hfs/hfs_format.h:801:61: note: expanded from macro 'JIB_RESERVED_SIZE'
  801 | #define JIB_RESERVED_SIZE  ((32*sizeof(u_int32_t)) - sizeof(uuid_string_t) - 48)
      |                                                             ^
/Library/Developer/CommandLineTools/SDKs/MacOSX15.2.sdk/usr/include/device/device_types.h:89:33: note: 'io_string_t' declared here
   89 | typedef char                    io_string_t[512];
      |                                 ^
In file included from /Volumes/OpenWrt/openwrt-24.10.0-rc5/build_dir/host/cmake-3.30.5/Source/CPack/cmCPackDragNDropGenerator.cxx:35:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX15.2.sdk/System/Library/Frameworks/CoreServices.framework/Headers/CoreServices.h:23:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX15.2.sdk/System/Library/Frameworks/CoreServices.framework/Frameworks/AE.framework/Headers/AE.h:20:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX15.2.sdk/System/Library/Frameworks/CoreServices.framework/Frameworks/CarbonCore.framework/Headers/CarbonCore.h:208:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX15.2.sdk/System/Library/Frameworks/CoreServices.framework/Frameworks/CarbonCore.framework/Headers/HFSVolumes.h:25:
/Library/Developer/CommandLineTools/SDKs/MacOSX15.2.sdk/usr/include/hfs/hfs_format.h:810:20: error: array is too large (18446744073709551184 elements)
  810 |         char            reserved[JIB_RESERVED_SIZE];
      |                                  ^~~~~~~~~~~~~~~~~
/Library/Developer/CommandLineTools/SDKs/MacOSX15.2.sdk/usr/include/hfs/hfs_format.h:801:28: note: expanded from macro 'JIB_RESERVED_SIZE'
  801 | #define JIB_RESERVED_SIZE  ((32*sizeof(u_int32_t)) - sizeof(uuid_string_t) - 48)
      |                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
3 errors generated.
[8/10] Linking CXX executable bin/ctest
ninja: build stopped: subcommand failed.
make[3]: *** [Makefile:59: /Volumes/OpenWrt/openwrt-24.10.0-rc5/build_dir/host/cmake-3.30.5/.built] Error 1
make[3]: Leaving directory '/Volumes/OpenWrt/openwrt-24.10.0-rc5/tools/cmake'
time: tools/cmake/compile#4.70#1.74#4.78
    ERROR: tools/cmake failed to build.
make[2]: *** [tools/Makefile:231: tools/cmake/compile] Error 1
make[2]: *** Waiting for unfinished jobs....
make[3]: Leaving directory '/Volumes/OpenWrt/openwrt-24.10.0-rc5/tools/mkimage'
time: tools/mkimage/compile#0.48#0.64#2.67
make[2]: Leaving directory '/Volumes/OpenWrt/openwrt-24.10.0-rc5'
make[1]: *** [tools/Makefile:227: /Volumes/OpenWrt/openwrt-24.10.0-rc5/staging_dir/host/stamp/.tools_compile_nyyynyyynnnyyyyynyyyyyyynyyynynyynnynnyyyyyynyyyyyyyyyyynynnyyyyynyyy] Error 2
make[1]: Leaving directory '/Volumes/OpenWrt/openwrt-24.10.0-rc5'
make: *** [/Volumes/OpenWrt/openwrt-24.10.0-rc5/include/toplevel.mk:233: world] Error 2
 openwrt-24.10.0-rc5$ 

### OpenWrt version

v24.10.0-rc5

### OpenWrt release

v24.10.0-rc5

### OpenWrt target/subtarget

v24.10.0-rc5

### Device

 openwrt-24.10.0-rc5$ sw_vers  ProductName:		macOS ProductVersion:		15.2 BuildVersion:		24C101  openwrt-24.10.0-rc5$ 

### Image kind

Self-built image

### Steps to reproduce

make -j$(nproc)  V=s

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

mvpbang, 2025-01-22:
> Please provide diffconfig, device tags, OSX and XCODE versions.

# .config
```
➜  openwrt-24.10.0-rc5 cat .config |egrep -v '^#|^$'
CONFIG_MODULES=y
CONFIG_HAVE_DOT_CONFIG=y
CONFIG_HOST_OS_MACOS=y
CONFIG_TARGET_x86=y
CONFIG_TARGET_x86_64=y
CONFIG_TARGET_x86_64_DEVICE_generic=y
CONFIG_HAS_SUBTARGETS=y
CONFIG_HAS_DEVICES=y
CONFIG_TARGET_BOARD="x86"
CONFIG_TARGET_SUBTARGET="64"
CONFIG_TARGET_PROFILE="DEVICE_generic"
CONFIG_TARGET_ARCH_PACKAGES="x86_64"
CONFIG_DEFAULT_TARGET_OPTIMIZATION="-Os -pipe"
CONFIG_CPU_TYPE=" "
CONFIG_LINUX_6_6=y
CONFIG_DEFAULT_base-files=y
CONFIG_DEFAULT_ca-bundle=y
CONFIG_DEFAULT_dnsmasq=y
CONFIG_DEFAULT_dropbear=y
CONFIG_DEFAULT_e2fsprogs=y
CONFIG_DEFAULT_firewall4=y
CONFIG_DEFAULT_fstools=y
CONFIG_DEFAULT_grub2-bios-setup=y
CONFIG_DEFAULT_kmod-amazon-ena=y
CONFIG_DEFAULT_kmod-amd-xgbe=y
CONFIG_DEFAULT_kmod-bnx2=y
CONFIG_DEFAULT_kmod-button-hotplug=y
CONFIG_DEFAULT_kmod-dwmac-intel=y
CONFIG_DEFAULT_kmod-e1000=y
CONFIG_DEFAULT_kmod-e1000e=y
CONFIG_DEFAULT_kmod-forcedeth=y
CONFIG_DEFAULT_kmod-fs-vfat=y
CONFIG_DEFAULT_kmod-igb=y
CONFIG_DEFAULT_kmod-igc=y
CONFIG_DEFAULT_kmod-ixgbe=y
CONFIG_DEFAULT_kmod-nft-offload=y
CONFIG_DEFAULT_kmod-r8169=y
CONFIG_DEFAULT_kmod-tg3=y
CONFIG_DEFAULT_libc=y
CONFIG_DEFAULT_libgcc=y
CONFIG_DEFAULT_libustream-mbedtls=y
CONFIG_DEFAULT_logd=y
CONFIG_DEFAULT_mkf2fs=y
CONFIG_DEFAULT_mtd=y
CONFIG_DEFAULT_netifd=y
CONFIG_DEFAULT_nftables=y
CONFIG_DEFAULT_odhcp6c=y
CONFIG_DEFAULT_odhcpd-ipv6only=y
CONFIG_DEFAULT_partx-utils=y
CONFIG_DEFAULT_ppp=y
CONFIG_DEFAULT_ppp-mod-pppoe=y
CONFIG_DEFAULT_procd-ujail=y
CONFIG_DEFAULT_uci=y
CONFIG_DEFAULT_uclient-fetch=y
CONFIG_DEFAULT_urandom-seed=y
CONFIG_DEFAULT_urngd=y
CONFIG_HAS_FPU=y
CONFIG_AUDIO_SUPPORT=y
CONFIG_GPIO_SUPPORT=y
CONFIG_PCI_SUPPORT=y
CONFIG_PCIE_SUPPORT=y
CONFIG_PCMCIA_SUPPORT=y
CONFIG_USB_SUPPORT=y
CONFIG_RTC_SUPPORT=y
CONFIG_USES_SQUASHFS=y
[... 684 lines cut ...]
CONFIG_PACKAGE_mkf2fs=y
CONFIG_PACKAGE_jshn=y
CONFIG_PACKAGE_libjson-script=y
CONFIG_STRACE_NONE=y
CONFIG_PACKAGE_ucode-mod-html=y
```

# device tags
x86/x86_64

# OSX
➜  openwrt-24.10.0-rc5 sw_vers 
ProductName:		macOS
ProductVersion:		15.2
BuildVersion:		24C101

# XCODE versions
not install xcode ide, installed "CommandLineTools"

➜  openwrt-24.10.0-rc5 pkgutil --pkg-info=com.apple.pkg.CLTools_Executables
package-id: com.apple.pkg.CLTools_Executables
version: 16.2.0.0.1.1733547573
volume: /
location: /
install-time: 1737255131

brada4, 2025-01-22:
Please compact your assorted outputs and provide precise details requested in issue template.
https://github.com/openwrt/openwrt/blob/main/.github/ISSUE_TEMPLATE/bug-report.yml

brada4, 2025-01-22:
It is beta version of xcode , repeat issue with something claimed stable.

mvpbang, 2025-01-22:
> It is beta version of xcode , repeat issue with something claimed stable.

ok, will try. release: v23.05.5 compile...