openwrt/openwrt #17688: Unable to traceroute after upgrading device to 22.03.7

bmrobson101, 2025-01-21:
### Describe the bug

I have a group of devices that were previously on OpenWRT version 21.02.0 that recently got upgraded to 22.03.7. During this upgrade, I noticed that the traceroute command (a command that used to work perfectly fine) is giving me a network unreachable error (!N). Has anyone else experienced this and/or have ideas for a potential solution? Are there any packages that got automatically installed between versions that may be messing with traceroute?

### OpenWrt version

r20341-591b7e93d3

### OpenWrt release

22.03.7

### OpenWrt target/subtarget

ath79/generic

### Device

GL-MT300N-V2 and TP-WR710N-V1

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-01-22:
Any custom firewal rules/includes used?
Bankup config,reset config and check whether traceroute works.

bmrobson101, 2025-01-23:
There are no firewall rules that I can see. I'll try resetting the config though and see what happens

brada4, 2025-01-23:
Check /etc/config/firewall in the backup file. There should be as a minimum lan and wan with NAT in between.

bmrobson101, 2025-01-23:
How do I actually reset the config? I tried looking online but couldn't find anything. Also I don't think I have a firewall file in my /etc/config directory

brada4, 2025-01-23:
It is in LuCI System/BackupFlashFirmware, router will return to 192.168.1.1 and no wifi.