openwrt/openwrt #17689: Banana Pi R4: switch speed degredation

wberrier, 2025-01-22:
### Describe the bug

I have a computer that supports "wake on lan", and it's gigabit ethernet device goes into 10mbit mode when "off" waiting for a wake on lan packet.

The issue is that this slows down another gigabit connection on the same switch (see output of various attachments).

### OpenWrt version

r28304-6dacba30a7

### OpenWrt release

24.10.0-rc5

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R4

### Image kind

Official downloaded image

### Steps to reproduce

Note, this firmware was built by the image builder...

Do iperf3 from router to another machine to validate 1gbps connection.

```
Connecting to host x.x.x.x, port 5201
Reverse mode, remote host x.x.x.x is sending
[  5] local 10.11.12.9 port 41762 connected to x.x.x.x port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec   112 MBytes   941 Mbits/sec                  
[  5]   1.00-2.00   sec   112 MBytes   942 Mbits/sec                  
[  5]   2.00-3.00   sec   112 MBytes   942 Mbits/sec                  
[  5]   3.00-4.00   sec   112 MBytes   942 Mbits/sec                  
[  5]   4.00-5.00   sec   112 MBytes   942 Mbits/sec                  
[  5]   5.00-6.00   sec   112 MBytes   942 Mbits/sec                  
[  5]   6.00-7.00   sec   112 MBytes   941 Mbits/sec                  
[  5]   7.00-8.00   sec   112 MBytes   942 Mbits/sec                  
[  5]   8.00-9.00   sec   112 MBytes   941 Mbits/sec                  
[  5]   9.00-10.00  sec   112 MBytes   942 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  1.10 GBytes   944 Mbits/sec    4             sender
[  5]   0.00-10.00  sec  1.10 GBytes   941 Mbits/sec                  receiver

iperf Done.
```

Then, plug in a 10mbit device on the switch.

Notice that the connection shows as 10mbit:

```
Settings for lan1:
	Supported ports: [ TP MII ]
	Supported link modes:   10baseT/Half 10baseT/Full 
	                        100baseT/Half 100baseT/Full 
	                        1000baseT/Full 
	Supported pause frame use: Symmetric Receive-only
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  10baseT/Half 10baseT/Full 
	                        100baseT/Half 100baseT/Full 
	                        1000baseT/Full 
	Advertised pause frame use: Symmetric Receive-only
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Link partner advertised link modes:  10baseT/Full 
	Link partner advertised pause frame use: Symmetric Receive-only
	Link partner advertised auto-negotiation: Yes
	Link partner advertised FEC modes: Not reported
	Speed: 10Mb/s
	Duplex: Full
	Port: Twisted Pair
	PHYAD: 1
	Transceiver: external
	Auto-negotiation: on
	MDI-X: Unknown
	Supports Wake-on: d
	Wake-on: d
	Link detected: yes
```

### Actual behaviour

Notice that the connection shows as 10mbit:

```
Settings for lan1:
	Supported ports: [ TP MII ]
	Supported link modes:   10baseT/Half 10baseT/Full 
	                        100baseT/Half 100baseT/Full 
	                        1000baseT/Full 
	Supported pause frame use: Symmetric Receive-only
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  10baseT/Half 10baseT/Full 
	                        100baseT/Half 100baseT/Full 
	                        1000baseT/Full 
	Advertised pause frame use: Symmetric Receive-only
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Link partner advertised link modes:  10baseT/Full 
	Link partner advertised pause frame use: Symmetric Receive-only
	Link partner advertised auto-negotiation: Yes
	Link partner advertised FEC modes: Not reported
	Speed: 10Mb/s
	Duplex: Full
	Port: Twisted Pair
	PHYAD: 1
	Transceiver: external
	Auto-negotiation: on
	MDI-X: Unknown
	Supports Wake-on: d
	Wake-on: d
	Link detected: yes
```

and the iperf is very slow:

```
Connecting to host berrier, port 5201
Reverse mode, remote host x.x.x.x is sending
[  5] local 10.11.12.9 port 57994 connected to x.x.x.x port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  1.12 MBytes  9.43 Mbits/sec                  
[  5]   1.00-2.00   sec  1.25 MBytes  10.5 Mbits/sec                  
[  5]   2.00-3.00   sec  1.00 MBytes  8.39 Mbits/sec                  
[  5]   3.00-4.00   sec  1.12 MBytes  9.44 Mbits/sec                  
[  5]   4.00-5.00   sec  1.12 MBytes  9.44 Mbits/sec                  
[  5]   5.00-6.00   sec  1.00 MBytes  8.39 Mbits/sec                  
[  5]   6.00-7.00   sec  1.25 MBytes  10.5 Mbits/sec                  
[  5]   7.00-8.00   sec  1.25 MBytes  10.5 Mbits/sec                  
[  5]   8.00-9.00   sec  1.00 MBytes  8.39 Mbits/sec                  
[  5]   9.00-10.00  sec  1.12 MBytes  9.44 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  12.3 MBytes  10.3 Mbits/sec    6             sender
[  5]   0.00-10.00  sec  11.2 MBytes  9.44 Mbits/sec                  receiver
```

although 400mbit sending:

```
Connecting to host berrier, port 5201
[  5] local 10.11.12.9 port 60644 connected to 10.11.12.3 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  52.2 MBytes   438 Mbits/sec    0   1.29 MBytes       
[  5]   1.00-2.00   sec  45.9 MBytes   385 Mbits/sec    0   2.84 MBytes       
[  5]   2.00-3.00   sec  51.5 MBytes   432 Mbits/sec    5   3.14 MBytes       
[  5]   3.00-4.00   sec  48.0 MBytes   403 Mbits/sec    3   3.14 MBytes       
[  5]   4.00-5.00   sec  45.4 MBytes   381 Mbits/sec    1   2.38 MBytes       
[  5]   5.00-6.00   sec  46.9 MBytes   393 Mbits/sec    0   2.57 MBytes       
[  5]   6.00-7.00   sec  46.8 MBytes   392 Mbits/sec    0   2.72 MBytes       
[  5]   7.00-8.00   sec  46.5 MBytes   390 Mbits/sec    0   2.84 MBytes       
[  5]   8.00-9.00   sec  49.8 MBytes   417 Mbits/sec    0   2.94 MBytes       
[  5]   9.00-10.00  sec  48.4 MBytes   406 Mbits/sec    2   3.01 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   483 MBytes   405 Mbits/sec   11             sender
[  5]   0.00-10.00  sec   482 MBytes   404 Mbits/sec                  receiver
```

### Expected behaviour

I would expect the port that is connected at 1000mbit to still perform at gigabit speeds.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

issuex, 2025-01-22:
already enable hardware/software flow offload ?

aimacintyre, 2025-01-23:
Is EEE enabled on either the port with the problematic wake-on-lan device or the port with collateral effect?  If so, does disabling (or enabling if disabled) EEE change the behaviour?

wberrier, 2025-01-23:
> Is EEE enabled on either the port with the problematic wake-on-lan device or the port with collateral effect? If so, does disabling (or enabling if disabled) EEE change the behaviour?

It looks like by default it is disabled:

```
root@router-2:~# ethtool --show-eee lan2
EEE Settings for lan2:
	EEE status: disabled
	Tx LPI: 30 (us)
	Supported EEE link modes:  100baseT/Full 
	                           1000baseT/Full 
	Advertised EEE link modes:  Not reported
	Link partner advertised EEE link modes:  100baseT/Full 
	                                         1000baseT/Full 
root@router-2:~# ethtool lan2
Settings for lan2:
	Supported ports: [ TP MII ]
	Supported link modes:   10baseT/Half 10baseT/Full 
	                        100baseT/Half 100baseT/Full 
	                        1000baseT/Full 
	Supported pause frame use: Symmetric Receive-only
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  10baseT/Half 10baseT/Full 
	                        100baseT/Half 100baseT/Full 
	                        1000baseT/Full 
	Advertised pause frame use: Symmetric Receive-only
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Link partner advertised link modes:  10baseT/Full 
	Link partner advertised pause frame use: Symmetric Receive-only
	Link partner advertised auto-negotiation: Yes
	Link partner advertised FEC modes: Not reported
	Speed: 10Mb/s
	Duplex: Full
	Port: Twisted Pair
	PHYAD: 2
	Transceiver: external
	Auto-negotiation: on
	MDI-X: Unknown
	Supports Wake-on: d
	Wake-on: d
	Link detected: yes
```

and enabling it doesn't help:

```
root@router-2:~# ethtool --set-eee lan2 eee on
root@router-2:~# ethtool --show-eee lan2
EEE Settings for lan2:
	EEE status: enabled - inactive
	Tx LPI: 30 (us)
	Supported EEE link modes:  100baseT/Full 
	                           1000baseT/Full 
	Advertised EEE link modes:  100baseT/Full 
	                            1000baseT/Full 
	Link partner advertised EEE link modes:  100baseT/Full 
	                                         1000baseT/Full 
```

wberrier, 2025-01-23:
Another odd finding:

with the dir 882, plugging the 10mbit device into a lan port (where another lan port is plugged into a lan port on the bpi-r4), iperf from the bpi-r4 to a neighbor device is degraded.

But, if the 10mbit device is plugged into the wan port on the dir 882 (as part of the br0 bridge), the degredation does not occur.  🤔

FerrumMaster, 2025-01-23:
> Another odd finding:
>
> with the dir 882, plugging the 10mbit device into a lan port (where another lan port is plugged into a lan port on the bpi-r4), iperf from the bpi-r4 to a neighbor device is [...]

I have read it is known behavior on BPI-R3 also, I haven't tried my BPI-R4 as I cannot get WIFI to work fully, ticket is open. As soon you do mixed ie 100/10 along with 1000 it degrades switch performance, and IMHO it was not fixed... getting a cheap 2.5Gbe switch is more easy these days either way.

wberrier, 2025-01-23:
I will try putting a dumb switch in the mix, but that's kind of what I'm doing already: using the dir 882 as that "switch", but that switch appears to have 2 "hardware ports", eth0 (with all lan ports) and wan.

Thanks for the heads up, if anything, this is a "known issue".  I wouldn't really worry too much about it, but I do like being able to use "wake on lan", and putting an extra switch in the mix just for that is certainly less than optimal.

FerrumMaster, 2025-01-23:
> I will try putting a dumb switch in the mix, but that's kind of what I'm doing already: using the dir 882 as that "switch", but that switch appears to have 2 "hardware ports", eth0 (with all lan [...]

Switch is a good thing... I use 30$ RTL8372N based 2.5GbE Switch 2.5GBASE-T 4 Port 2.5G RJ45 with 2 Port 10G SFP and connect the device using SFP 10G at max speed with the switch. What's the point having speedy WIFIs, they go past gigabit speeds. I use the switch for a year already, despite the super low cost, it works fine and is very power efficient unlike older devices supporting such speeds.

There was and is something off... Daniel fixed something and not sure how it went.

https://forum.banana-pi.org/t/bpi-r4-100mbit-broken/17049

I cannot find the R3 thread explaining the issue anymore.

wberrier, 2025-01-24:
Agreed, I have a similar setup with my other bpi-r4 (10gbps DAC cable to 2.5gbps switch).  I upgraded to wifi 7 and had to upgrade all of my switches 😄 

This unit I'm reporting about here is a non-routing AP bridge providing wifi and only this one ethernet client.  I have some sfp modules on order, so I'm hoping once this bridge is connected via sfp, it won't matter if the 10/1000mbps client degrades the internal lan switch... 🤞

Even still, it would be nice to not have to resort to workarounds for hardware that should handle 10/100/1000.

Thanks for finding and posting that thread!  That's super helpful.