openwrt/openwrt #17698: Lan port problems on FRITZ!Box 7362 SL running 23.05.5

skleeschulte, 2025-01-22:
### Describe the bug

Hi! I observe two problems with the LAN ports:

(1) If I connect my printer (a Brother DCP-9055CDN) to LAN3 or LAN4, it does not get network connectivity after OpenWrt boot. The OpenWrt status page correctly shows a 100M link for the port and some data going out, but no data coming in. As soon as I replug the printer (same or other LAN port), the connection is working. If I plug the printer to LAN1 or LAN2, it works right after boot. With my notebook, I do not have the problem - it get's network connectivity on any LAN port right after OpenWrt reboot. Maybe important: LAN1 and LAN2 are GbE ports, LAN3 and LAN4 are 100M ports. My printer has a 100M port, while my notebook has a GbE port.

This is how lan3 is shown on the status page after boot with my printer connected:
[image]

(2) Sometimes, ports are shown with "no link" on the status page, despite the link working fine and received and transmitted packages counting up on the status page.

### OpenWrt version

r24106-10cc5fcd00

### OpenWrt release

23.05.5

### OpenWrt target/subtarget

lantiq/xrx200

### Device

AVM FRITZ!Box 7362 SL

### Image kind

Official downloaded image

### Steps to reproduce

Regarding problem (1): To narrow this down, maybe someone else can check if this is a problem of 100M devices being connected to a 100M port on the router (LAN3 or LAN4).

Regarding prolem (2): Just have a look at the status page now and then with devices connected to the LAN ports.

### Actual behaviour

See above.

### Expected behaviour

Devices should have network connectivity on every port right after boot, without the need of replugging them.

The status page should correctly show the link status.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

skleeschulte, 2025-01-22:
Update: When I restart the network service after a reboot (LuCI -> System -> Startup -> Restart-Button in network line) the printer gets connectivity.

brada4, 2025-01-23:
Does it lose connectivity afterwards in some way? IF/WHEN it loses connection again, say in a week(end) check the logs.  Probably go in LuCI to System/System/Logging and change default 64kB log size to 1024kB to not lose log entries.

skleeschulte, 2025-01-25:
> Does it lose connectivity afterwards in some way? IF/WHEN it loses connection again, say in a week(end) check the logs. Probably go in LuCI to System/System/Logging and change default 64kB log size to 1024kB to not lose log entries.

So far, i did not observe that, but I increased the log size just in case.

As I can "fix" the problem by restarting the network service after a reboot, can I use that to narrow the problem down?