openwrt/openwrt #17704: 24.10.0-rc6: build problem

mic-user, 2025-01-23:
### Describe the bug

Can't build as below.

[image]

### OpenWrt version

24.10.0-rc6

### OpenWrt release

N/A

### OpenWrt target/subtarget

mvebu/cortexa9

### Device

Linksys WRT32X

### Image kind

Self-built image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

arixmkii, 2025-01-23:
It has not been announced yet on official web page or in GH releases. You are probably seeing this because this bug is still present https://github.com/openwrt/openwrt/issues/12502 I would say you need just wait for the official announcement of RC6.

namiltd, 2025-01-23:
There are always these types of reports right after a new version is released. You have to wait a few days for everything to be built.

mic-user, 2025-01-24:
build works.

closed 2025-01-24 by mic-user