openwrt/openwrt #17709: OpenWrt 24.10.0-rc6 build problem

uyguremre, 2025-01-23:
### Describe the bug

[image]

Generate local signing keys...
Generate local certificate...
Package list missing or not up-to-date, generating it.

Building package index...
Downloading https://downloads.openwrt.org/releases/24.10.0-rc6/targets/lantiq/xrx200/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-lantiq/../../../../builder/dl/openwrt_core
Downloading https://downloads.openwrt.org/releases/24.10.0-rc6/targets/lantiq/xrx200/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/24.10.0-rc6/packages/mips_24kc/base/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-lantiq/../../../../builder/dl/openwrt_base
Downloading https://downloads.openwrt.org/releases/24.10.0-rc6/packages/mips_24kc/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/24.10.0-rc6/targets/lantiq/xrx200/kmods/6.6.73-1-fe0372157644410cd19931638b8a5657/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-lantiq/../../../../builder/dl/openwrt_kmods
Downloading https://downloads.openwrt.org/releases/24.10.0-rc6/targets/lantiq/xrx200/kmods/6.6.73-1-fe0372157644410cd19931638b8a5657/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/24.10.0-rc6/packages/mips_24kc/luci/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-lantiq/../../../../builder/dl/openwrt_luci
Downloading https://downloads.openwrt.org/releases/24.10.0-rc6/packages/mips_24kc/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/24.10.0-rc6/packages/mips_24kc/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-lantiq/../../../../builder/dl/openwrt_packages
Downloading https://downloads.openwrt.org/releases/24.10.0-rc6/packages/mips_24kc/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/24.10.0-rc6/packages/mips_24kc/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-lantiq/../../../../builder/dl/openwrt_routing
Downloading https://downloads.openwrt.org/releases/24.10.0-rc6/packages/mips_24kc/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/24.10.0-rc6/packages/mips_24kc/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-lantiq/../../../../builder/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/24.10.0-rc6/packages/mips_24kc/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-lantiq/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * opkg_install_cmd: Cannot install package Ath10k-board-qca988x.
 * opkg_install_cmd: Cannot install package libubus20231128.
 * opkg_install_cmd: Cannot install package libuci20130104.
 * resolve_conffiles: Existing conffile /builder/build_dir/target-mips_24kc_musl/root-lantiq/etc/ppp/chap-secrets is different from the conffile in the new package. The new conffile will be placed at /builder/build_dir/target-mips_24kc_musl/root-lantiq/etc/ppp/chap-secrets-opkg.
make[2]: *** [Makefile:234: package_install] Error 255
make[1]: *** [Makefile:171: _call_manifest] Error 2
make: *** [Makefile:349: manifest] Error 2

### OpenWrt version

24.10.0-rc6 

### OpenWrt release

24.10.0-rc6 

### OpenWrt target/subtarget

lantiq

### Device

homhub 5a

### Image kind

Self-built image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2025-01-23:
There are always these types of reports right after a new version is released. You have to wait a few days for everything to be built.

mic-user, 2025-01-24:
build works.

xirixiz, 2025-01-24:
You're too fast. Sometimes it takes some time before the firmware is fully availble. It is available in the firmware list, but in the background processes are still running to make "everything" available for your profile.

mic-user, 2025-01-24:
> You're too fast. Sometimes it takes some time before the firmware is fully availble. It is available in the firmware list, but in the background processes are still running to make "everything" available for your profile.

Hey. you xirixiz. 
You must NOT make wrong perception or blame or mockery people eventhough you have totally wrong percetion.

Firest, The people here is helping/letting know/report to openwrt by taking time whatif missing since every release same issues repeat over years.

Second, regarding the system, 
if that the release version is not ready to release then the system must NOT show the un-ready version to the public which every professional system does. Blue-green or something else what openwrt can imploy.
Of course openwrt contributers are(as a single team) really put precious time for global efficiency(I thanks to all of them),
BUT, current process is not so professional figure so far while decades of operation.

So, I want to tell you that watch your mouth to freely contributing people rather thanks. Got that?

closed 2025-01-24 by hauke

xirixiz, 2025-01-25:
@mic-user First, let me clarify: I was not making any mockery or blaming anyone. You’re interpreting my response with your own perspective. I was simply trying to assist the original poster in understanding what was happening. My apologies if the issue starter or others felt this way. It was definitely not my intention. I just don't understand why you respond in this tone yourself when you accuse someone else of not responding appropriately.

It’s important to recognize that many people spend their free time and effort working on this project for the benefit of the entire community. Instead of directing complaints toward those volunteering their time, perhaps consider contributing directly to help improve the project. Constructive participation is far more valuable than criticism alone.

Maintaining OpenWRT requires significant effort, yet users often open new issues without reviewing existing ones or checking for similar reports. Despite this, contributors consistently take the time to respond and provide assistance.

mic-user, 2025-01-25:
@xirixiz I also must clearify what is wrong with openWRT at the moment including this reply.
That was not explosion of emotion or biased perspective, 
but just watching pathetic situation if you can understand it.
Not to start fight but want to deliver what I'm saying it, what OpenWRT would do.

See the below process.
#1. At the public access openwrt image build portal > people access > build > got error > report > OpenWRT side people whining/appeal to the PUBLIC that "takes time until official release" > and next time > unfinished build image setting up while portal open > open public access portal(still) > repeat eternally.

#2. At the Q&A forum > people got errors from OpenWRT(not a productive opinion) > ask > Replied by someone(especially the potato guy, blue icon mambo jumbo documentary guy or some under development country young guy, etc) who don't even understand what the question is, that even question back. > the people reply back sincerely to help themself to get right answer > another reply from the guys who don't understand but just make eternal ping-pong post > repeat several times > Askers get tired and stressful, even spent time for nothing >  No solution finally get > the asker finally leave and say'ok, I'll find another way' > repeat eternally.

Above both are happened at this moment.
You guys making own self destruction cycle. Make problem OpenWRT-self > No solution for the public > whining/appeal to people > repeat.
Means your OpenWRT are the EVERY PROBLEM PROVIDER not outside others.
Your single error affect to thousand of people. Under the situation, if there is several report or articles, 
then you guys must say 'thanks god what a luck' rather not get stress.

So, stop asking to slow down people or  stop request 'search' same issue but change the process.
1. Employee blue-green release process or appropriate release system to isolate un-ready image from the public. Or disable build button at least while you guys working. Is that so hard?
2. Take out role or permission to post action by useless people who spam posting gibberish text rather than solution. So sick & tired to ask in the forum.
These two are what I'm claiming of if I need to say directly.
This kind of improvement will make better operation environment for OpenWRT and socierty(including me).