openwrt/openwrt #17711: OpenWrt 24.10.0-rc6 WAN interface issue

BUG-reporter, 2025-01-24:
### Describe the bug

Once I upgraded from 23.05.5 to 24.10.0-rcX (6-th at the moment), WAN interface are not able to receive any configs from ISP-s DHCP. Some messages in logs found related to WAN:

```
Wed Jan 22 19:54:15 2025 daemon.notice netifd: wan (2756): udhcpc: received SIGTERM
Wed Jan 22 19:54:15 2025 daemon.notice netifd: wan (2756): udhcpc: entering released state
Wed Jan 22 19:54:15 2025 daemon.notice netifd: wan (2756): Command failed: ubus call network.interface notify_proto { "action": 0, "link-up": false, "keep": false, "interface": "wan" } (Permission denied)
Wed Jan 22 19:54:15 2025 daemon.notice netifd: Interface 'wan' is now down
Wed Jan 22 19:54:15 2025 daemon.notice netifd: Interface 'wan' is setting up now
Wed Jan 22 19:54:15 2025 daemon.notice netifd: wan (3192): udhcpc: started, v1.36.1
Wed Jan 22 19:54:15 2025 daemon.notice netifd: wan (3192): udhcpc: broadcasting discover
Wed Jan 22 19:54:16 2025 daemon.warn odhcpd[1915]: No default route present, overriding ra_lifetime to 0!
```

[Kernel.log](https://github.com/user-attachments/files/18528448/Kernel.log)
[System.log](https://github.com/user-attachments/files/18528449/System.log)

### OpenWrt version

24.10.0-rc6

### OpenWrt release

24.10

### OpenWrt target/subtarget

n/a

### Device

ASUS AC-85P

### Image kind

Official downloaded image

### Steps to reproduce

Just upgrade to 24.10-rcX

### Actual behaviour

WAN interface has no network config from ISP.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

castillofrancodamian, 2025-01-24:
I too see a renewal even before its time. Apparently the allocation lasts 6 hours, but OpenWrt asks every 3 hours.

```
Mon Jan 20 10:40:48 2025 daemon.notice netifd: wan (8310): udhcpc: sending renew to server 172.20.2.95
Mon Jan 20 10:40:48 2025 daemon.notice netifd: wan (8310): udhcpc: lease of 190.244.140.16 obtained from 172.20.2.95, lease time 21600
Mon Jan 20 10:40:49 2025 user.notice firewall: Reloading firewall due to ifupdate of wan (wan)
Mon Jan 20 13:40:48 2025 daemon.notice netifd: wan (8310): udhcpc: sending renew to server 172.20.2.95
Mon Jan 20 13:40:48 2025 daemon.notice netifd: wan (8310): udhcpc: lease of 190.244.140.16 obtained from 172.20.2.95, lease time 21600
Mon Jan 20 16:40:47 2025 daemon.notice netifd: wan (8310): udhcpc: sending renew to server 172.20.2.95
Mon Jan 20 16:40:47 2025 daemon.notice netifd: wan (8310): udhcpc: lease of 190.244.140.16 obtained from 172.20.2.95, lease time 21600
Mon Jan 20 19:40:48 2025 daemon.notice netifd: wan (8310): udhcpc: sending renew to server 172.20.2.95
Mon Jan 20 19:40:48 2025 daemon.notice netifd: wan (8310): udhcpc: lease of 190.244.140.16 obtained from 172.20.2.95, lease time 21600
Mon Jan 20 20:53:06 2025 daemon.notice netifd: wan2 (2806): udhcpc: sending renew to server 192.168.70.1
Mon Jan 20 20:53:06 2025 daemon.notice netifd: wan2 (2806): udhcpc: lease of 192.168.70.135 obtained from 192.168.70.1, lease time 86400
Mon Jan 20 22:40:48 2025 daemon.notice netifd: wan (8310): udhcpc: sending renew to server 172.20.2.95
Mon Jan 20 22:40:48 2025 daemon.notice netifd: wan (8310): udhcpc: lease of 190.244.140.16 obtained from 172.20.2.95, lease time 21600
Tue Jan 21 01:40:48 2025 daemon.notice netifd: wan (8310): udhcpc: sending renew to server 172.20.2.95
Tue Jan 21 01:40:48 2025 daemon.notice netifd: wan (8310): udhcpc: lease of 190.244.140.16 obtained from 172.20.2.95, lease time 21600
Tue Jan 21 04:40:47 2025 daemon.notice netifd: wan (8310): udhcpc: sending renew to server 172.20.2.95
Tue Jan 21 04:40:47 2025 daemon.notice netifd: wan (8310): udhcpc: lease of 190.244.140.16 obtained from 172.20.2.95, lease time 21600
Tue Jan 21 07:40:48 2025 daemon.notice netifd: wan (8310): udhcpc: sending renew to server 172.20.2.95
Tue Jan 21 07:40:48 2025 daemon.notice netifd: wan (8310): udhcpc: lease of 190.244.140.16 obtained from 172.20.2.95, lease time 21600
Tue Jan 21 08:53:06 2025 daemon.notice netifd: wan2 (2806): udhcpc: sending renew to server 192.168.70.1
Tue Jan 21 08:53:06 2025 daemon.notice netifd: wan2 (2806): udhcpc: lease of 192.168.70.135 obtained from 192.168.70.1, lease time 86400
Tue Jan 21 10:40:48 2025 daemon.notice netifd: wan (8310): udhcpc: sending renew to server 172.20.2.95
Tue Jan 21 10:40:48 2025 daemon.notice netifd: wan (8310): udhcpc: lease of 190.244.140.16 obtained from 172.20.2.95, lease time 21600
Tue Jan 21 13:40:48 2025 daemon.notice netifd: wan (8310): udhcpc: sending renew to server 172.20.2.95
Tue Jan 21 13:40:48 2025 daemon.notice netifd: wan (8310): udhcpc: lease of 190.244.140.16 obtained from 172.20.2.95, lease time 21600
Tue Jan 21 16:40:48 2025 daemon.notice netifd: wan (8310): udhcpc: sending renew to server 172.20.2.95
Tue Jan 21 16:40:48 2025 daemon.notice netifd: wan (8310): udhcpc: lease of 190.244.140.16 obtained from 172.20.2.95, lease time 21600
Tue Jan 21 19:40:47 2025 daemon.notice netifd: wan (8310): udhcpc: sending renew to server 172.20.2.95
Tue Jan 21 19:40:47 2025 daemon.notice netifd: wan (8310): udhcpc: lease of 190.244.140.16 obtained from 172.20.2.95, lease time 21600
Tue Jan 21 20:53:06 2025 daemon.notice netifd: wan2 (2806): udhcpc: sending renew to server 192.168.70.1
Tue Jan 21 20:53:06 2025 daemon.notice netifd: wan2 (2806): udhcpc: lease of 192.168.70.135 obtained from 192.168.70.1, lease time 86400
Tue Jan 21 22:40:48 2025 daemon.notice netifd: wan (8310): udhcpc: sending renew to server 172.20.2.95
Tue Jan 21 22:40:48 2025 daemon.notice netifd: wan (8310): udhcpc: lease of 190.244.140.16 obtained from 172.20.2.95, lease time 21600
Wed Jan 22 01:40:48 2025 daemon.notice netifd: wan (8310): udhcpc: sending renew to server 172.20.2.95
Wed Jan 22 01:40:48 2025 daemon.notice netifd: wan (8310): udhcpc: lease of 190.244.140.16 obtained from 172.20.2.95, lease time 21600
Wed Jan 22 04:40:48 2025 daemon.notice netifd: wan (8310): udhcpc: sending renew to server 172.20.2.95
Wed Jan 22 04:40:48 2025 daemon.notice netifd: wan (8310): udhcpc: lease of 190.244.140.16 obtained from 172.20.2.95, lease time 21600
Wed Jan 22 07:40:47 2025 daemon.notice netifd: wan (8310): udhcpc: sending renew to server 172.20.2.95
Wed Jan 22 07:40:47 2025 daemon.notice netifd: wan (8310): udhcpc: lease of 190.244.140.16 obtained from 172.20.2.95, lease time 21600
Wed Jan 22 08:53:05 2025 daemon.notice netifd: wan2 (2806): udhcpc: sending renew to server 192.168.70.1
Wed Jan 22 08:53:05 2025 daemon.notice netifd: wan2 (2806): udhcpc: lease of 192.168.70.135 obtained from 192.168.70.1, lease time 86400
Wed Jan 22 10:40:47 2025 daemon.notice netifd: wan (8310): udhcpc: sending renew to server 172.20.2.95
Wed Jan 22 10:40:48 2025 daemon.notice netifd: wan (8310): udhcpc: lease of 190.244.140.16 obtained from 172.20.2.95, lease time 21600
Wed Jan 22 13:40:48 2025 daemon.notice netifd: wan (8310): udhcpc: sending renew to server 172.20.2.95
Wed Jan 22 13:40:48 2025 daemon.notice netifd: wan (8310): udhcpc: lease of 190.244.140.16 obtained from 172.20.2.95, lease time 21600
[... cut ...]
Thu Jan 23 19:40:49 2025 daemon.notice netifd: wan (8310): udhcpc: lease of 190.244.140.16 obtained from 172.20.2.95, lease time 21600
Thu Jan 23 20:53:06 2025 daemon.notice netifd: wan2 (2806): udhcpc: sending renew to server 192.168.70.1
Thu Jan 23 20:53:06 2025 daemon.notice netifd: wan2 (2806): udhcpc: lease of 192.168.70.135 obtained from 192.168.70.1, lease time 86400
Thu Jan 23 22:40:49 2025 daemon.notice netifd: wan (8310): udhcpc: sending renew to server 172.20.2.95
Thu Jan 23 22:40:49 2025 daemon.notice netifd: wan (8310): udhcpc: lease of 190.244.140.16 obtained from 172.20.2.95, lease time 21600
```

aimacintyre, 2025-01-24:
> Apparently the allocation lasts 6 hours, but OpenWrt asks every 3 hours.

This is normal DHCP operation: the first lease renewal attempt is made half-way through the lease period.

aimacintyre, 2025-01-24:
@BUG-reporter : there are reports of some Mediatek switches with EEE enabled in 24.10, where it wasn't previously, causing issues; it might be helpful if you could investigate this (see [this post](https://forum.openwrt.org/t/wax206-24-10-0-rc4-wan-port-ooo/220279/5) for some more info as a starting point).

BUG-reporter, 2025-01-24:
> Invalid Version reported. `24.10.0-rc6` Is this from a clean repository?

Of course yes. Downloaded from OpenWrt repository through firmware selector.

BUG-reporter, 2025-01-24:
> [@BUG-reporter](https://github.com/BUG-reporter) : there are reports of some Mediatek switches with EEE enabled in 24.10, where it wasn't previously, causing issues; it might be helpful if you could [...]

Ok, do you need me try to switch off EEE as well and report a result?

BUG-reporter, 2025-01-29:
I checked. This is not my case becase my wan doesn't support eee.

BUG-reporter, 2025-01-29:
> [@BUG-reporter](https://github.com/BUG-reporter) : there are reports of some Mediatek switches with EEE enabled in 24.10, where it wasn't previously, causing issues; it might be helpful if you could [...]

I checked. This is not my case becase my wan doesn't support eee.

BUG-reporter, 2025-02-03:
The similar old reports from the past:

https://github.com/openwrt/openwrt/issues/9069 <-- this one is mine
https://forum.openwrt.org/t/internet-speed-capped-at-100mbps-by-the-router/132045 <--this one is from others

So, please, unmark my report as invalid!

BUG-reporter, 2025-02-05:
Well, 24.10 released and this bug is still not fixed.

[kernel_24.10.log](https://github.com/user-attachments/files/18664762/kernel_24.10.log)
[system_24.10.log](https://github.com/user-attachments/files/18664761/system_24.10.log)
[ethtool_wan.txt](https://github.com/user-attachments/files/18664764/ethtool_wan.txt)

Hantick, 2025-02-09:
Same problem on my side. No Wan after upgrading to 24.10.0 on my Cudy WR3000S

BUG-reporter, 2025-02-09:
> Same problem on my side. No Wan after upgrading to 24.10.0 on my Cudy WR3000S

Did you try a mentioned woraround ?

Hantick, 2025-02-10:
Yes, the workaround did not resolve my issue. Although resetting to default settings and then uploading the backup config once again resolved the issue

BUG-reporter, 2025-02-10:
> Yes, the workaround did not resolve my issue. Although resetting to default settings and then uploading the backup config once again resolved the issue

Then your case is not the same as mine.

BUG-reporter, 2025-02-10:
Another one issue I discovered is that my download speed maximum is 30Mbps while connecting directly to the ISP without a router I have full 100mbps.

lvqier, 2025-02-14:
> Found a workaround:
> 
> `ethtool -s wan autoneg off speed 100 duplex full`
> 
> After run this command, wan interface becomes configured with network config. BTW, when my device first got Owrt support there were exactly the same issue.

Same problem on xiaomi_mi-router-3g. 
This workarround works.

BUG-reporter, 2025-02-14:
> > Found a workaround:
> > `ethtool -s wan autoneg off speed 100 duplex full`
> > After run this command, wan interface becomes configured with network config. BTW, when my device first got Owrt support there were exactly the same issue.
> 
> Same problem on xiaomi_mi-router-3g. This workarround works.

Could you please also check the download speed on 24.10 and 23.05 and compare them?

lvqier, 2025-02-14:
> Could you please also check the download speed on 24.10 and 23.05 and compare them?

I would like to test download speed on 24.10, but no further device for 23.05. 
Would you like to share your test method?

kofec, 2025-02-14:
In different issue: https://github.com/openwrt/openwrt/issues/15348#issuecomment-2499740405
 I wrote:
I add info that when using:
ethtool -s lan2 speed 100 duplex full autoneg off
I had low performance - about 20Mb/s +/- 3Mb/s
so I had to change to
ethtool -s lan2 speed 100 duplex full autoneg on
And now all is fine

Dmole, 2025-02-16:
Some are forced to downgrade due to wan not working anymore:

https://forum.openwrt.org/t/openwrt-24-10-0-first-stable-release/223845

BUG-reporter, 2025-02-16:
> Some are forced to downgrade due to wan not working anymore:
> 
> https://forum.openwrt.org/t/openwrt-24-10-0-first-stable-release/223845

Same as me - downgraded to 23.05.5

Yohan54, 2025-03-10:
I having same issue were WAN DHCP client (Openwrt) is requesting a renew lease every hour even though setting is set 12h. Appears reading above the only fix is downgrade from  24.10 to 23.05. Shouldn't 24.10 be pulled if has bug, just asking.

aimacintyre, 2025-03-10:
> I having same issue were WAN DHCP client (Openwrt) is requesting a renew lease every hour to ISP modem, even though setting is set 12h.  Appears reading above the only fix is downgrade from 24.10 to 23.05.

If you're successfully getting WAN leases renewed, as distinct from leases expiring and then a new lease being granted, you don't actually have any problem (and certainly not any problem relating to this issue) - the lease period would be 7200 seconds (2 hours) for 1 hour renewals; 12 hour leases would be 43200 seconds with renewals every 6 hours.  If the ISP modem is only giving your device a 2 hour lease when you think it is set for 12 hours, the problem is with the ISP modem.

> Shouldn't 24.10 be pulled if has bug, just asking.

No bug has been confirmed.

BUG-reporter, 2025-03-10:
Could someone remove "invalid" from my report, please?

basrieter, 2025-03-10:
I disabled EEE on WAN which fixed my issue on a Linksys EA7500.

LGA1150, 2025-03-10:
See #17351

LGA1150, 2025-03-10:
> Found a workaround:
> 
> `ethtool -s wan autoneg off speed 100 duplex full`
> 
> After run this command, wan interface becomes configured with network config. BTW, when my device first got Owrt support there were exactly the same issue.

> Another one issue I discovered is that my download speed maximum is 30Mbps while connecting directly to the ISP without a router I have full 100mbps.

If you turn off autoneg only on one side, the other side cannot determine the duplex and have to fallback to _half duplex_ -- probably not what you wanted.

BUG-reporter, 2025-03-21:
SNAPSHOT (r29064-696ad7b1aa)
Still not fixed.

BUG-reporter, 2025-05-06:
24.10.1. Not fixed.

BUG-reporter, 2025-05-20:
SNAPSHOT r29684-a238170e57
Not fixed.

Neustradamus, 2025-05-24:
Linked to:
- https://github.com/openwrt/openwrt/issues/18100
- https://github.com/openwrt/openwrt/pull/18758
- https://github.com/openwrt/openwrt/issues/18052
- https://github.com/openwrt/openwrt/issues/17711
- https://github.com/openwrt/openwrt/issues/17351