openwrt/openwrt #17714: make gcc/toolchain enabel libgomp  WARNING: skipping libgomp -- package has no install section

hyjk555, 2025-01-24:
### Describe the bug

1  .config:

CONFIG_PACKAGE_jsonfilter=y
CONFIG_PACKAGE_libasan=y
CONFIG_PACKAGE_libatomic=y
CONFIG_PACKAGE_libc=y
CONFIG_PACKAGE_libgcc=y
CONFIG_PACKAGE_libgomp=y
CONFIG_PACKAGE_liblsan=y
CONFIG_PACKAGE_libpthread=y
CONFIG_PACKAGE_librt=y
CONFIG_PACKAGE_libstdcpp=y

2.  edit gcc enable libgomp:  /devel/gcc/Makefile
    --disable-libgomp \  -->  --enable-libgomp 

3.gcc default version:12.3.0

4. use type : glibc

xysds@ubuntu:~/friendlywrt$ make -j1 V=s package/feeds/packages/gcc/compile
make -r package/feeds/packages/gcc/compile: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/home/xysds/friendlywrt/include/toplevel.mk:232：package/feeds/packages/gcc/compile] 错误 1
xysds@ubuntu:~/friendlywrt$ make -j1 V=s package/feeds/packages/gcc/compile
Collecting package info: done
make[2]: Entering directory '/home/xysds/friendlywrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/xysds/friendlywrt/scripts/config'
make[1]: Entering directory '/home/xysds/friendlywrt'
cd "/home/xysds/friendlywrt"; git log --format=%h -1 toolchain > /home/xysds/friendlywrt/tmp/.ver_check
cmp -s /home/xysds/friendlywrt/tmp/.ver_check /home/xysds/friendlywrt/staging_dir/toolchain-aarch64_generic_gcc-12.3.0_glibc/stamp/.ver_check || { \
	rm -rf /home/xysds/friendlywrt/build_dir/target-aarch64_generic_glibc /home/xysds/friendlywrt/staging_dir/target-aarch64_generic_glibc /home/xysds/friendlywrt/staging_dir/toolchain-aarch64_generic_gcc-12.3.0_glibc /home/xysds/friendlywrt/build_dir/toolchain-aarch64_generic_gcc-12.3.0_glibc; \
	mkdir -p /home/xysds/friendlywrt/staging_dir/toolchain-aarch64_generic_gcc-12.3.0_glibc/stamp; \
	mv /home/xysds/friendlywrt/tmp/.ver_check /home/xysds/friendlywrt/staging_dir/toolchain-aarch64_generic_gcc-12.3.0_glibc/stamp/.ver_check; \
}
+ mkdir -p /home/xysds/friendlywrt/staging_dir/target-aarch64_generic_glibc
+ cd /home/xysds/friendlywrt/staging_dir/target-aarch64_generic_glibc
+ mkdir -p bin lib stamp usr/include usr/lib
mkdir -p /home/xysds/friendlywrt/build_dir/target-aarch64_generic_glibc/stamp
touch /home/xysds/friendlywrt/staging_dir/target-aarch64_generic_glibc/.prepared
make[2]: Entering directory '/home/xysds/friendlywrt/package/libs/toolchain'
Makefile:735: WARNING: skipping libgomp -- package has no install section

My opewnrt type is glibc instead of musl

Since my software requires this, I tried to check and make it
But this warning appears. No output libgomp so
I see that there are definitions for 'define Package/libgomp/install' in the toolchain
I have also tried make toolchain, but it did not output libgomp so

I don't know how to get him to output this module

### OpenWrt version

r23809-234f1a2efa

### OpenWrt release

23.05.3

### OpenWrt target/subtarget

rockchip/armv8

### Device

FriendlyElec NanoPC-T6 LTS

### Image kind

Self-built image

### Steps to reproduce

1.  CONFIG_PACKAGE_libgomp=y
2.  /devel/gcc/Makefile
    --disable-libgomp \  -->  --enable-libgomp 

### Actual behaviour

No output libgomp so

### Expected behaviour

output:libgomp.so

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-01-24:
What software requires -lgomp ?

hyjk555, 2025-01-26:
> What software requires -lgomp ?

Python's Ninja

hyjk555, 2025-01-26:
I have found a way:
1. Edit: toolchain\gcc\common.mk Modify -- enable libcomp\
2. Edit: package\libs\toolchain\Makefile Add install option

define Package/libgfortran/install
$(INSTALL_DIR) $(1)/usr/lib
$(CP) $(TOOLCHAIN_DIR)/lib/libgfortran.so.* $(1)/usr/lib/
endef

#----------add-----------------
define Package/libgomp/install
$(INSTALL_DIR) $(1)/lib
$(CP) $(TOOLCHAIN_DIR)/lib/libgomp.so.* $(1)/lib/
$(CP) $(TOOLCHAIN_DIR)/lib/libgomp.spec $(1)/lib/
endef
#----------end -----------------

3. Add dependencies to feeds\packages\imagemagick\Makefile, I don't know why not add libgomp. This dependency will compile with an error
DEPENDS:=+libltdl +libpthread +zlib +libfreetype +libpng +libjpeg +libtiff +libgomp

feeds\packages\devel\gcc
This configuration can be left unchanged, I don't understand why there is also a configuration here. I think the content in toolchain\gcc\common.mk is similar.

Finally, remember to clean before making
make toolchain/gcc/final/clean
make toolchain/gcc/final/compile
make package/libs/toolchain/compile

Finally, libgomp. *****. ipk will be generated
In the end, I didn't figure out where the Configuring PACKAGEnlibgomp=y configuration item is valid

closed 2025-01-26 by hyjk555

badguy001, 2025-03-03:
https://github.com/openwrt/openwrt/tree/main/scripts/ext-toolchain.sh
It seems that libgomp.so only avalible when use external toolchain and gcc which in external toolchain had been enabled libgomp. package\libs\toolchain\Makefile will not install libgomp when not use external toolchain(control by CONFIG_EXTERNAL_TOOLCHAIN in Makefile). So if you want to use toolchain compiled from openwrt and you hope compile libgomp.so, you have to change src by youself which just like what you had did.