openwrt/openwrt #17715: WiFi Speed Banana BPI-R4 issue ( 24.10.0-rc6 )

KFOq, 2025-01-24:
### Describe the bug

This issue on all versions not solved until to now .. OpenWRT Snapshot , 24.10.0-rc2 , 24.10.0-rc3 , 24.10.0-rc4 , 24.10.0-rc5 ,24.10.0-rc6

**Issue :**
There is speeds limitations due to OpenWRT WIFI packages **doesn't support devices that has Qualcomm WCN6856 Wi-Fi Driver**

**On My Laptop (Alienware AMD 19) : 60mb/s**
WIFI- 160mhz (Qualcomm WCN6856 Wi-Fi Driver)

[image]

**On My Phone Samsung : 400 mb/s** and other laptops has not have Qualcomm WiFi 

[image]

Laptop WIFI Information

[Qualcomm WCN6856 Wi-Fi Driver](https://docs.qualcomm.com/bundle/publicresource/topics/80-WL542-10/overview.html)

[image]

I installed the Latest driver for WIFI on my computer but no improvement

Note: On BPI-R4 System that developed by BPI-R4 Company , I'm getting full speeds on WIFI for all devices ....

**###💯 Solution shared by OpenWRT members :** 
https://forum.openwrt.org/t/banana-bpi-r4-wifi7-status/201051/1065?u=kfo

- https://patchwork.kernel.org/project/linux-wireless/patch/20241112083846.32063-1-MeiChia.Chiu@mediatek.com/
- for mainline 24.10 repo, try to use this [patch](https://github.com/openwrt/openwrt/blob/8e359882eae462462a495443b3fc0bf2e443e340/package/kernel/mac80211/patches/subsys/400-mac80211-Support-EHT-1024-aggregation-size-in-T.patch)

[image]

### OpenWrt version

r28388-58d0057481

### OpenWrt release

24.10.0-rc6

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R4

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

hauke, 2025-01-26:
Do I understand it correctly that you use the Bananapi BPI-R4 with the wifi module from Bananapi which comes with a MT7995AV chip. When you connect to the AP opened by the MT7995AV with a Laptop with a Qualcomm WCN6856 wifi card it is slow?

@nbd168  Are you aware if this problem?

pmarques, 2025-01-27:
@KFOq are you also limited to 7dBm on 5Ghz? If yes, this is probably related to #17489

Pix13, 2025-01-28:
I'd like this issue fixed and included in the 24.10 final release too, as without it, the device is pretty much unuseable with all the Qualcomm clients

pgera, 2025-01-28:
I think this affects other mediatek based APs too. I am seeing this on wax220 AP with QCNFA765 client.

EPinci, 2025-01-30:
I'm not a committer but I can give this a try in my build.
However, @KFOq, the patch you mentioned at the beginning does not match the ones I see in rmandrad repo.
The original patch doesn't build in 24.10 ToT.

@rmandrad, did you deconstruct the original patch in your repo's 0015, 0018 and 0022?

rmandrad, 2025-01-30:
I had this patch applied last year but is now incorporated on mac80211 (at least the snapshot version) ... for instance the BA extension is now included on the header file

KFOq, 2025-01-31:
@EPinci 
I tried to build the image by that patch but I couldn't succuss  , 

I think as said Mr. @rmandrad (I had this patch applied last year but is now incorporated on mac80211 )

maybe you can try @romanovj  (his link) 
https://github.com/cmonroe/feed-wifi-master/blob/smartrg-master/mac80211/patches/subsys/mtk-0025-mac80211-add-EHT-BA1024-support.patch

i'm ready to test and share the feedback

danpawlik, 2025-01-31:
@KFOq could you share results when you switch to `AX` version (not BE as I guess you are using)?

EPinci, 2025-01-31:
> [@EPinci](https://github.com/EPinci) I tried to build the image by that patch but I couldn't succuss ,
>
> I think as said Mr. [@rmandrad](https://github.com/rmandrad) (I had this patch applied last [...]

I tried building with @romanovj patch but it fails against 24.10 ToT.
But... if that patch is already incorporated per @rmandrad ... why are we trying to build it in again?

KFOq, 2025-01-31:
@danpawlik 
My results based on **AX mode** not on BE mode ... also if I switched it to BE I will get very low speeds 

[image]

Refer to this subject as reference : I shared my comments 
https://forum.openwrt.org/t/banana-bpi-r4-wifi7-status/201051/1182?u=kfo

I tried MTK-SDK by one of openwrt members I got full speeds but not stable ... ( He said , he added the support of Qualcomm devices to OpenWRT system drivers = Patch  )
[image]

danpawlik, 2025-01-31:
@KFOq did you check using linux live distro eg. Fedora or Ubuntu, what results you will have?

EPinci, 2025-01-31:
@KFOq the patch you wanted to try patches a BE behavior, not AX.

KFOq, 2025-01-31:
> [@KFOq](https://github.com/KFOq) did you check using linux live distro eg. Fedora or Ubuntu, what results you will have?
@danpawlik 
I'm using Windows 11  Alienware R1 (Model: 2024) .. all drivers at latest update on Laptop ... 
same thins my PC using WiFi Qualcomm Chip getting very low speeds 

Note: 
- when I tried other laptop has intel wifi chip .. i'm getting full speeds on BE or AX
- I know how to build image on Linux Mint (Virtual Machine ) but still i'm noob 

----------------------------

> [@KFOq](https://github.com/KFOq) the patch you wanted to try patches a BE behavior, not AX.

@EPinci 
no issue if its BE or AX , at least I can see speeds above 100Mb/s ⬆️ 🥴

danpawlik, 2025-01-31:
@KFOq just take a pendrive, download newest Fedora linux workstation iso , use `rufus` tool and write the image to the pendrive.
After that, reboot Windows and boot your PC from the pendrive (you can try to press F10 or F12 during boot, or Google how to enter into UEFI menu/ Boot menu on Windows 11). That test will bring better view on the issue.

EPinci, 2025-02-01:
> [@EPinci](https://github.com/EPinci) sorry, for rc-7 minimum two patches are required from cmonroe feed

Yup, with those two I can build an image. It doesn't break any of my scenarios but I don't have any qualcomm device to test out the effectiveness of it.

KFOq, 2025-02-01:
@EPinci 

Please share with me the image so I can test them ..

danpawlik, 2025-02-01:
@KFOq you don't need to replace Windows or install Linux with Windows as dual boot. I just suggest to run live-cd - you will not install anything, it will just boot your laptop from the pendrive, run linux and thats it. Check on youtube what is livecd - there are many examples. Just don't install / replace your Windows.

KFOq, 2025-02-01:
@EPinci 

If you can share your build so I can try it ...

I added the the two files that mentioned by @romanovj  but I got filed to build ... can you guide me how you built the image 

- https://github.com/cmonroe/feed-wifi-master/blob/smartrg-master/mac80211/patches/subsys/mtk-0014-mac80211-fix-tx-amsdu-aggregation.patch

- https://github.com/cmonroe/feed-wifi-master/blob/smartrg-master/mac80211/patches/subsys/mtk-0025-mac80211-add-EHT-BA1024-support.patch

[image]

EPinci, 2025-02-01:
> Please share with me the image so I can test them ..

You can get the SD image and sysupgrade I built for you from [here](https://echopod.net/openwrt-mediatek-filogic-bananapi_bpi-r4_kfoq.zip): image has no config and is built from [openwrt-24.10 at current ToT](https://github.com/openwrt/openwrt/commit/c1d5de0c59efd25ecb5922cb11e5d38b072cf993).

KFOq, 2025-02-01:
@EPinci 

sadly the speeds still low 

[image]

EPinci, 2025-02-01:
Uhm, not sure what we're testing in here... remember that the patch you asked me to incorporate only affects BE.
You should test speeds from your laptop **to the router** not to the Internet that can get a gazilion things in the way.

Maybe you can use iperf3 or librespeedtest that are already baked in the image to test the actual wifi speed?
Or some more empirical file transfer test between clients on the same SSID?
Beacuse the image you posted shows the client connectet at about 2400mpbs...

KFOq, 2025-02-02:
> Maybe you can use iperf3 or librespeedtest that are already baked in the image to test the actual wifi speed?

@EPinci 
Yes I tested on all apps the actual speeds are 80Mb/s or lower  ...on Fast on OKAA and iperf3  

As I mentioned only one image gave me a good speeds 

This build by @woziwrt only gave me speeds above the speeds +100Mb/s ( **around 600Mb/s but the image is not stable** )
https://github.com/woziwrt/bpi-r4-mtk

tutugreen, 2025-02-12:
I have also encountered same issue, the speed is very slow under BE

https://forum.banana-pi.org/t/whats-the-best-firmware-for-bpi-r4-now-extreme-slow-wifi-7-speed-on-openwrt-snapshot/19528/5

If I create a WiFi 7 BE  AP with WPA3 , it will encounter serious speed issues.
If created as BE @ WPA2, it will display as WiFi 7 in scan, but WiFi 6 after connect. (on android client)

@KFOq You said your non-qualcomm laptop client is working properly, which model? like BE200? with WPA3 & negotiation rate of 2880?
﻿
All of my WiFi7 clients, including laptops (Qualcomm 7800), tablets (8Gen3), phones (8Elite), and headsets, are Qualcomm chipsets. I thought this router was slow for all clients in BE mode.
﻿
I also have a USB WiFi 7 tri band dongle (CF-983BE) based on Realtek chip, I will try it out to see if it works properly.

I have seen many replies/builds, and I have my vacation end last week. I will test them one by one these days to see if there are any improvements.

KFOq, 2025-02-15:
> If I create a WiFi 7 BE AP with WPA3 , it will encounter serious speed issues.
> If created as BE @ WPA2, it will display as WiFi 7 in scan, but WiFi 6 after connect. 

@tutugreen 
In my case all AX and BE at 80mhz and 160 mhz and 320 mhz .. i'm getting very low speeds around 70Mb/s 
Phones and Huaweii laptop and PC (that has not Qualcomm wifi chip) getting full speeds 

@danpawlik 
I run the linux system on my Laptop as you guided me what is next ... ( sorry for late respond I was traveling )

danpawlik, 2025-02-15:
@KFOq no problem. Have you done speedtest or iperf3 ? Are the results comparable or on linux are better/same/worse ?

KFOq, 2025-02-15:
> [@KFOq](https://github.com/KFOq) no problem. Have you done speedtest or iperf3 ? Are the results comparable or on linux are better/same/worse ?

@danpawlik 
Better ..... 

With Windows the speeds are limited at 70Mb/s 
I tried Fedora Workstation as you told me .. I got 

Pictures : 

[image]

[image]

[image]

I'm using the latest update for WIFI driver for this laptop on w Windows 11

danpawlik, 2025-02-15:
That's really something odd. Could you check also with iperf3 or install e.g. librespeed-go and check what is the result? 
If you don't know how to perform iperf3 or enable librespeed-go - let me know

KFOq, 2025-02-15:
@danpawlik 

I tested speeds on librespeed-go and speedtest and Fast 

On Windows  (getting low speeds) : 
[image]

On Fedora I'm getting a good speeds

tutugreen, 2025-02-16:
> [@danpawlik](https://github.com/danpawlik) To be honest , I don't know how to ,do speedtest with librespeed-go (I said to my self let me go to search how to do librespeed-go bettert than bother you to teach me ) ... so I asked Chatgpt and gave me this step and told me to install package librespeed-go .... after that gave me the page to test the speeds

If you have another PC, try run this server:

https://apps.microsoft.com/detail/9pbkqkr4x79r

and on client, use browser to get easy speedtest result.

KFOq, 2025-02-17:
@tutugreen (Thank you for guiding me) 

@danpawlik 
I tested the speeds on other apps as per the advice ...

**WiFi BE (6ghz 320mhz)** 
[image]

[image]

[image]

I didn't choose the server by my self (automatically) I tried muilti servers but still speeds are below 70Mb/s
[image]

**WiFi AX (5ghz 160mhz)** 

[image]

[image]

**This is my actual speeds on WIFI 6 and 5 ... when I used Mr @woziwrt  Image SDK **  

[image]

danpawlik, 2025-02-17:
@KFOq if you can not ping me each time...

still you are sharing speeds from the speedtest which are relying on your 5G ISP network (IIRC).

That's why I suggest to make a local test just to your router. For example:

* install `librespeed-go` and perform a test, for example:
```
<ssh to your router>
apk add librespeed-go
sed -i "s/enabled '0'/enabled '1'/g" /etc/config/librespeed-go
/etc/init.d/librespeed-go start
```
Then open in your browser: `http://<your router address>:8989/` and make a test

* do a `iperf3` test when you were running on fedora:
```
<ssh to your router>
apk add iperf3
iperf3 -s -D
```

then on your laptop, when you are using e.g. fedora, open a `console` and run:
```
iperf3 -c <your router ip address>
```

And share the results. Thanks.

KFOq, 2025-02-17:
> And share the results. 

I was putting your name because you asked me to do other test so I wanted you to see the results ... Anyhow  no ping 😏now 

**This with Official OpenWRT** 
[image]

**This with Mr.Woziwrt MTK-SDK image** 
[image]

--------------------------------------------------------------------------------------------------

tutugreen, 2025-02-18:
I've just did a quick test on OpenWRT 24.10 and Woziwrt's BPI-R4_MLO_beta_2_EXT_upd-1.2.zip too,
﻿
24.10:
8 Gen3 (Lenovo Y700 2025 Tablet TB321FU, FastConnect 7800), 5Ghz (6Ghz didn't try because firmware country code limit, try later) : slow speed 30~70Mbps
8 Elite (OnePlus Ace5Pro, FastConnect 7900), both 5Ghz , 6Ghz reach: ~1000Mbps+ (don't remember accurate, but it's a resonable number)

7988 CPU usage when 1000Mbps+: >80% high usage on one core
﻿
don't know why that much different.....
﻿
oziwrt's BPI-R4_MLO_beta_2_EXT_upd-1.2.zip:
﻿
8 Gen3 (Lenovo Y700 2025 Tablet TB321FU): 797Mbps DL, 1662.5 Mbps UL (much better than 24.10)
8 Elite (OnePlus Ace5Pro) 5Ghz: ~1.4Gbps DL/ 1709Mbps UL
8 Elite (OnePlus Ace5Pro) 6Ghz (320Mhz): ~ Avg 3.6Gbps DL, it Peak at 4.2Gbps, wow! 😀 (UL 1.9Gbps)

7988 CPU usage when 4.2Gbps: below 4% on all cores.
﻿
test using dedicated lan server run openspeedtest, connect to BPI R4 WAN SFP+, so it's the speed after NAT.
﻿
Tablet / Phone is placed just on the surface of BPI-R4 case, once moved, the speed decreases a lot.
﻿
for status shows on analiti, 4096QAM is used.
﻿
It's shows MLO support on ssid, but show connection speed as: 0Mbps (2.4Ghz) or 0Mbps (5Ghz),  1000+Mbps  (6Ghz) in android settings, so looks not actually aggererated. (the MLO SSID test speed is same as the pure 6ghz one).
﻿
indeed seen significant improvements on community firmware.

pmarques, 2025-02-18:
Thanks for sharing the results, I also did some [tests](https://github.com/openwrt/openwrt/issues/17489#issuecomment-2662942397) with 5Ghz and with a few drywalls and a cupboard in the mix, so far the results are a bit disappointing.

KFOq, 2025-02-26:
@nbd168 @hauke 
Please can you update WiFi drivers for OpenWRT ( Where the as I think the OpenWRT drivers for Wifi are old) because facing this issue 

https://forum.openwrt.org/t/banana-bpi-r4-wifi7-status/201051/1236?u=kfo

KFOq, 2025-03-15:
@nbd168 @hauke 

any support for this issue .. 

I hope if OpenWRT can extend their support to solve this issue ...

I tried the latest snapshot for this device and still the speeds are below 70Mb/s on all WIFI channels (6Ghz ,5Ghz and 2Ghz) 

The network speeds on system is around 680Mb/s and **WiFi 6ghz and 5Ghz below 70Mb/s** 
[image]

danpawlik, 2025-03-15:
@KFOq this is not firmware issue. You can replace firmware with the binaries that are in mediatek image and put it into `/lib/firmware/mediatek/mt7996/` and nothing will change. I got such device that has an issue on 6GHz network, when country is not set to US and it is using BE (for AX works better). From the other side, phones with Mediatek chipset works well - over 1.6 Gbs even on old firmware.
The community folks are doing what they can and they are supporting many devices, not BPi-R4 - it is not a main device here. Please be patient and wait for patches. 
Pinging people will not help, it even will make worse.

KFOq, 2025-03-17:
Dear @danpawlik  I avoided to ping you becuase you requested that , 
so for me Pinging the admins here is to bring focus to this issue where already highlighted by admin hauke here but no respond from 

> [@nbd168](https://github.com/nbd168) Are you aware if this problem?

I'm on this issue like a 5-7 months not solved ... its patch issue but its need to be added so this issue can be solved .

danpawlik, 2025-04-08:
@KFOq hey, so I have done few tests and the issue is not in the firmware, but in the hostapd parameters that are set in hostapd.
After making some comparison between mtk feed mac80211.sh [script](https://github.com/danpawlik/bpi-r4-mtk/blob/main/example/mac80211.sh) and openwrt pure [script](https://github.com/openwrt/openwrt/blob/main/package/network/config/wifi-scripts/files-ucode/lib/netifd/wireless/mac80211.sh) you will see, that there are many changes - especially for 6g.
I have done some PoC available [here](https://github.com/danpawlik/bpi-r4-mtk/blob/main/example/mac80211-my.sh) and from 60mbit now I have 1300. Still not perfect, but much better.

danpawlik, 2025-04-23:
@KFOq hey, finally got 5 min to make some tests.
I'm also affected with that bug on my OnePlus 12 when I enable `BE` mode, where on `AX` network works perfect.
So in short, I figured out, that after commenting:

```
#       if [ "$enable_be" != "0" ]; then
#               append base_cfg "ieee80211be=1" "$N"
#               [ "$hwmode" = "a" ] && {
#                       append base_cfg "eht_oper_chwidth=$eht_oper_chwidth" "$N"
#                       append base_cfg "eht_oper_centr_freq_seg0_idx=$eht_center_seg0" "$N"
#               }
#       fi
```
https://github.com/openwrt/openwrt/blob/main/package/network/config/wifi-scripts/files/lib/netifd/wireless/mac80211.sh#L518-L524

+ restart wifi interface (or reboot), I can get 1.4Gbit via download, but upload is near 300mbit (comparing to AX mode, where I can have 1.4Gbit its poor), but it is.. small step to figure out how to fix that :)

KFOq, 2025-04-24:
Hi @danpawlik 
sorry for late respond , I gave up to be honest with you .. I bought Xiamoi Mesh .. and I disabled BPI-R4 Wifi .... 

on my cause last time when only AX active ... i'm getting low speeds ... 
when only BE active .... i'm getting low speeds ...

not one router BPI-R4 ... 7 routers BPI-R4 I tried them has same issue lol 🤣

also I don't know from where to start to solve this issue ... but when I used Mr.Woziwrt image I didn't face this issue ..and Mr. Woziwrt he doesn't know which patch / file that on his image making BPI-R4 WiFi doing a good performance on Qualcomm devices ..

BTW I have Oneplus 13 ,, it the best phone ... I took it from Alibaba website ( Chiness version ) COS for 502$ dollars With shipping to my place ...I changed the system to Oxygen ..working a great 
its giving me a high speeds on WIFI ... better than any device ...

danpawlik, 2025-06-07:
Even with new mac80211 there is a problem.

closed 2025-09-27 by KFOq