openwrt/openwrt #17716: 24.10.0-rc6: ipq4019 - Kernel execption while booting

GoetzGoerisch, 2025-01-24:
### Describe the bug

24.10.0-rc6 dmesg output while booting:

```
[   34.582411] ath10k_ahb a800000.wifi: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
[   34.586256] ------------[ cut here ]------------
[   34.597251] WARNING: CPU: 1 PID: 902 at backports-6.12.6/net/mac80211/main.c:272 0xbf23385c [mac80211@8dad8409+0x77000]
[   34.602005] Modules linked in: nft_fib_inet nf_flow_table_inet ath10k_pci(O) ath10k_core(O) ath(O) nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mac80211(O) cfg80211(O) vpd_sysfs nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 memconsole_coreboot memconsole libcrc32c hwmon coreboot_table compat(O) sha512_arm ghash_arm_ce cmac leds_gpio ramoops reed_solomon pstore gpio_button_hotplug(O) f2fs ext4 mbcache jbd2 crc32c_generic crc32_generic
[   34.650004] CPU: 1 PID: 902 Comm: hostapd Tainted: G        W  O       6.6.73 #0
[   34.672185] Hardware name: Generic DT based system
[   34.679567] Function entered at [<c030d1d8>] from [<c0308ec8>]
[   34.684162] Function entered at [<c0308ec8>] from [<c0a3ff8c>]
[   34.689976] Function entered at [<c0a3ff8c>] from [<c032165c>]
[   34.695793] Function entered at [<c032165c>] from [<c03217dc>]
[   34.701608] Function entered at [<c03217dc>] from [<bf23385c>]
[   34.707425] Function entered at [<bf23385c>] from [<bf2339b0>]
[   34.713240] Function entered at [<bf2339b0>] from [<c0859960>]
[   34.719057] Function entered at [<c0859960>] from [<c0859dc4>]
[   34.724874] Function entered at [<c0859dc4>] from [<c0859e5c>]
[   34.730688] Function entered at [<c0859e5c>] from [<c0929d0c>]
[   34.736503] Function entered at [<c0929d0c>] from [<c092c564>]
[   34.742319] Function entered at [<c092c564>] from [<c082bc64>]
[   34.748138] Function entered at [<c082bc64>] from [<c04c6554>]
[   34.753952] Function entered at [<c04c6554>] from [<c0300040>]
[   34.759766] Exception stack(0xc3771fa8 to 0xc3771ff0)
[   34.765589] 1fa0:                   00000010 be959738 00000010 00008914 be959738 00000000
[   34.770714] 1fc0: 00000010 be959738 00000010 00000036 00000001 00000001 00000000 00000020
[   34.778872] 1fe0: 00663e60 be9595f8 00579bc8 b6ec966c
[   34.787077] ---[ end trace 0000000000000000 ]---

```

24.10.0-rc5 for reference:
```
[   41.332283] ath10k_ahb a800000.wifi: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
[   41.336133] ------------[ cut here ]------------
[   41.347109] WARNING: CPU: 1 PID: 1221 at backports-6.12.6/net/mac80211/main.c:272 0xbf23685c [mac80211@c669c40f+0x77000]
[   41.351878] Modules linked in: nft_fib_inet nf_flow_table_inet ath10k_pci(O) ath10k_core(O) ath(O) nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mac80211(O) cfg80211(O) vpd_sysfs nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 memconsole_coreboot memconsole libcrc32c hwmon coreboot_table compat(O) sha512_arm ghash_arm_ce cmac leds_gpio ramoops reed_solomon pstore gpio_button_hotplug(O) f2fs ext4 mbcache jbd2 crc32c_generic crc32_generic
[   41.399955] CPU: 1 PID: 1221 Comm: hostapd Tainted: G           O       6.6.69 #0
[   41.422151] Hardware name: Generic DT based system
[   41.429532] Function entered at [<c030d1d8>] from [<c0308ec8>]
[   41.434213] Function entered at [<c0308ec8>] from [<c0a3faec>]
[   41.440029] Function entered at [<c0a3faec>] from [<c032165c>]
[   41.445843] Function entered at [<c032165c>] from [<c03217dc>]
[   41.451660] Function entered at [<c03217dc>] from [<bf23685c>]
[   41.457484] Function entered at [<bf23685c>] from [<bf2369b0>]
[   41.463291] Function entered at [<bf2369b0>] from [<c08595ac>]
[   41.469109] Function entered at [<c08595ac>] from [<c0859a10>]
[   41.474922] Function entered at [<c0859a10>] from [<c0859aa8>]
[   41.480738] Function entered at [<c0859aa8>] from [<c092988c>]
[   41.486556] Function entered at [<c092988c>] from [<c092c0e4>]
[   41.492373] Function entered at [<c092c0e4>] from [<c082b93c>]
[   41.498187] Function entered at [<c082b93c>] from [<c04c65d0>]
[   41.504003] Function entered at [<c04c65d0>] from [<c0300040>]
[   41.509818] Exception stack(0xc50abfa8 to 0xc50abff0)
[   41.515639] bfa0:                   00000010 bead4828 00000010 00008914 bead4828 00000000
[   41.520766] bfc0: 00000010 bead4828 00000010 00000036 00000001 00000001 00000000 00000020
[   41.528924] bfe0: 00599e60 bead46e8 004afbb8 b6ec066c
[   41.537130] ---[ end trace 0000000000000000 ]---
```

### OpenWrt version

r28388-58d0057481

### OpenWrt release

24.10.0-rc6

### OpenWrt target/subtarget

ipq40xx/chromium

### Device

Google WiFi (Gale)

### Image kind

Official downloaded image

### Steps to reproduce

- install 24.10.0-rc6
- boot device
- look into dmesg output

### Actual behaviour

- kernel execptions during boot

### Expected behaviour

- no execptions

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

hnyman, 2025-01-25:
That is annoying but harmless.
As discussed earlier, the  classification of that informational message inside kernel driver has been changed so that it gets reported as a kernel error with the newer kernels.

For reference:

Duplicate of https://github.com/openwrt/openwrt/issues/15959#issuecomment-2259603978
Mentioned in https://github.com/openwrt/openwrt/pull/16036

GoetzGoerisch, 2025-01-25:
@hnyman thanks for your pointers.
Could and should this be fixed, although it is not an operational issue?

DragonBluep, 2025-04-01:
Fixed by 88234a03bc645d327016cc807c35972fcb6834eb. Please try snapshot version. And this ticket can be closed.

GoetzGoerisch, 2025-04-17:
Confirmed fixed with 24.10.1!

Thank you @DragonBluep

closed 2025-04-17 by GoetzGoerisch