openwrt/openwrt #17722: Built-in MMC card reader (still) does not function on 24.10.0-rc6

BorreSanders, 2025-01-24:
### Describe the bug

After power on and device start up built-in MMC card reader slot does not function, and (even) without SD card inserted error below is reported repeatedly. The same issue was also reported for rc4, see #17411.

### OpenWrt version

r28388-58d0057481

### OpenWrt release

24.10.0-rc6

### OpenWrt target/subtarget

ramips/mt76x8

### Device

RAVPower RP-WD009

### Image kind

Official downloaded image

### Steps to reproduce

1. Install 24.10.0-rc6 on RAVPower RP-WD009
2. Observe MMC errors in kernel and system logs.

### Actual behaviour

Builtin MMC card reader does not function, and repeatedly logs error -22.
It does not make a difference if I insert an SD card in the slot or not.

### Expected behaviour

Builtin MMC card reader does function, and logs no errors

### Additional info

https://github.com/openwrt/openwrt/pull/17446 discusses a collection of changes that could fix the problem. I suggest following up on the recommendation of [naseef](https://github.com/naseef) there.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2025-01-25:
Look at this: https://community.st.com/t5/stm32-mpus-products/mmci-pl18x-58005000-mmc-no-support-for-card-s-volts/td-p/86216

hauke, 2025-01-29:
Should be fixed in 1a75172721a1c917312105e9e0449fa431c0154c.

closed 2025-01-29 by hauke

BorreSanders, 2025-02-06:
Fix confirmed. Just installed the released version of 24.10.0, and SD card in MMC slot is functional again. Thanks all for your support for restoring the MMC card reader functionality for this device! For detailed proof, see attached kernel logging.

[kernel.log.24.10.0-released.txt](https://github.com/user-attachments/files/18688022/kernel.log.24.10.0-released.txt)

PR #17754 (2025-01-29): 24.10 - Fix missing backport from #17446

PR #17446 has total 6 commits, but only 5 are backported to 24.10 branch. As a result the MMC functionality is not working in 24.10 mt76x8 targets. This PR cherrypicks the missing commit from snapshot

@DragonBluep @hauke @BorreSanders

Fixes #17722