openwrt/openwrt #17723: in status -> realtime graphs -> connections the 'Enable DNS Lookup' does not seem to work

peterb99, 2025-01-25:
### Describe the bug

Enable DNS Lookup does not seem to work, only IP addresses shown

### OpenWrt version

r28388-58d0057481

### OpenWrt release

24.10.0-rc6

### OpenWrt target/subtarget

mediatek/filogic

### Device

OpenWrt One

### Image kind

Official downloaded image

### Steps to reproduce

go to status -> realtime graphs -> connections  and select 'Enable DNS Lookup' 

### Actual behaviour

Only IP addresses of source and targets shown no hostnames used to work in RC2

### Expected behaviour

IP addresses changing to hostnames for source and targets

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

hnyman, 2025-01-26:
Seems to work ok, with up-to-date main/master build.

[image]

peterb99, 2025-01-28:
Changed to following release:
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='24.10.0-rc6'
DISTRIB_REVISION='r28388-58d0057481'
DISTRIB_TARGET='mediatek/filogic'
DISTRIB_ARCH='aarch64_cortex-a53'
DISTRIB_DESCRIPTION='OpenWrt 24.10.0-rc6 r28388-58d0057481'

Now when press enable dns, it works, HOWEVER pressing disable and then Enable it does not work anymore