openwrt/openwrt #17727: WAX630 - netifd/hostapd can't modify bridge devices

Aelobin, 2025-01-25:
### Describe the bug

When choosing the network to attach to a wireless network (client mode), if that interface is using a bridge device then the wireless network fails to enable, with the following error:

```
Sat Jan 25 22:38:45 2025 daemon.notice netifd: radio0 (18265): WARNING: Variable 'data' does not exist or is not an array/object
Sat Jan 25 22:38:45 2025 daemon.notice hostapd: Set new config for phy phy0:
Sat Jan 25 22:38:45 2025 daemon.notice wpa_supplicant[1718]: Set new config for phy phy0
Sat Jan 25 22:38:46 2025 daemon.notice netifd: radio0 (18265): Interface 0 setup failed: BRIDGE_NOT_ALLOWED
Sat Jan 25 22:38:46 2025 daemon.notice wpa_supplicant[1718]: Set new config for phy phy0
Sat Jan 25 22:38:46 2025 daemon.notice hostapd: Set new config for phy phy0:
Sat Jan 25 22:38:46 2025 daemon.notice netifd: radio0 (18265): command failed: No such device (-19)
Sat Jan 25 22:38:46 2025 daemon.notice netifd: Wireless device 'radio0' is now up
```

And when trying to use the wpa_psk_file options to assign dynamic vlans, this similar error occurs:
```
Sat Jan 25 22:44:41 2025 daemon.notice netifd: radio2 (18832): WARNING: Variable 'data' does not exist or is not an array/object
Sat Jan 25 22:44:41 2025 daemon.notice hostapd: Set new config for phy phy2:
Sat Jan 25 22:44:41 2025 daemon.notice wpa_supplicant[1718]: Set new config for phy phy2
Sat Jan 25 22:44:41 2025 daemon.notice wpa_supplicant[1718]: Set new config for phy phy2
Sat Jan 25 22:44:41 2025 daemon.notice hostapd: Set new config for phy phy2: /var/run/hostapd-phy2.conf
Sat Jan 25 22:44:41 2025 daemon.notice hostapd: Restart interface for phy phy2
Sat Jan 25 22:44:41 2025 daemon.notice hostapd: Configuration file: data: driver=nl80211 logger_syslog=127 logger_syslog_level=2 logger_stdout=127 logger_stdout_level=2 country_code=GB ieee80211d=1 ieee80211h=1 hw_mode=a beacon_int=100 stationary_ap=1 chanlist=132 tx_queue_data2_burst=2.0 #num_global_macaddr=1 #macaddr_base= ieee80211n=1 ht_coex=0 ht_capab=[LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935][DSSS_CCK-40] ieee80211ac=1 vht_oper_chwidth=0 vht_oper_centr_freq_seg0_idx= vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][SU-BEAMFORMER][SU-BEAMFORMEE][MU-BEAMFORMER][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][SOUNDING-DIMENSION-4][BF-ANTENNA-4][MAX-MPDU-11454][MAX-A-MPDU-LEN-EXP7] channel=132  interface=wlan5 bssid=e0:46:ee:b5:ec:40 ctrl_interface=/var/run/hostapd bss_load_update_period=60 chan_util_avg_period=600 disassoc_low_ack=1 skip_inactivity_poll=0 preamble=1 wmm_enabled=1 ignore_broadcast_ssid=0 uapsd_advertisement_enabled=1 utf8_ssid=1 multi_ap=0 wpa_passphrase=passwordhere
Sat Jan 25 22:44:41 2025 daemon.notice hostapd: wlan5: interface state UNINITIALIZED->COUNTRY_UPDATE
Sat Jan 25 22:44:42 2025 daemon.err hostapd: Failed to create interface wlan5.108: -95 (Not supported)
Sat Jan 25 22:44:42 2025 daemon.err hostapd: VLAN: Could not add VLAN wlan5.108: No such device
Sat Jan 25 22:44:42 2025 daemon.err hostapd: VLAN initialization failed.
Sat Jan 25 22:44:42 2025 daemon.err hostapd: Interface initialization failed
Sat Jan 25 22:44:42 2025 daemon.notice hostapd: wlan5: interface state COUNTRY_UPDATE->DISABLED
Sat Jan 25 22:44:42 2025 daemon.notice hostapd: wlan5: AP-DISABLED
Sat Jan 25 22:44:42 2025 daemon.err hostapd: wlan5: Unable to setup interface.
Sat Jan 25 22:44:42 2025 daemon.notice hostapd: nl80211: deinit ifname=wlan5 disabled_11b_rates=0
Sat Jan 25 22:44:42 2025 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Sat Jan 25 22:44:42 2025 daemon.notice hostapd: wlan5: CTRL-EVENT-TERMINATING
Sat Jan 25 22:44:42 2025 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan5 wasn't started
Sat Jan 25 22:44:42 2025 daemon.notice hostapd: hostapd.add_iface failed for phy phy2 ifname=wlan5
Sat Jan 25 22:44:42 2025 daemon.notice netifd: Wireless device 'radio2' is now up
```

### OpenWrt version

r28388-58d0057481

### OpenWrt release

24.10.0-rc6

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Netgear WAX630

### Image kind

Official downloaded image

### Steps to reproduce

With the following configs, the above errors happen when trying to enable the wireless networks

/etc/config/network
```
config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'lan1'
	list ports 'lan2'

config bridge-vlan
	option device 'br-lan'
	option vlan '101'
	list ports 'lan1:t'

config bridge-vlan
	option device 'br-lan'
	option vlan '108'
	list ports 'lan1:t'

config device
	option type 'bridge'
	option name 'vlan.101'
	list ports 'br-lan.101'

config device
	option type 'bridge'
	option name 'vlan.108'
	list ports 'br-lan.108'

config interface 'iot'
	option proto 'dhcp'
	option device 'vlan.108'

config interface 'wwan'
	option proto 'none'
	option device 'vlan.101'
```

/etc/config/wireless
```
config wifi-device 'radio0'
	option type 'mac80211'
	option path 'platform/soc@0/c000000.wifi'
	option band '5g'
	option channel '52'
	option htmode 'HE80'
	option cell_density '0'
	option country 'GB'
	option disabled '1'

config wifi-device 'radio1'
	option type 'mac80211'
	option path 'platform/soc@0/c000000.wifi+1'
	option band '2g'
	option channel '6'
	option htmode 'HE20'
	option disabled '1'
	option country 'GB'
	option cell_density '0'

config wifi-device 'radio2'
	option type 'mac80211'
	option path 'platform/soc@0/c000000.wifi+2'
	option band '5g'
	option channel '132'
	option htmode 'VHT20'
	option country 'GB'
	option cell_density '0'
	option disabled '1'

config wifi-iface 'wifinet1'
	option device 'radio2'
	option mode 'ap'
	option ssid 'WifiWithVLANs'
	option encryption 'psk2'
	option key 'passwordhere'
	option ifname 'wlan5'
	option ieee80211r '1'
	option ft_over_ds '0'
	option ft_psk_generate_local '1'
	option wpa_psk_file '/etc/hostapd.wpa_psk'
	option vlan_file '/etc/hostapd.vlan5'
	option dynamic_vlan '2'
	option vlan_bridge 'vlan'
	option disabled '1'

config wifi-iface 'wifinet3'
	option device 'radio0'
	option mode 'sta'
	option bssid 'D8:EC:E5:3E:92:56'
	option ssid 'WifiAttachedToBridge'
	option encryption 'sae'
	option key 'passwordhere'
	option ocv '0'
	option network 'wwan'
	option disabled '1'
```

/etc/hostapd.wpa_psk
`vlanid=108 00:00:00:00:00:00 otherpasswordhere`

/etc/hostapd.vlan5
`108	wlan5.108	vlan.108`

### Actual behaviour

The wireless networks remain disabled when trying to bring them up, with errors in the System Log

### Expected behaviour

The Client Mode wireless network should bridge to the "wwan" interface (in the vlan.101 bridge, alongside the br-lan.101 device)

The Access Point wireless network should add a new wifi device (wlan5.101) to the vlan.101 bridge for clients connecting with the "otherpasswordhere" PSK

### Additional info

Apologies if I'm just doing something wrong, I'm still very new to OpenWrt!

I tested the same configs on another device (UniFi U6-LR v2) and both the wifi networks work as expected, so I'm unsure if there's something the WAX630 chip doesn't support?

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-01-26:
Wifi does not support vlan tags.

Aelobin, 2025-01-26:
> Wifi does not support vlan tags.

I'm not trying to send vlan tags over wifi 😃 My understanding is that tagged traffic entering via ethernet should get untagged by the vlan filtering on br-lan and sent to the corresponding br-lan.101 and br-lan.108 devices. Both those devices are part of their own bridges, vlan.101 and vlan.108, which are used as the interfaces for the wireless networks.

The vlan.101 bridge ("wwan" interface) should have wifinet3's device added to it when that wireless network is up. The vlan.108 bridge ("iot" interface) should have a device added to it (wlan.108) which would be created by the [wpa_psk_file](https://openwrt.org/docs/guide-user/network/wifi/basic#wpa_psk_file) options in wifinet1 (when up). Both of those wireless networks fail to start though, due to errors about creating/modifying the bridges. I may be doing something wrong (appreciate any corrections!), however this exact same config works like I described when used on another device 🤔

closed 2025-02-03 by Aelobin