openwrt/openwrt #17733: Zyxel PMG3000-D20B GPON SFP module vanishes after upgrade (21.02→24.10/snapshot)

twwn, 2025-01-26:
### Describe the bug

My Zyxel PMG3000-D20B SFP module doesn't come up with 24.10/snapshot. At all: no errors, nothing. I can only see it with `i2cdump`. In the OEM's 21.02 build, it's working properly.

This module is the fiber GPON one chosen by the Deutsche Telekom. That is, the module required for much of German's public fiber network.

My hardware is a Banana Pi BPI-R4 with its device-particular [soldering mod](https://forum.banana-pi.org/t/bpi-r4-zyxel-pmg3000-d20b-sfp-module-not-detected/18413/3). (It works around the device's SFP powering issue that clashes with the module's late EEPROM emulation. Other GPON modules [suffer from this](https://forum.banana-pi.org/t/bpi-r4-sfp-gpon-onu-34-20bi-openwrt-tx-fault/21810/1) as well, but afterwards at least throw an error in 24.10.)

All builds (that have the tool) can read from the module via `i2cdump`, so it does physically connect (iow, my soldering didn't break it):

```
i2cdump -y 3 0x50
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: 03 04 01 00 00 00 02 00 00 00 00 03 0c 00 14 c8    ???...?....??.??
10: 00 00 00 00 5a 59 58 45 4c 00 00 00 00 00 00 00    ....ZYXEL.......
20: 00 00 00 00 00 00 00 00 50 4d 47 33 30 30 30 2d    ........PMG3000-
30: 44 32 30 42 00 00 00 00 56 31 2e 30 05 1e 00 55    D20B....V1.0??.U
40: 00 1a 00 00 53 32 32 34 31 34 32 35 30 32 38 30    .?..S23267120070
50: 35 00 00 00 31 35 30 35 32 35 20 20 68 f0 01 9b    5...150525  h???
```

### OpenWrt version

7322d32

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek

### Device

Banana Pi BPI-R4

### Image kind

Self-built image

### Steps to reproduce

1. Install OpenWrt snapshot

### Actual behaviour

SFP module fails to appear.

### Expected behaviour

SFP module comes up & `ethtool -m` can evaluate it.

### Additional info

* ***OEM's (Sinovoip) [20240620 MTK MP4.0 build](https://docs.banana-pi.org/en/BPI-R4/BananaPi_BPI-R4#_system_image) (21.02 @ 5.4.271; [repo](https://github.com/BPI-SINOVOIP/BPI-R4-MT76-OPENWRT-V21.02)):***
Device is present:

```
dmesg | grep sfp
[   43.422277] sfp sfp@1: SM: enter empty:down:down event dev_up
[   43.428016] sfp sfp@1: SM: exit empty:up:down
[   43.539184] sfp sfp@0: SM: enter empty:down:down event dev_up
[   43.544925] sfp sfp@0: SM: exit empty:up:down
[   51.211299] sfp sfp@0: SM: enter empty:up:down event insert
[   51.216885] sfp sfp@0: SM: exit probe:up:down
[   51.527710] sfp sfp@0: SM: enter probe:up:down event timeout
[   51.544514] sfp sfp@0: module ZYXEL PMG3000-D20B rev V1.0 sn S23267120070 dc 150525
[   51.552432] sfp sfp@0: sfp: support mode 00,00000200,00006040
[   51.588241] sfp sfp@0: tx disable 1 -> 0
[... 251 lines cut ...]
CONFIG_PACKAGE_sysfsutils=y
CONFIG_PACKAGE_terminfo=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_umbim=y
CONFIG_PACKAGE_usbids=y
CONFIG_PACKAGE_usbutils=y
CONFIG_PACKAGE_wifischedule=y
CONFIG_PACKAGE_wireguard-tools=y
# CONFIG_PACKAGE_wpad-basic-mbedtls is not set
CONFIG_PACKAGE_wpad-openssl=y
CONFIG_PACKAGE_wsdd2=y
CONFIG_PACKAGE_wwan=y
CONFIG_PACKAGE_xfs-admin=y
CONFIG_PACKAGE_xfs-fsck=y
CONFIG_PACKAGE_xtables-nft=y
CONFIG_KERNEL_WERROR=y
# CONFIG_PACKAGE_kmod-crypto-kpp is not set
```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

rmandrad, 2025-01-26:
this may help you

https://forum.banana-pi.org/t/huawei-optixstar-s800e-xgspon-spf-onu-not-detected/19075/39

check glassdoor solution (entry 38)

twwn, 2025-01-26:
> this may help you
> 
> https://forum.banana-pi.org/t/huawei-optixstar-s800e-xgspon-spf-onu-not-detected/19075/39

Thanks. I just tried, for me it makes the R4 no longer boot properly (`Connection refused`):

```
diff --git i/package/base-files/files/etc/rc.local w/package/base-files/files/etc/rc.local
index 5639477..55edce7 100644
--- i/package/base-files/files/etc/rc.local
+++ w/package/base-files/files/etc/rc.local
@@ -1,4 +1,7 @@
 # Put your custom commands here that should be executed once
 # the system init finished. By default this file does nothing.
 
+echo 594 > /sys/class/gpio/export
+echo out > /sys/class/gpio/gpio594/direction
+echo 0 > /sys/class/gpio/gpio594/value
 exit 0
diff --git i/target/linux/mediatek/files-6.6/arch/arm64/boot/dts/mediatek/mt7988a-bananapi-bpi-r4.dtsi w/target/linux/mediatek/files-6.6/arch/arm64/boot/dts/mediatek/mt7988a-bananapi-bpi-r4.dtsi
index 8dba5b4..cd85f0e 100644
--- i/target/linux/mediatek/files-6.6/arch/arm64/boot/dts/mediatek/mt7988a-bananapi-bpi-r4.dtsi
+++ w/target/linux/mediatek/files-6.6/arch/arm64/boot/dts/mediatek/mt7988a-bananapi-bpi-r4.dtsi
@@ -36,18 +36,6 @@
 		reg = <0x00 0x40000000 0x00 0x10000000>;
 	};
 
-	/* SFP1 cage (WAN) */
-	sfp1: sfp1 {
-		compatible = "sff,sfp";
-		i2c-bus = <&i2c_sfp1>;
-		los-gpios = <&pio 54 GPIO_ACTIVE_HIGH>;
-		mod-def0-gpios = <&pio 82 GPIO_ACTIVE_LOW>;
-		tx-disable-gpios = <&pio 70 GPIO_ACTIVE_HIGH>;
-		tx-fault-gpios = <&pio 69 GPIO_ACTIVE_HIGH>;
-		rate-select0-gpios = <&pio 21 GPIO_ACTIVE_LOW>;
-		maximum-power-milliwatt = <3000>;
-	};
-
 	gpio-keys {
 		compatible = "gpio-keys";
 
@@ -86,11 +74,9 @@
 };
 
 &gmac2 {
-	sfp = <&sfp1>;
 	managed = "in-band-status";
-	phy-mode = "usxgmii";
+	phy-mode = "10gbase-r";
 	status = "okay";
-	openwrt,netdev-name = "sfp-wan";
 };
 
 &switch {
```

twwn, 2025-01-26:
> Invalid Target/Subtarget reported. `mediatek` Is this from a supported device?

fixed

> Invalid Version reported. `7322d32` Is this from a clean repository?

It is, I just cloned with `--depth=1`, which makes the pseudo-SVN count fail. Please remove the invalid label!

hauke, 2025-01-27:
The fix in `drivers/net/phy/sfp.c` looks good. The SFP runs Linux with OpenWrt which needs to boot up before the EEPROM is available. 
Is the 2.5G connection to the SFP working? 

Please create a PR and also send this to the upstream Linux kernel.

twwn, 2025-01-27:
It's not a fix, it doesn't actually help with anything. It's just what I tried for not knowing better. I don't know what's wrong or how to proceed.

This particular module only works in 1G mode by default (there is [a way](https://github.com/xvzf/zyxel-gpon-sfp) to enable 2.5G, but it requires access to the module's SSH interface and manual negotiation afterwards).

The breakage is somewhere directly in the kernel. [On the BPI forums](https://forum.banana-pi.org/t/bpi-r4-zyxel-pmg3000-d20b-sfp-module-not-detected/18413/52), @BtBN reports that with [Wunderlich's kernel](https://github.com/frank-w/BPI-Router-Linux/) (and Gentoo) the module worked out-of-the-box (after said soldering) since at least kernel 6.9. But again, most of Wunderlich's work seems upstreamed only with 6.13.

twwn, 2025-01-27:
I've now built OpenWrt from the final main commit that allowed kernel 6.1(.89) for mediatek, r0+26156-[fb2475e6bd](https://github.com/openwrt/openwrt/commit/fb2475e6bd).

It it, the module is very much still present. `ethtool -m eth2` (this is before the renaming to `sfp-wan`) reports its full info.

Mostly. The module seems to be in a kind of reset loop, and I'm unsure what's the cause (the module itself, my soldering, or the kernel).

Occasionally `ethtool` (patched to 6.11) will break off its readout halfway through and instead go

```
[…]
	Vendor SN                                 : S23267120070___
	Date code                                 : 150525

netlink error: Operation timed out
Failed to read Page A2h.
root@OpenWrt:~#  ethtool -m eth2
netlink error: Operation timed out
```

I'll try to build later commits when I've got more time.

I do suggest somebody to at least look at this and confirm that other devices on 24.10 have no issues with this module. Otherwise, 24.10 gets released an quite a few people may find themselves knocked off the internet after updating.

Would somebody **please** remove the invalid label‽

twwn, 2025-01-30:
@hauke @dangowrt

I got my hands on 3 Huawei-branded SFP modules and checked them with both my modified & unmodified SFP ports:

* **WTD** RTXM228-402 (SFP-10G-1310nm-10km)
* **Hisense** LTD1302-BC+1 (1.25G)
* **HG GENUINE** MXPD-243S (1.25G)

__Sinovoip 21.02 5.4.271 MTK:__ Instantly recognized, `ethtool -m` data complete

__OpenWrt @c87a7678:__ Reported as such in `dmesg`, `ethtool -m` data just hex mess

So unlike the Zyxel, they at least show, but are still unusable.

Is it safe to say that SFP (& wifi) on OpenWrt for this device will just be a broken mess, at least with 24.10?

dangowrt, 2025-01-31:
We may have to downgrade `ethtool` or further investigate why dumping the EEPROM isn't working there. Otherwise SFP(+) all work very well on the BPi-R4, I'm using an XGSPON SFP and DAC cable in production, and have tested various RJ-45 SFP(+) modules.

The main difference with the Sinovoip firmware is that OpenWrt actually handles the LOS and TX_FAULT signals and lets the Linux kernel parse the EEPROM, while Sinovoip sets the port to a fixed speed and relies exclusively on in-band signalling, letting Linux ignore SFP EEPROM and out-of-band signals. This means that some SFP modules will require setting up kernel quirks in order to work with OpenWrt, while with Sinovoip firmware only 10G SFPs with in-band signalling work while 1G and 2.5G modules cannot work at all.

twwn, 2025-01-31:
Might be less effort to do the kernel 6.12 update, add the upstreamed 6.14 fixes, and go from there, though.

Anyway, there's one report of the Zyxel [apparently working](https://forum.banana-pi.org/t/bpi-r4-zyxel-pmg3000-d20b-sfp-module-not-detected/18413/32) on a September snapshot (that is, he just [mentioned](https://forum.banana-pi.org/t/bpi-r4-zyxel-pmg3000-d20b-sfp-module-not-detected/18413/16) using snapshot earlier). But for me, nothing at all past `i2cdump`. I'll try the Luleey LL-XS2510 next.

twwn, 2025-01-31:
@dangowrt I do indeed have one installed (WD Blue SN580).

But from looking into this just (and only) now, I also have a V1.1 board ([photo](https://i.imgur.com/RK64Q0d.jpeg)), and from [this one picture](https://i.imgur.com/PcSstlD.jpeg) I snapped before installation, the [relevant troublesome resistors](https://forum.banana-pi.org/t/bpi-r4-nvme-i2c/17152/8) aren't present on it?

Should I try with the SSD uninstalled? (I will if it's relevant, I'm just weary of the installation and ripping out the wifi thermal pads for the n-th time.)

twwn, 2025-01-31:
@dangonwrt

Never mind, I went ahead anyway.

That 86251a7 and yesterday's c87a767 build. No 5G modem, no SSD: same broken result.

dangowrt, 2025-01-31:
If R228 and R230 are not present on your board then you are not affected by the issue, and your picture shows they are not populated, so there is no need to test without the SSD. The issue has to be something else.

Can you try the GPON SFP in a simple switch or media converter for starters and make sure that
 - it runs up-to-date firmware
 - it is configured for 2500Base-X mode
 
Did I understand correctly that the module is not detected at all during boot, ie. no trace of it in `dmesg`?

twwn, 2025-01-31:
I don't have another device with SFP here, unfortunately.

I also just checked with snapshot from the official firmware builder (again c87a767801), as well as using the OEM-supplied power adapter instead of USB-PD… again nothing. Except I now know to never use that adapter again as it *very* loudly  whistled.

> * it is configured for 2500Base-X mode

IIRC the Zyxel does 2.5G (instead of 1G) only unofficially, and only if set manually, then needing to further disable auto-negotiation.

> Did I understand correctly that the module is not detected at all during boot, ie. no trace of it in `dmesg`?

Yes. No trace, except on Sinovoip's 5.4 build.

twwn, 2025-01-31:
@dangowrt 

I'd have liked to access the module via ssh/webinterface to get its actual firmware version, using [the wiki](https://openwrt.org/docs/guide-user/network/wan/access.modem.through.nat) and its [Hack GPON entry](https://hack-gpon.org/ont-zyxel-pmg3000-d20b/), but it doesn't want to be reached?

Can you please check whether I doing this wrong in the R4?

```
root@OpenWrt:~# uci show network.modem
network.modem=interface
network.modem.proto='static'
network.modem.ipaddr='10.10.1.2'
network.modem.netmask='255.0.0.0'
network.modem.device='eth2'
root@OpenWrt:~# uci show firewall.@zone[1].network
firewall.cfg03dc81.network='wan' 'wan6' 'WWAN' 'WWAN6' 'WWAN_Q' 'WWAN6_Q' 'modem'
root@OpenWrt:~# uci commit network; service network restart; uci commit firewall; service firewall restart
[…]
root@OpenWrt:~# ssh admin@10.10.1.1

ssh: Connection to admin@10.10.1.1:22 exited: Connect failed: Host is unreachable
root@OpenWrt:~# ping 10.10.1.1
PING 10.10.1.1 (10.10.1.1): 56 data bytes
^C
--- 10.10.1.1 ping statistics ---
20 packets transmitted, 0 packets received, 100% packet loss
root@OpenWrt:~# uname -a
Linux OpenWrt 5.4.271 #0 SMP Wed Jun 5 06:11:29 2024 aarch64 GNU/Linux
root@OpenWrt:~# ethtool -m eth2
[can post full version if of interest]
	Transceiver type                          : Ethernet: 1000BASE-LX
	Vendor name                               : ZYXEL___________
	Vendor OUI                                : 00:00:00
	Vendor PN                                 : PMG3000-D20B____
	Vendor rev                                : V1.0
```

twwn, 2025-02-01:
Took it to a Unifi switch (ancient 3.18.24 kernel), [this is what showed](https://i.imgur.com/9DmO4Lf.png). Unlike the Huawai sticks, the switch detected some host in the Zyxel.

Unfortunately I couldn't ssh into the stick, no IP to the interface didn't work and for setting one, the Unifi controller demands squatting on the .1 one.

dangowrt, 2026-01-28:
> I gues it must be RX side of the module then if its really UART on those pins.

Nah, that must be some other probablem, LOS usually isn't used for debug UART.

AreYouLoco, 2026-01-30:
I am trying to investigate with some user from TurrisOS forum. Ignoring tx_fault and los is not enough. If someone is able to build a kernel like OP. I would try this quirk set at the same time

https://github.com/torvalds/linux/blob/4d310797262f0ddf129e76c2aad2b950adaf1fda/drivers/net/phy/sfp.c#L426

mosterta, 2026-08-28:
I had the same issue and improved the patch a bit. I get the ZYXEL PMG3000-D20B up and running with the following patch based on OpenWrt 25.12.05

```
diff --git a/drivers/net/phy/sfp.c b/drivers/net/phy/sfp.c
index 0000000..0000000 100644
--- a/drivers/net/phy/sfp.c
+++ b/drivers/net/phy/sfp.c
@@
 static void sfp_fixup_zyxel(struct sfp *sfp)
 {
+	/*
+	 * Workaround for the ZYXEL PMG3000-D20B:
+	 * the module exposes a valid PHY, but its LOS/TX_FAULT state is noisy
+	 * during early startup. Ignore the false transitions and keep the
+	 * longer module startup window.
+	 */
     sfp_fixup_ignore_los(sfp);
     sfp_fixup_ignore_tx_fault(sfp);
     sfp->module_t_wait = T_START_UP_BAD_GPON;
 }
@@
 static void sfp_quirk_zyxel_linkmode(const struct sfp_eeprom_id *id,
                    struct sfp_module_caps *caps)
 {
+	/*
+	 * The module reports a noisy link mode mask during startup; for the
+	 * ZYXEL PMG3000-D20B we only disable autonegotiation and leave the
+	 * actual link negotiation to the PHY.
+	 */
     sfp_quirk_disable_autoneg(id, caps);
 }
@@
+	SFP_QUIRK("ZYXEL", "PMG3000-D20B", sfp_quirk_zyxel_linkmode, sfp_fixup_zyxel),
 };

```

LorbusChris, 2026-09-09:
I got it to work on Turris OS with the changes from https://github.com/turris-cz/os-build/pull/19, those probably also apply to stock OpenWrt