openwrt/openwrt #17735: IPv6-PD update fails with dynamic IPv6 address on Fritz!Box

bdrung, 2025-01-26:
### Describe the bug

My provider gives me a dynamic IPv6 address (dynamic /56 prefix), which has a lease time of one hour. The Internet facing router is a FRITZ!Box 7590 with FRITZ!OS 7.59 (latest version). The WAN port of the OpenWRT router is connected to FRITZ!Box. The OpenWRT router (even in the stock configuration) gets an IPv6 address and IPv6-PD. Example output from Network -> Interfaces -> wan6:
```
Protocol: DHCPv6 client
Uptime: 0h 20m 36s
MAC: D4:6E:0E:65:AD:EB
RX: 12.51 MB (18678 Pkts.)
TX: 2.15 MB (12848 Pkts.)
IPv6: 2a00:1f:9b01:af01:<redacted>/128
IPv6: fd<redacted>/64
IPv6-PD: 2a00:1f:9b01:aff0::/60
````

When the FRITZ!Box gets a new IPv6 prefix or when I force a reconnect, the IPv6-PD on the OpenWRT router is not updated. Example output after a force reconnect:

```
Protocol: DHCPv6 client
Uptime: 0h 22m 26s
RX: 12.66 MB (19300 Pkts.)
TX: 2.25 MB (13379 Pkts.)
IPv6: 2a00:1f:9b01:af01:<redacted>/128
IPv6: fd<redacted>/64
IPv6: 2a00:1f:9b01:e301:<redacted>/64
IPv6-PD: 2a00:1f:9b01:aff0::/60
```

Updating from OpenWRT 23.05.5 to 24.10.0-rc6 did not change anything. Is this a bug in OpenWRT or in FRITZ!OS instead?

### OpenWrt version

r28388-58d0057481

### OpenWrt release

24.10.0-rc6

### OpenWrt target/subtarget

ipq806x/generic

### Device

TP-Link Archer C2600

### Image kind

Official downloaded image

### Additional info

My issue seems to be similar to bug #9689 (which seemed to be a FRITZ!OS bug that has been fixed since then).

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

bdrung, 2025-01-26:
Looks like the github action failed to extract the data.

hauke, 2025-01-27:
I think I had the same problem. I think the fritzbox did not send a `RECONFIGURE` message when it got a new prefix. I got a new IPv6 prefix every day and the prefix delegation broke after 1 day. The linked ticket also says that AVM is working on a fix on their side which will not be added to 7.29. Maybe you need a more recent FritzOS version.

bdrung, 2025-01-28:
I reported this issue to AVM today.

Then I noticed that AVM released FRITZ!OS 8. I upgraded the FRITZ!Box 7590 to FRITZ!OS 8.02, but that version shows the same issue.

bdrung, 2025-02-10:
I tried to dump the DHCP traffic, but that failed:

```
root@router:~# tcpdump -evni eth0 upd port 67
tcpdump: unsupported DSA tag: qca
```

So I followed https://openwrt.org/docs/guide-user/network/protocol.dhcp and added DHCP client scripts:

```sh
cat << "EOF" > /etc/odhcp6c.user.d/00-logger
logger -t ${0##*/} ${@} $(env)
EOF
```

The new logs on reconnecting the Fritz!Box:

```
Sun Feb  9 12:33:26 2025 user.notice dhcpv6.script: wan ra-updated J_V_routes6=J_A5 K_J_T6= target netmask metric valid T_J_T8_target=string T_J_T10_valid=int T_J_T11_metric=int RA_ADDRESSES=2a00:1f:9b03:7401:<redacted>/64,3141,6741 fd<redacted>/64,3599,7199 RA_REACHABLE=0 T_J_A2_1=object J_T6_netmask=64 K_J_T7= target netmask metric valid T_J_T9_target=string T_J_T11_valid=int T_J_T1_passthru=string T_J_A2_2=object J_T7_netmask=64 K_J_T8= target netmask gateway metric valid source J_T8_gateway=<redacted> T_J_T10_target=string J_V_ip6prefix=J_A12 CER= PASSTHRU=<redacted> J_T8_netmask=56 K_J_T9= target netmask gateway metric valid source J_T9_gateway=<redacted> T_J_T11_target=string T_J_A5_1=object J_T9_netmask=56 K_J_T10= target netmask gateway metric valid source J_T10_gateway=<redacted>
Sun Feb  9 12:33:30 2025 user.notice dhcpv6.script: wan ra-updated J_V_routes6=J_A5 K_J_T6= target netmask metric valid T_J_T8_target=string T_J_T10_valid=int T_J_T11_metric=int RA_ADDRESSES=2a00:1f:9b03:7401:<redacted>/64,0,7200 fd<redacted>/64,3600,7200 RA_REACHABLE=0 T_J_A2_1=object J_T6_netmask=64 K_J_T7= target netmask metric valid T_J_T9_target=string T_J_T11_valid=int T_J_T12_metric=int T_J_T1_passthru=string T_J_A2_2=object J_T7_netmask=64 K_J_T8= target netmask gateway metric valid source J_T8_gateway=<redacted> T_J_T10_target=string T_J_T12_valid=int J_V_ip6prefix=J_A13 CER= PASSTHRU=<redacted> J_T8_netmask=56 K_J_T9= target netmask gateway metric valid source J_T9_gateway=<redacted> T_J_T11_target=string T_J_A5_1=object J_T9_netmask=56 K_J_T10= target netmask gateway metric vali
Sun Feb  9 12:33:32 2025 user.notice dhcpv6.script: wan ra-updated J_V_routes6=J_A6 T_J_T8_target=string T_J_T10_valid=int T_J_T11_metric=int T_J_T17_netmask=string T_J_T18_gateway=string T_J_T20_metric=int RA_ADDRESSES=2a00:1f:9b03:7401:<redacted>/64,0,7198 fd<redacted>/64,3598,7198 2a00:1f:9b01:f201:<redacted>/64,3598,7198 RA_REACHABLE=0 T_J_A2_1=object K_J_T7= target netmask metric valid T_J_T9_target=string T_J_T11_valid=int T_J_T12_metric=int T_J_T18_netmask=string T_J_T20_valid=int T_J_T21_metric=int T_J_T1_passthru=string T_J_A2_2=object J_T7_netmask=64 K_J_T8= target netmask metric valid T_J_T10_target=string T_J_T12_valid=int T_J_T13_metric=int T_J_T19_netmask=string T_J_T21_valid=int T_J_T22_metric=int J_V_ip6prefix=J_A24 CER= PASSTHRU=<redacted> T_J_A2_3=object J_T8_netmask=64 K_J_T9= target netmask
Sun Feb  9 12:33:34 2025 user.notice dhcpv6.script: wan ra-updated J_V_routes6=J_A6 T_J_T8_target=string T_J_T10_valid=int T_J_T11_metric=int T_J_T17_netmask=string T_J_T18_gateway=string T_J_T20_metric=int RA_ADDRESSES=2a00:1f:9b03:7401:<redacted>/64,0,7196 fd<redacted>/64,3599,7199 2a00:1f:9b01:f201:<redacted>/64,3596,7196 RA_REACHABLE=0 T_J_A2_1=object K_J_T7= target netmask metric valid T_J_T9_target=string T_J_T11_valid=int T_J_T12_metric=int T_J_T18_netmask=string T_J_T20_valid=int T_J_T21_metric=int T_J_T1_passthru=string T_J_A2_2=object J_T7_netmask=64 K_J_T8= target netmask metric valid T_J_T10_target=string T_J_T12_valid=int T_J_T13_metric=int T_J_T19_netmask=string T_J_T21_valid=int T_J_T22_metric=int J_V_ip6prefix=J_A24 CER= PASSTHRU=<redacted> T_J_A2_3=object J_T8_netmask=64 K_J_T9= target netmask
```

The ip config afterwards:

```
8: wan@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP qlen 1000
    link/ether <redacted>
    inet <redacted> scope global wan
       valid_lft forever preferred_lft forever
    inet6 2a00:1f:9b01:f201:<redacted>/64 scope global dynamic noprefixroute 
       valid_lft 7136sec preferred_lft 3536sec
    inet6 fd<redacted>/64 scope global dynamic noprefixroute 
       valid_lft 7139sec preferred_lft 3539sec
    inet6 2a00:1f:9b03:7401:<redacted>/128 scope global dynamic noprefixroute 
       valid_lft 6995sec preferred_lft 3395sec
    inet6 fe80::<redacted>/64 scope link 
       valid_lft forever preferred_lft forever
```

bdrung, 2025-04-20:
After an update to OpenWRT 24.10.1 and upgrading the Fritz!Box to FRITZ!OS 8.03, I was able to collect logs with tcpdump. The Fritz!Box had the prefix 2a00:1f:9b01:a400::/56 and OpenWRT -> Network -> Interface showed IPv6-PD: 2a00:1f:9b01:a4f0::/60 on wan6.

```
root@router:~# ip -6 a s wan
8: wan@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP qlen 1000
    link/ether d4:6e:0e:<redacted router> brd ff:ff:ff:ff:ff:ff
    inet6 fd11::d66e:eff:fe<redacted router>/64 scope global dynamic noprefixroute 
       valid_lft 6733sec preferred_lft 3133sec
    inet6 2a00:1f:9b01:a401:d66e:eff:fe<redacted router>/128 scope global dynamic noprefixroute 
       valid_lft 6220sec preferred_lft 2620sec
    inet6 fe80::d66e:eff:fe<redacted router>/64 scope link 
       valid_lft forever preferred_lft forever
```

Then I started tcpdump and then let the Fritz!Box reconnected:

```
root@router:~# tcpdump -evni wan 'port 546 or port 547 or (icmp6 and ip6[40] = 134)'
14:09:15.737039 3c:a6:2f:<redacted fritzbox> > 33:33:00:00:00:01, ethertype IPv6 (0x86dd), length 254: (hlim 255, next-header ICMPv6 (58) payload length: 200) fe80::3ea6:2fff:fe<redacted fritzbox> > ff02::1: [icmp6 sum ok] ICMP6, router advertisement, length 200
	hop limit 255, Flags [other stateful], pref medium, router lifetime 0s, reachable time 0ms, retrans timer 0ms
	  prefix info option (3), length 32 (4): 2a00:1f:9b01:8001::/64, Flags [onlink, auto], valid time 0s, pref. time 0s
	  prefix info option (3), length 32 (4): 2a00:1f:9b01:a401::/64, Flags [onlink, auto], valid time 6070s, pref. time 2470s
	  prefix info option (3), length 32 (4): fd11::/64, Flags [onlink, auto], valid time 7200s, pref. time 3600s
	  rdnss option (25), length 40 (5):  lifetime 1200s, addr: fd11::3ea6:2fff:fe<redacted fritzbox> addr: 2a00:1f:9b01:a401:3ea6:2fff:fe<redacted fritzbox>
	  mtu option (5), length 8 (1):  1492
	  route info option (24), length 16 (2):  2a00:1f:9b01:a400::/56, pref=medium, lifetime=1800s
	  route info option (24), length 16 (2):  fd11::/64, pref=medium, lifetime=1800s
	  source link-address option (1), length 8 (1): 3c:a6:2f:<redacted fritzbox>
14:09:20.440054 3c:a6:2f:<redacted fritzbox> > 33:33:00:00:00:01, ethertype IPv6 (0x86dd), length 238: (hlim 255, next-header ICMPv6 (58) payload length: 184) fe80::3ea6:2fff:fe<redacted fritzbox> > ff02::1: [icmp6 sum ok] ICMP6, router advertisement, length 184
	hop limit 255, Flags [other stateful], pref medium, router lifetime 0s, reachable time 0ms, retrans timer 0ms
	  prefix info option (3), length 32 (4): 2a00:1f:9b01:8001::/64, Flags [onlink, auto], valid time 0s, pref. time 0s
	  prefix info option (3), length 32 (4): 2a00:1f:9b01:a401::/64, Flags [onlink, auto], valid time 6065s, pref. time 0s
	  prefix info option (3), length 32 (4): fd11::/64, Flags [onlink, auto], valid time 7200s, pref. time 3600s
	  rdnss option (25), length 40 (5):  lifetime 1200s, addr: fd11::3ea6:2fff:fe<redacted fritzbox> addr: 2a00:1f:9b01:a401:3ea6:2fff:fe<redacted fritzbox>
	  mtu option (5), length 8 (1):  1492
	  route info option (24), length 16 (2):  fd11::/64, pref=medium, lifetime=1800s
	  source link-address option (1), length 8 (1): 3c:a6:2f:<redacted fritzbox>
14:09:20.445081 3c:a6:2f:<redacted fritzbox> > 33:33:00:00:00:01, ethertype IPv6 (0x86dd), length 230: (hlim 255, next-header ICMPv6 (58) payload length: 176) fe80::3ea6:2fff:fe<redacted fritzbox> > ff02::1: [icmp6 sum ok] ICMP6, router advertisement, length 176
	hop limit 255, Flags [managed, other stateful], pref medium, router lifetime 1800s, reachable time 0ms, retrans timer 0ms
	  prefix info option (3), length 32 (4): 2a00:1f:9b01:a401::/64, Flags [onlink, auto], valid time 7200s, pref. time 3600s
	  prefix info option (3), length 32 (4): fd11::/64, Flags [onlink, auto], valid time 7200s, pref. time 3600s
	  rdnss option (25), length 40 (5):  lifetime 1200s, addr: fd11::3ea6:2fff:fe<redacted fritzbox> addr: 2a00:1f:9b01:a401:3ea6:2fff:fe<redacted fritzbox>
	  mtu option (5), length 8 (1):  1492
	  route info option (24), length 8 (1):  ::/0, pref=medium, lifetime=1800s
	  route info option (24), length 16 (2):  2a00:1f:9b01:a401::/64, pref=medium, lifetime=1800s
	  route info option (24), length 16 (2):  fd11::/64, pref=medium, lifetime=1800s
	  source link-address option (1), length 8 (1): 3c:a6:2f:<redacted fritzbox>
14:09:20.451413 3c:a6:2f:<redacted fritzbox> > 33:33:00:00:00:01, ethertype IPv6 (0x86dd), length 206: (hlim 255, next-header ICMPv6 (58) payload length: 152) fe80::3ea6:2fff:fe<redacted fritzbox> > ff02::1: [icmp6 sum ok] ICMP6, router advertisement, length 152
	hop limit 255, Flags [other stateful], pref medium, router lifetime 0s, reachable time 0ms, retrans timer 0ms
	  prefix info option (3), length 32 (4): 2a00:1f:9b01:a401::/64, Flags [onlink, auto], valid time 7200s, pref. time 0s
	  prefix info option (3), length 32 (4): fd11::/64, Flags [onlink, auto], valid time 7200s, pref. time 3600s
	  rdnss option (25), length 40 (5):  lifetime 1200s, addr: fd11::3ea6:2fff:fe<redacted fritzbox> addr: 2a00:1f:9b01:a401:3ea6:2fff:fe<redacted fritzbox>
	  mtu option (5), length 8 (1):  1492
	  route info option (24), length 16 (2):  fd11::/64, pref=medium, lifetime=1800s
	  source link-address option (1), length 8 (1): 3c:a6:2f:<redacted fritzbox>
[... cut ...]
    inet6 fe80::d66e:eff:fe<redacted router>/64 scope link 
       valid_lft forever preferred_lft forever
```

But OpenWRT -> Network -> Interface still says for wan6:

```
Protocol: DHCPv6 client
Uptime: 0h 20m 13s
MAC: D4:6E:0E:65:AD:EB
RX: 154.91 MB (196367 Pkts.)
TX: 35.92 MB (147801 Pkts.)
IPv6: 2a00:1f:9b01:a401:d66e:eff:fe<redacted router>/128
IPv6: fd11::d66e:eff:fe<redacted router>/64
IPv6: 2a00:1f:9b01:b501:d66e:eff:fe<redacted router>/64
IPv6-PD: 2a00:1f:9b01:a4f0::/60
```

There is a dhcp6 reconfigure packet. So is this a bug in OpenWRT?

jess-sch, 2025-07-27:
[This german language Heise/c't article from 2023](https://www.heise.de/select/ct/2023/2/2230709000622055021) mentions that the Fritz!Box does not implement DHCPv6 Reconfigure in a standards-compliant way and therefore only works correctly with other AVM routers.

Given that I experience the same problem of ignored Reconfigure messages trying to run MikroTik RouterOS behind a Fritz!Box, it seems that this issue still hasn't been fixed on the AVM side.

My current workaround is to enable SLAAC on the upstream interface and run a background script that renews the DHCPv6 lease whenever the SLAAC interface addresses change - maybe you could do something similar?

```
:local interface vlan2

:local previousList [:toarray ""]
:while (true) do={
  :local currentList [/ipv6/address/find interface=$interface global dynamic]
  :if ($previousList != $currentList) do={
    :log info "Triggering DHCPv6 reload due to SLAAC address change"
    :local dhcpClient [/ipv6/dhcp-client/find interface=$interface]
    /ipv6/dhcp-client/renew $dhcpClient
  }
  :set previousList $currentList
  :delay 0.5
}
```

JuliusZet, 2025-07-27:
I am over it by now. I replaced my FRITZ!Box with a simple DSL modem which forwards PPPoE to my OpenWrt. I then set the IPv6 prefix lifetime to 10 minutes on all LAN interfaces, so that all devices will receive the new prefix after maximum 5 minutes. I think I can live with that. ;)

bdrung, 2025-10-26:
@JuliusZet I configured the Fritz!Box to send IA_PD (Heimnetz -> Netzwerk -> Netzwerkeinstellungen -> Erweiterte Netzwerkeinstellungen ändern -> IPv6 -> DHCPv6-Server).

I tried again with OpenWrt 24.10.4 and with FRITZ!OS-Version 8.20 on the Fritz!Box, but the problem stayed the same.

But I found a workaround: Disable IA_PD (Heimnetz -> Netzwerk -> Netzwerkeinstellungen -> Erweiterte Netzwerkeinstellungen ändern -> IPv6 -> DHCPv6-Server -> Nur DNS-Server zuweisen). Then configure OpenWRT to use IPv6 relay: https://openwrt.org/docs/guide-user/network/ipv6/configuration#ipv6_relay

```
# cat /etc/config/dhcp
config dhcp lan
    option dhcpv6 relay
    option ra relay
    option ndp relay
    ...
 
config dhcp wan6
    option dhcpv6 relay
    option ra relay
    option ndp relay
    option master 1
    option interface wan6
```