openwrt/openwrt #17744: OpenWrt 24.10.0-rc5 syslog-ng does not start at boot

Konstantin-Glukhov, 2025-01-27:
### Describe the bug

syslog-ng does not start during boot.
I put debug 'set -x' in /etc/init.d/syslog-ng and there is no hint of an error there:
```
cat /etc/init.d/syslog-ng
#!/bin/sh /etc/rc.common
# Copyright (C) 2006-2019 OpenWrt.org

START=50

USE_PROCD=1

msg() {
    date "+%H:%M:%S $*"
}

start_service() {
    local _LOG
    _LOG=/tmp/syslog-ng.debug
    msg '*** reset syslog-ng ***' >> $_LOG
        [ -f /etc/syslog-ng.conf ] || return 1
    msg 'start_service start syslog-ng' >> $_LOG
    set -x
        procd_open_instance >> $_LOG 2>&1
        procd_set_param command /usr/sbin/syslog-ng --foreground >> $_LOG 2>&1
        procd_close_instance >> $_LOG 2>&1
    set +x
    msg 'start_service end   syslog-ng' >> $_LOG
}

reload_service() {
        /usr/sbin/syslog-ng-ctl reload
}
```
There is no problem of starting syslog-ng manually after reboot:

service syslog-ng start

The debug files are identical between boot and manual start.

### OpenWrt version

r28304-6dacba30a7

### OpenWrt release

24.10.0-rc5

### OpenWrt target/subtarget

ipq806x/generic

### Device

TP-Link Talon AD7200

### Image kind

Official downloaded image

### Steps to reproduce

opkg remove logd
opkg install syslog-ng
reboot

after reboot check for running syslog-ng

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

Chr1s70ph, 2025-01-29:
For services to start at boot, you have to `enable` them, in addition to starting them.

To do this, you have to do the following:
```
/etc/init.d/syslog-ng enable
/etc/init.d/syslog-ng start
```
> Note: this is also mentioned in the [Docs](https://openwrt.org/docs/guide-user/perf_and_log/log.syslog-ng3#startup), but its not explained why you have to enable the service. While this is a standard procedure in Unix, it might be confusing for newcomers.

I hope this solves your problem!

Konstantin-Glukhov, 2025-01-29:
If it was that simple, I would not have opened the ticket :)
The service is enabled, but does not start at boot.
As I mentioned, I put the debug statements in the script, but they did not yield additional information.

Chr1s70ph, 2025-01-29:
Without providing any system logs or anything alike, helping would be impossible.

You are also running neither the latest RC, nor do you have the default syslog-ng service file.
```sh
#!/bin/sh /etc/rc.common
# Copyright (C) 2006-2019 OpenWrt.org

START=50

USE_PROCD=1

start_service() {
	[ -f /etc/syslog-ng.conf ] || return 1
	procd_open_instance
	procd_set_param command /usr/sbin/syslog-ng --foreground
	procd_close_instance
}

reload_service() {
	/usr/sbin/syslog-ng-ctl reload
}
```

Try those 2 options, and if that does not work, provide us some logs so we can trace back the issue together.

Konstantin-Glukhov, 2025-01-29:
I checked the change log for the latest rc6 and none of them affect the syslog-ng.
I wonder if people read the whole thread at all...
I explicitly stated that I modified the standard syslog-ng startup script for debugging purposes.
What logs do you expect to see?
There are no logs because syslog-ng is the only logger and it does not start during boot.
Did you try to reproduce the problem?
Did you install syslog-ng and checked if it starts on reboot?

Chr1s70ph, 2025-01-30:
> I checked the change log for the latest rc6 and none of them affect the syslog-ng.
> I wonder if people read the whole thread at all...
Don't get snarky now, please 
> I explicitly stated that I modified the standard syslog-ng startup script for debugging purposes.
> What logs do you expect to see?
Kernel logs and System logs. Also, all logs that syslog-ng generates (Tested it myself, I get logs)
> There are no logs because syslog-ng is the only logger and it does not start during boot.
> Did you try to reproduce the problem?
Yes, and I have no problems on rc-7
> Did you install syslog-ng and check if it starts on reboot?
Yes, I have no issues.

Please don't get mad at people trying to help you. You wouldn't help people yourself, if you're not talked to adequately. This is all voluntary work.

Konstantin-Glukhov, 2025-02-03:
The problem "syslog-ng doesn't start at boot" is related to **extroot**. I found a workaround by shifting **syslog-ng** startup to the end of the boot process via creating a link `/etc/rc.d/S99syslog-ng` or adding startup logic to `/etc/rc.local`.

Somehow version 24.10 changed the boot process with **extroot** in the way that the original `/etc/rc.d/S50syslog-ng` does not start **syslog-ng** anymore.

Without **extroot** the problem does not manifest itself in 24.10. **syslog-ng** starts OK from `/etc/rc.d/S50syslog-ng` without **extroot**.

c-spiess, 2025-02-05:
The problem can also be observed with 23.05.5 on systems with overlay filesystem (here ath79/generic and octeon/generic).
On an x86/64 with ext4 filesystem it can not be observed.

As workaround i disabled syslog-ng (`/etc/init.d/syslog-ng disable`),
set start order to 60 (seems to be sufficient),
edit /etc/init.d/syslog-ng and set `START=60` ,
enable again (`/etc/init.d/syslog-ng enable`) and reboot

Konstantin-Glukhov, 2025-02-13:
I could only make it start at boot with START=70

c-spiess, 2025-02-14:
Tested with 24.10.0 on my backup device (x86/64 APU 2B4):
`START=70` isn't sufficient, with `START=80` it works (didn't check 71..79).
Smells like an ugly timing problem :-(

c-spiess, 2025-02-16:
I could debug a bit:
A short shell script `/etc/syslog-ng.sh`
```
#!/bin/sh
dbg=/dev/kmsg

echo $0 start >$dbg
/usr/sbin/syslog-ng --foreground -e 2>$dbg
echo $0 end $? >$dbg
```
replaces the call in `/etc/init.d/syslog-ng`
```
#procd_set_param command /usr/sbin/syslog-ng --foreground
procd_set_param command /etc/syslog-ng.sh
```
Now i get the follwing messages
```
[   36.276435] /etc/syslog-ng.sh start
[   48.213536] [2025-02-16T11:32:53.050934] Error binding socket; addr='AF_INET6([::1]:514)', error='Address not available (99)'
[   48.227011] [2025-02-16T11:32:53.050934] Error initializing message pipeline; plugin_name='network', location='source generator network_localhost:3:1'
[   48.253928] /etc/syslog-ng.sh end 2
```
It seems that the source `network_localhost()` is causing the trouble.
I guess IPv6 isn't initialized yes at this point.
After removing `network_localhost()` from my `syslog-ng.conf` syslog-ng starts.
But that's not the solution, any hint's how to fix it properly?

c-spiess, 2025-02-16:
Another strange thing:
`network_localhost()` binds only to the IPv6 port ::1:514,
the IPv4 port 127.0.0.1:514 isn't shown in netstat.

c-spiess, 2025-02-17:
In the forum i found a [hint](https://forum.openwrt.org/t/solved-start-init-script-at-boot-after-localhost-is-ready/11507/4) how to solve the dependency on loopback interface.
To `/etc/init.d/syslog-ng ` i added 
```
boot() {
	ubus -t 60 wait_for network.interface.loopback 2>/dev/null
	start $@
}
```
@Konstantin-Glukhov can you try this?

c-spiess, 2025-02-23:
> It seems like the wait_for does the trick with START=50 (at least for my router).

Thanks for testing, [PR #26043](https://github.com/openwrt/packages/pull/26043) created.

Neustradamus, 2025-10-25:
@c-spiess: Can you check your PR comments?