openwrt/openwrt #17747: Rockchip: NanoPi device missing realtek kernel module

TyroSpy, 2025-01-27:
Package **kmod-phy-realtek** building with missing **realtek.ko** module.
https://github.com/openwrt/openwrt/blob/d981f28f762a559ae50386fa089b050aaaa78555/package/kernel/linux/modules/netdevices.mk#L453

hauke, 2025-01-27:
The PHY support is compiled into the kernel.
```
target/linux/rockchip/armv8/config-6.6:CONFIG_REALTEK_PHY=y
```
The Kernel module is not needed and OpenWrt automatically creates an empty package.

closed 2025-01-27 by hauke

TyroSpy, 2025-01-28:
I have low speed via WAN interface about 550Mbps. I tested speed with wget, and I looked at what other devices have **realtek.ko** 
```
/dev/null           100%[===================>] 100.00M  66.4MB/s    in 1.5s

2025-01-27 18:03:54 (66.4 MB/s) - '/dev/null' saved [104857600/104857600]
```
If you are sure that an empty kernel module is ok, then I will create another issue