openwrt/openwrt #17750: Network throughput much slower than expected in repeater/sta/wds/client mode

wirl, 2025-01-27:
### Describe the bug

Recently upgraded to this Zyxel T-56router to replace my old repeater but the speeds are a lot slower, I have tried both the last stable, latest rc releases and currently running a snapshot build from source for testing. I'm expecting 400mbit/s+ speeds but I'm getting just shy over 100mbit/s. I used iperf3 and speedtest-go for this on the router itself but also some wired clients connected to it. The router only runs wireless in sta mode and does not provide any wireless access point.

I have tried client mode (sta) and wds and both yield the same results, the latter slightly slower.

I have enabled/disabled various options like h/w off loading and irq balance, and combinations of them. none of it changed these speeds. It almost seems as if there is a cap somewhere. On both of my test routers also running OpenWrt snapshots I tried with those two features off and the speeds were still very high on those. Barely made a difference.

I've done a clean flash of all 3 branches at least three times just to be sure I didn't just mess up a setting or something. I really do not want to open issues that are user error and waste everybody's time. But I'm at my wits end with this. On the same shelve I have three routers all running the latest snapshot of OpenWrt and only one of them has slow speeds when using sta/repeater mode. I don't know about access point mode speeds. It's beyond the scope of my use case.

I have no idea where to start trouble shooting or what kind of logs to provide. I just know these aren't the right speeds. I searched the existing issues for anything similar and the only thing for this chipset I found was something about low tx power. I am not sure if that could be causing this issue or not.

I'm not new to compiling things from source and doing basic testing and light debugging but my skills end there. I do have a permanent serial usb plug jerry-rigged to the back for easy access in case recovery is needed. 

[image]

I did notice the tx speed is very low, could that be the related to the lower tx power issue?

Please let me know what information I can provide other than what I already supplied. I am not actively using the router yet and can do all kinds of testing and hopefully help make OpenWrt better for this router. 

### OpenWrt version

r28714-d981f28f76

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

Zyxel EX5601-T0 ubootmod

### Image kind

Self-built image

### Steps to reproduce

- Build latest snapshot
- Setup STA/client mode
- Compare speeds to a router with a different chipset (I used a glinet travel router and an older (ac only) Fritzbox for my testing)

### Actual behaviour

Slow throughput both upload and download speeds: 100-150mbit/s maximum

### Expected behaviour

High throughput of both upload and download speeds: 350-400mbit/s minimum

### Additional info

Diffconfig attached but this also happens on stock images, I'm just doing as I'm told 😺 🥇 

### Diffconfig

```text
CONFIG_TARGET_mediatek=y
CONFIG_TARGET_mediatek_filogic=y
CONFIG_TARGET_mediatek_filogic_DEVICE_zyxel_ex5601-t0-ubootmod=y
CONFIG_BUSYBOX_CUSTOM=y
CONFIG_BUSYBOX_CONFIG_FEATURE_GETOPT_LONG=y
CONFIG_BUSYBOX_CONFIG_FEATURE_IPCALC_FANCY=y
CONFIG_BUSYBOX_CONFIG_FEATURE_IPCALC_LONG_OPTIONS=y
CONFIG_BUSYBOX_CONFIG_FEATURE_SEAMLESS_XZ=y
CONFIG_BUSYBOX_CONFIG_FEATURE_TAR_LONG_OPTIONS=y
[... 256 lines cut ...]
CONFIG_PACKAGE_wget-ssl=y
CONFIG_PACKAGE_whois=y
CONFIG_PACKAGE_wireless-tools=y
CONFIG_PACKAGE_xtables-nft=y
CONFIG_PACKAGE_xz=y
CONFIG_PACKAGE_xz-utils=y
CONFIG_PREINITOPT=y
CONFIG_TARGET_PREINIT_BROADCAST="172.16.0.255"
CONFIG_TARGET_PREINIT_IP="172.16.0.1"
CONFIG_TARGET_PREINIT_NETMASK="255.255.0.0"
CONFIG_ZSTD_OPTIMIZE_O3=y
# CONFIG_PACKAGE_kmod-mppe is not set
CONFIG_PACKAGE_kmod-ppp=y
# CONFIG_PACKAGE_kmod-ppp-synctty is not set
# CONFIG_PACKAGE_kmod-pppoa is not set
CONFIG_PACKAGE_kmod-pppoe=y
# CONFIG_PACKAGE_kmod-pppol2tp is not set
CONFIG_PACKAGE_kmod-pppox=y
# CONFIG_PACKAGE_kmod-pptp is not set
CONFIG_PACKAGE_kmod-slhc=y
```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

wirl, 2025-01-29:
I did more testing and noticed the tx link goes from vht-msc 7-9 to 0 instantly when under load and goes back to 7-9 once the load is stopped. I do not observe this on the other two routers where the vht-mcs stays high throughout. I hope this can help solve this issue.

While researching this I found several issues here (some dating back to 2020) and some topics on the forum relating to this, one suggested to lower the tc power is when its too high causes a lot of tx retries, I noticed my retries are very high too. It seems there have been lingering issues with mt-chips on 5ghz in relation to low tx speeds.

I tested with lowered tx speeds but the vht-mcs stills drops to 0, went down to 5mw in small steps. I also tried all available channel widths which made no difference as well. Interestingly the channel width is always 80mhz, no matter what I configure. The AP it connects to supports 20/40 and 80. It seems the setting has no impact. Changing the tx power is working but unfortunately has no effect on the issue. Also tried changing reg-db country which had no effect either. Same applies to changing between N/AC/AX modes.

I really hope this can be fixed but I'm not very hopeful after reading previous issues and forum threads on mt7x chip.

wirl, 2025-01-29:
did more testing and I give up. This issue is above my pay grade and I have no idea why it's so slow. I bought this router as it was suggested a lot on the forum but seems fort my use case I just got myself a paperweight. Very sad.

closed 2025-01-29 by wirl

wirl, 2025-01-29:
Did more testing and I give up. This issue is above my pay grade and I have no idea why it's so slow. I bought this router as it was suggested a lot on the forum and the specs are great but seems for my use case I just got myself a paperweight. Very sad.

I'm pretty sad the router can not be used for what I bought it for, Guess I should be happy I still have an old router that works.Just sad this highly recommended router has an old issue lingering for years affecting in just the area I needed to use it. Wish I did better research and bought another router, Now my budget is gone and I'm stuck with a router I have no use for.

wirl, 2025-01-29:
I'm a dumbass who clicked the wrong button and closed the issue instead of only adding a comment.

zearp, 2025-07-28:
Issue still persists in the current master branch (r30616-6b981c5464). So disheartening to see that link speed drop to 6mbit and just crap itself. Guess this is one of those bugs that will never be fixed. No STA/WDS or mesh for the Zyxel T-56 unless you don't mind it being capped around ~100mbit and have many retries and maybe some airtime congestion/pollution due to those. I get about 90k retries every 15 minutes on an idle wds link to another OpenWrt router.