openwrt/openwrt #17756: Crash within qcom_socinfo driver when attempting to read from various /sys/kernel/debug/qcom_socinfo/ locations

plaes, 2025-01-28:
### Describe the bug

Seems like Qualcomm's `qcom_socinfo` driver is causing some issues (or is not properly initialized), as attempts to read from various locations under `/sys/kernel/debug/qcom_socinfo/...` cause crashes:

```
cat /sys/kernel/debug/qcom_socinfo/cnss/oem 
[  238.293679] 8<--- cut here ---
[  238.293710] Unable to handle kernel NULL pointer dereference at virtual address 000006da when read
[  238.295632] [000006da] *pgd=44632835, *pte=00000000, *ppte=00000000
[  238.304676] Internal error: Oops: 17 [#1] SMP ARM
[  238.310732] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_inet ath10k_pci(O) ath10k_core(O) ath(O) pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mac80211(O) cfg80211(O) slhc nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c crc_ccitt compat(O) ledtrig_usbport sha512_arm seqiv geniv cmac leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_qcom ohci_platform ohci_hcd phy_qcom_ipq806x_usb ahci fsl_mph_dr_of ehci_platform ehci_fsl sd_mod ahci_platform libahci_platform libahci libata scsi_mod scsi_common ehci_hcd gpio_button_hotplug(O) crc32c_generic
[  238.366105] CPU: 0 PID: 2592 Comm: cat Tainted: G           O       6.6.73 #0
[  238.388339] Hardware name: Generic DT based system
[  238.395456] PC is at 0xc06e0f78
[  238.400136] LR is at 0xc04dc464
[  238.403175] pc : [<c06e0f78>]    lr : [<c04dc464>]    psr: 00000013
[  238.406305] sp : c2f3be38  ip : c3002000  fp : c207d3d8
[  238.412553] r10: 00400cc0  r9 : 7ffff000  r8 : 00000000
[  238.417761] r7 : c207d3e8  r6 : 00001000  r5 : 00000001  r4 : c207d3c0
[  238.422972] r3 : c06e0f74  r2 : 0000067a  r1 : 00000001  r0 : c207d3c0
[  238.429571] Flags: nzcv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment none
[  238.436081] Control: 10c5787d  Table: 438bc06a  DAC: 00000051
[  238.443282] Register r0 information: slab seq_file start c207d3c0 pointer offset 0 size 80
[  238.449016] Register r1 information: non-paged memory
[  238.457171] Register r2 information: non-paged memory
[  238.462293] Register r3 information: non-slab/vmalloc memory
[  238.467328] Register r4 information: slab seq_file start c207d3c0 pointer offset 0 size 80
[  238.473061] Register r5 information: non-paged memory
[  238.481130] Register r6 information: non-paged memory
[  238.486252] Register r7 information: slab seq_file start c207d3c0 pointer offset 40 size 80
[  238.491290] Register r8 information: NULL pointer
[  238.499446] Register r9 information: non-paged memory
[  238.504306] Register r10 information: non-paged memory
[  238.509342] Register r11 information: slab seq_file start c207d3c0 pointer offset 24 size 80
[  238.514382] Register r12 information: slab kmalloc-cg-4k start c3002000 pointer offset 0 size 4096
[  238.522977] Process cat (pid: 2592, stack limit = 0xa52e7704)
[  238.531739] Stack: (0xc2f3be38 to 0xc2f3c000)
[  238.537553] be20:                                                       00000000 c2f3bea8
[  238.541901] be40: c2f3be88 00000180 ffffffff 00000180 c04ba980 c2f3bf88 c2e6e540 00001000
[  238.550060] be60: 00004004 00000000 c2f3bf88 00000003 00000000 c04dc89c 00001000 c0d10c38
[  238.558219] be80: bed86db0 00001000 00000000 00000001 00000000 c2f3be80 00001000 00000000
[  238.566379] bea0: 00000001 00000000 c2e6e540 00000000 00000000 00000000 00000000 00000000
[  238.574539] bec0: 00000000 00004004 00000000 00000000 c04dc828 c2e6e540 bed86db0 00001000
[  238.582700] bee0: c1407330 c0595b04 c2e6e540 c2f3bf88 bed86db0 00000001 00001000 c0595ab0
[  238.590859] bf00: 00000003 c04afa80 c2f3bf48 00000000 00000000 00000000 c2e6e540 01000000
[  238.599020] bf20: 00000000 c04aef3c 01000000 00000000 c03002c4 c2f3bfb0 bed86dac c0224180
[  238.607179] bf40: 00000000 00000000 00000000 00000000 bed86000 00000000 00000000 00000000
[  238.615338] bf60: 00000000 c2e6e540 c2e6e540 00000000 00000000 c03002c4 c31d2800 00000003
[  238.623497] bf80: 00000000 c04b0474 00000000 00000000 00000003 00000000 00000000 00000003
[  238.631658] bfa0: 00000003 c0300060 00000000 00000000 00000003 bed86db0 00001000 00000000
[  238.639817] bfc0: 00000000 00000000 00000003 00000003 00077600 00000003 00000000 00000000
[  238.647977] bfe0: bed86d58 bed86d48 b6f25c00 b6f24e14 60000010 00000003 00000000 00000000
[  238.656141] Function entered at [<c06e0f78>] from [<c2f3bea8>]
[  238.664321] Code: e34c106e e5932194 eaf7ea7f e5902048 (e5d23060) 
[  238.670118] ---[ end trace 0000000000000000 ]---
```

### OpenWrt version

r28388-58d0057481

### OpenWrt release

24.10.0-rc6

### OpenWrt target/subtarget

ipq806x/generic

### Device

Extreme Networks AP3935

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-01-28:
Could you try to list all affected files using `tar cvf /dev/null sys/kernel/debug/qcom_socinfo`

plaes, 2025-01-28:
> Could you try to list all affected files using `tar cvf /dev/null sys/kernel/debug/qcom_socinfo`

Unfortunately it reboots immediately when it crashes. Most of the files there cause crashes, including `name`.