openwrt/openwrt #17761: MicroTik hAP ac3 wireless not detected

microftxx, 2025-01-28:
### Describe the bug

There is no Network / Wireless item in LuCi's menu, nothing in /etc/config/wireless and so on.

I've tried to receive any recommendation on OpenWRT forum, but i got no decision.

The situation is the same on 2 such devices.

Very important info: i've installed OpenWrt 24.10.0-rc6. WiFi works fine. But I need OpenWrt 23.05.5 exactly. 
OpenWrt 23.05.5 vs MikroTik hAP ac3 are described as good friends (see https://openwrt.org/toh/start)

### OpenWrt version

r24106-10cc5fcd00

### OpenWrt release

23.05.5

### OpenWrt target/subtarget

ipq40xx/mikrotik

### Device

MicroTik hAP ac3

### Image kind

Official downloaded image

### Steps to reproduce

Just install OpenWrt image onto device. That's all.

### Actual behaviour

WiFi is not available.

### Expected behaviour

I want WiFi to work on my device.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

mrkiko, 2025-01-28:
Hello!!

It may be nice if you could provide the dmesg from 24.x running on the device.
To see why, have a look at commit 7d33aedd10bab0d8023d608f6f900b2915c69456 in openwrt main GIT tree.
In other words, your device may contain LZ77 compressed WAN calibration data, and you might need to backport support for that in 23.x in you really with to stay there.

microftxx, 2025-01-28:
> Hello!!
> 
> It may be nice if you could provide the dmesg from 24.x running on the device. To see why, have a look at commit [7d33aed](https://github.com/openwrt/openwrt/commit/7d33aedd10bab0d8023d608f6f900b2915c69456) in openwrt main GIT tree. In other words, your device may contain LZ77 compressed WAN calibration data, and you might need to backport support for that in 23.x in you really with to stay there.

Yes, I want to stay on 23.x. 

This is it:

```
root@OpenWrt:~# dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 6.6.73 (builder@buildhost) (arm-openwrt-linux-muslgnueabi-gcc (OpenWrt GCC 13.3.0 r28388-58d0057481) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 SMP Wed Jan 22 19:52:54 2025
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: MikroTik hAP ac3
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] OF: reserved mem: 0x87e00000..0x87e7ffff (512 KiB) nomap non-reusable smem@87e00000
[    0.000000] OF: reserved mem: 0x87e80000..0x87ffffff (1536 KiB) nomap non-reusable tz@87e80000
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000080000000-0x000000008fffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080000000-0x0000000087dfffff]
[    0.000000]   node   0: [mem 0x0000000087e00000-0x0000000087ffffff]
[    0.000000]   node   0: [mem 0x0000000088000000-0x000000008fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080000000-0x000000008fffffff]
[    0.000000] percpu: Embedded 13 pages/cpu s21524 r8192 d23532 u53248
[    0.000000] pcpu-alloc: s21524 r8192 d23532 u53248 alloc=13*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Kernel command line:
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64960
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 246136K/262144K available (7670K kernel code, 644K rwdata, 1140K rodata, 1024K init, 251K bss, 16008K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 48.00MHz (virt).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xb11fd3bfb, max_idle_ns: 440795203732 ns
[    0.000001] sched_clock: 56 bits at 48MHz, resolution 20ns, wraps every 4398046511096ns
[    0.000021] Switching to timer-based delay loop, resolution 20ns
[    0.000332] Calibrating delay loop (skipped), value calculated using timer frequency.. 96.00 BogoMIPS (lpj=480000)
[    0.000355] CPU: Testing write buffer coherency: ok
[    0.000407] pid_max: default: 32768 minimum: 301
[    0.010249] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.010273] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.018858] qcom_scm: convention: smc legacy
[    0.020408] RCU Tasks Trace: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1 rcu_task_cpu_ids=4.
[    0.020602] Setting up static identity map for 0x80300000 - 0x8030003c
[    0.020797] rcu: Hierarchical SRCU implementation.
[    0.020805] rcu:     Max phase no-delay instances is 1000.
[    0.021528] smp: Bringing up secondary CPUs ...
[    0.025330] smp: Brought up 1 node, 4 CPUs
[    0.025353] SMP: Total of 4 processors activated (384.00 BogoMIPS).
[    0.025366] CPU: All CPU(s) started in SVC mode.
[    0.033458] VFP support v0.3: implementor 41 architecture 2 part 30 variant 7 rev 5
[    0.033649] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.033681] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.038874] pinctrl core: initialized pinctrl subsystem
[    0.042136] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[    0.042570] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.044186] thermal_sys: Registered thermal governor 'step_wise'
[    0.044302] cpuidle: using governor ladder
[    0.044341] cpuidle: using governor menu
[... 181 lines cut ...]
[   39.514902] br-lan: port 1(lan1) entered disabled state
[   39.520353] qca8k-ipq4019 c000000.switch lan1: entered allmulticast mode
[   39.525323] ipqess-edma c080000.ethernet eth0: entered allmulticast mode
[   39.533034] qca8k-ipq4019 c000000.switch lan1: entered promiscuous mode
[   39.556491] qca8k-ipq4019 c000000.switch lan2: configuring for phy/psgmii link mode
[   39.557898] br-lan: port 2(lan2) entered blocking state
[   39.563179] br-lan: port 2(lan2) entered disabled state
[   39.568288] qca8k-ipq4019 c000000.switch lan2: entered allmulticast mode
[... cut ...]
[   39.631085] qca8k-ipq4019 c000000.switch lan4: entered allmulticast mode
[   39.636938] qca8k-ipq4019 c000000.switch lan4: entered promiscuous mode
[   39.658234] qca8k-ipq4019 c000000.switch wan: configuring for phy/psgmii link mode
[   42.808937] qca8k-ipq4019 c000000.switch wan: Link is Up - 1Gbps/Full - flow control rx/tx
[   43.690561] qca8k-ipq4019 c000000.switch lan1: Link is Up - 1Gbps/Full - flow control rx/tx
[   43.690717] br-lan: port 1(lan1) entered blocking state
[   43.697770] br-lan: port 1(lan1) entered forwarding state
```

mrkiko, 2025-01-28:
Ok, I think I misinterpreted the commit message - sorry.

Still, I would give it a try: try to backport the commit adding LZ77 support and see if it helps in 23.x.

microftxx, 2025-01-29:
> Ok, I think I misinterpreted the commit message - sorry.
> 
> Still, I would give it a try: try to backport the commit adding LZ77 support and see if it helps in 23.x.

So, I have to rebuild OpenWrt image with new config files?

dchard, 2025-04-10:
Has nothing to do with LZ77, according to the dmesg both wifi interfaces are properly initialized. Are any radios actually enabled?