openwrt/openwrt #17764: MT7621: DNA Valokuitu Plus EX400 Starts One Core Only

msandber, 2025-01-28:
### Describe the bug

Only one out of two cores are being brought up.

OpenWrt master:
```
[    0.000000] Linux version 6.6.74 (builder@buildhost) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r28694-730db6b893) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 SMP Sun Jan 26 00:17:11 2025
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is DNA Valokuitu Plus EX400
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2} total 2
...
[    0.212485] smp: Brought up 1 node, 2 CPUs
...
```

After removing a patch `'target/linux/ramips/patches-6.6/320-MIPS-add-support-for-buggy-MT7621S-core-detection.patch'` and rebuilding:

```
[    0.000000] Linux version 6.6.74 (ubuntu@ubuntu) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r16536+12016-a3d431ccf7) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 SMP Tue Jan 28 09:51:46 2025
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is DNA Valokuitu Plus EX400
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
...
[    0.388710] smp: Brought up 1 node, 4 CPUs
...
```

### OpenWrt version

r28694-730db6b893

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ramips/mt7621

### Device

DNA Valokuitu Plus EX400

### Image kind

Official downloaded image

### Steps to reproduce

N/A, happens always.

### Actual behaviour

Only one out of two cores is started.

### Expected behaviour

Both cores on an MT7621 SoC would be brought up.

### Additional info

The said patch was quickly reverted after it's submittal in upstream with commit [740da9d7ca4](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/arch/mips/include/asm/mips-cps.h?id=740da9d7ca4e25f5d87db9f80d75432681b61305).

Table of hardware reports 12 devices with CPU 'MT7621S. It is unknown if other dual-core MT7621 devices are affected by this.

### Diffconfig

```text
CONFIG_TARGET_ramips=y
CONFIG_TARGET_ramips_mt7621=y
CONFIG_TARGET_ramips_mt7621_DEVICE_dna_valokuitu-plus-ex400=y
CONFIG_HTOP_LMSENSORS=y
CONFIG_PACKAGE_cgi-io=y
CONFIG_PACKAGE_htop=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-ucode=y
CONFIG_PACKAGE_libncurses=y
CONFIG_PACKAGE_luci=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-package-manager=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-light=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_terminfo=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

mrkiko, 2025-01-29:
May I ask for yourhelp @arinc9 ? Hoping I am tagging you right. :)

arinc9, 2025-01-29:
Sorry, this is neither my area of expertise nor my responsibility. You should find the person that put this patch in OpenWrt which causes the bad behaviour.

mrkiko, 2025-01-29:
Sorry, I think I tagged the wrong person. :)

namiltd, 2025-01-29:
Please check this patch and provide syslog information: [320-MIPS-add-support-for-buggy-MT7621S-core-detection.patch](https://github.com/user-attachments/files/18588347/320-MIPS-add-support-for-buggy-MT7621S-core-detection.patch)

namiltd, 2025-01-29:
We will probably need a few more such logs from other single and dual-core MT7620 devices that have no problem with correct detection.

namiltd, 2025-01-29:
Try this: [320-MIPS-add-support-for-buggy-MT7621S-core-detection.patch](https://github.com/user-attachments/files/18591291/320-MIPS-add-support-for-buggy-MT7621S-core-detection.patch)

msandber, 2025-01-30:
This seems to have done the trict at least for my device.

```
[    0.000000] Linux version 6.6.74 (ubuntu@ubuntu) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r16536+12016-a3d431ccf7) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 SMP Tue Jan 28 09:51:46 2025
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is DNA Valokuitu Plus EX400
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] percpu: Embedded 12 pages/cpu s18816 r8192 d22144 u49152
[    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Writing ErrCtl register=00030340
[    0.000000] Readback ErrCtl register=00030340
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64960
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 247500K/262144K available (8039K kernel code, 654K rwdata, 1712K rodata, 1232K init, 219K bss, 14644K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 256
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000004] sched_clock: 64 bits at 880MHz, resolution 1ns, wraps every 4398046511103ns
[    0.008050] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[    0.066203] pid_max: default: 32768 minimum: 301
[    0.079470] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.086704] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.102918] RCU Tasks Trace: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1 rcu_task_cpu_ids=4.
[    0.112671] rcu: Hierarchical SRCU implementation.
[    0.117370] rcu: 	Max phase no-delay instances is 1000.
[    0.123340] smp: Bringing up secondary CPUs ...
[    0.128597] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.128622] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.128636] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.128684] CPU1 revision is: 0001992f (MIPS 1004Kc)
[    0.182773] Synchronize counters for CPU 1: done.
[    0.214674] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.214697] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.214709] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.214740] CPU2 revision is: 0001992f (MIPS 1004Kc)
[    0.273655] Synchronize counters for CPU 2: done.
[    0.304336] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.304355] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.304368] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.304402] CPU3 revision is: 0001992f (MIPS 1004Kc)
[    0.358858] Synchronize counters for CPU 3: done.
[    0.388719] smp: Brought up 1 node, 4 CPUs
```

```
cat /proc/cpuinfo 
system type		: MediaTek MT7621 ver:1 eco:3
machine			: DNA Valokuitu Plus EX400
processor		: 0
cpu model		: MIPS 1004Kc V2.15
BogoMIPS		: 586.13
wait instruction	: yes
[... 47 lines cut ...]
package			: 0
core			: 1
VPE			: 0
VCED exceptions		: not available
VCEI exceptions		: not available

processor		: 3
cpu model		: MIPS 1004Kc V2.15
BogoMIPS		: 586.13
wait instruction	: yes
microsecond timers	: yes
tlb_entries		: 32
extra interrupt vector	: yes
hardware watchpoint	: yes, count: 4, address/irw mask: [0x0ffc, 0x0ffc, 0x0ffb, 0x0ffb]
isa			: mips1 mips2 mips32r1 mips32r2
ASEs implemented	: mips16 dsp mt
Options implemented	: tlb 4kex 4k_cache prefetch mcheck ejtag llsc pindexed_dcache userlocal vint perf_cntr_intr_bit cdmm perf
shadow register sets	: 1
kscratch registers	: 0
package			: 0
core			: 1
VPE			: 1
VCED exceptions		: not available
VCEI exceptions		: not available
```

namiltd, 2025-01-30:
Whether we will implement the patch depends on its behavior on other mt7621s, including mt7621s.

msandber, 2025-01-30:
> Patch for testing on other routers: [320-MIPS-add-support-for-buggy-MT7621S-core-detection.patch](https://github.com/user-attachments/files/18606208/320-MIPS-add-support-for-buggy-MT7621S-core-detection.patch)

Regarding the patch, `pr_debug` did not display anything, I used `pr_info`. `%d` did not compile and I used `%lu` instead.

namiltd, 2025-01-30:
Corrected: [320-MIPS-add-support-for-buggy-MT7621S-core-detection.patch](https://github.com/user-attachments/files/18606376/320-MIPS-add-support-for-buggy-MT7621S-core-detection.patch)

namiltd, 2025-01-30:
Corrected: [320-MIPS-add-support-for-buggy-MT7621S-core-detection.patch](https://github.com/user-attachments/files/18606496/320-MIPS-add-support-for-buggy-MT7621S-core-detection.patch)

namiltd, 2025-01-30:
For others mt7621 (not mt7621s) is:
```
[    0.000000] cpu0:0 flags = 0,
[    0.000000] cpu0:1 flags = 1,
[    0.000000] cpu1:0 flags = 1,
[    0.000000] cpu1:1 flags = 1
```
I still need tests for mt7621s.

lexmark3200, 2025-01-31:
I have sold my r6220 so I can't help you.
Now I have er-x, dir-878, dir-2660 and dir-2150 a1.
All of them mt7621.

namiltd, 2025-01-31:
Ok, thanks for the information.

andyboeh, 2025-02-01:
I have a few NA502/NA502s in the closet, but it's going to take a few weeks until I have time to test this.

mrkiko, 2025-02-02:
Going to test with an R6220. Before the patch is applied, the device is reporting the following (running an old version of OpenWrt):

[    0.000000] Linux version 5.15.107 (builder@buildhost) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 12.2.0 r22608-1f68baf5f0) 12.2.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Fri Apr 21 16:44:40 2023
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is Netgear R6220
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2} total 2
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] percpu: Embedded 11 pages/cpu s15856 r8192 d21008 u45056
[    0.000000] pcpu-alloc: s15856 r8192 d21008 u45056 alloc=11*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
[    0.000000] Kernel command line: console=ttyS0,57600 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Writing ErrCtl register=00020600
[    0.000000] Readback ErrCtl register=00020600
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 118584K/131072K available (7433K kernel code, 626K rwdata, 1500K rodata, 1244K init, 251K bss, 12488K reserved, 0K cma-reserved, 0K highmem)

Patching & compiling; will be reporting back ASAP.

mrkiko, 2025-02-02:
After updating via compilation & applying the patch, the device reports:

[    0.000000] Linux version 6.6.74 (mrkiko@mStation) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r28739-69890e16b3) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 SMP Sat Feb  1 05:06:59 2025
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is Netgear R6220
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {
[    0.000000] cpu0:0 flags = 32, 
[    0.000000] cpu0:0 flags = 1, 
[    0.000000] cpu1:0 flags = 16711680, 
[    0.000000] cpu1:1 flags = 16449536
[    0.000000] 2} total 2
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] percpu: Embedded 12 pages/cpu s18816 r8192 d22144 u49152
[    0.000000] pcpu-alloc: s18816 r8192 d22144 u49152 alloc=12*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 
[    0.000000] Kernel command line: console=ttyS0,57600 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Writing ErrCtl register=00020600
[    0.000000] Readback ErrCtl register=00020600
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 117776K/131072K available (8039K kernel code, 654K rwdata, 1712K rodata, 1232K init, 219K bss, 13296K reserved, 0K cma-reserved)

namiltd, 2025-02-02:
Thank you very much. The fix works as it should.

namiltd, 2025-02-07:
Could you please check this patch?: [320-MIPS-add-support-for-buggy-MT7621S-core-detection.patch](https://github.com/user-attachments/files/18710834/320-MIPS-add-support-for-buggy-MT7621S-core-detection.patch)

namiltd, 2025-02-07:
Could you please check this patch?: [320-MIPS-add-support-for-buggy-MT7621S-core-detection.patch](https://github.com/user-attachments/files/18711915/320-MIPS-add-support-for-buggy-MT7621S-core-detection.patch)

msandber, 2025-02-07:
There was one version of the patch that did not compile so copypasting here what I tried with just in case.

```
--- a/arch/mips/include/asm/mips-cps.h
+++ b/arch/mips/include/asm/mips-cps.h
@@ -11,6 +11,8 @@
 #include <linux/io.h>
 #include <linux/types.h>
 
+#include <asm/mach-ralink/mt7621.h>
+
 extern unsigned long __cps_access_bad_size(void)
 	__compiletime_error("Bad size for CPS accessor");
 
@@ -165,6 +167,9 @@ static inline unsigned int mips_cps_numc
 	if (!mips_cm_present())
 		return 0;
 
+	if (IS_ENABLED(CONFIG_SOC_MT7621))
+		return 1 + FIELD_GET(0x20000, __raw_readl(MT7621_SYSC_BASE + SYSC_REG_CHIP_REV));
+ 
 	/* Add one before masking to handle 0xff indicating no cores */
 	return FIELD_GET(CM_GCR_CONFIG_PCORES,
 			 mips_cps_cluster_config(cluster) + 1);
```

I get.

```
[    0.000000] VPE topology {2,2} total 4
...
[    0.388686] smp: Brought up 1 node, 4 CPUs
```

Looks like it's working.

closed 2025-02-17 by openwrt-bot