openwrt/openwrt #17768: strongswan packages disappeared from 24.10.0-rc7 aarch64-a53 repository

jonkerj, 2025-01-29:
### Describe the bug

With 24.10.0-rc7 being released, I tried to imagebuild an updated image for my router (bpi-r4), but this failed due to strongswan packages not being present in the repositories.

A quick grep for `strongswan-minimal` in [rc7 package index](https://mirror-03.infra.openwrt.org/releases/24.10.0-rc7/packages/aarch64_cortex-a53/packages/Packages) confirms this. The strange thing is, it seems to be removed from the rc6 index as well, while my rc6 build on Jan 24th seems to have successfully downloaded both an index containing `strongswan-minimal` and the `strongswan-minimal_5.9.14-r4_aarch64_cortex-a53.ipk`, so rc6 currently does not build with `PACKAGES=strongswan=minimal` anymore.

So somehow, the packages appear to have been removed from the index afterwards last week.

### OpenWrt version

r28417-daef29c75d

### OpenWrt release

24.10.0-rc7

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R4

### Image kind

Self-built image

### Steps to reproduce

1. Download [filogic imagebuilder](https://downloads.openwrt.org/releases/24.10.0-rc7/targets/mediatek/filogic/openwrt-imagebuilder-24.10.0-rc7-mediatek-filogic.Linux-x86_64.tar.zst)
2. make image PACKAGES="strongswan-minimal"

### Actual behaviour

```
 * opkg_install_cmd: Cannot install package strongswan-minimal.
make[2]: *** [Makefile:234: package_install] Error 255
make[1]: *** [Makefile:161: _call_image] Error 2
make: *** [Makefile:337: image] Error 2
```

### Expected behaviour

An image containing strongswan

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2025-01-29:
`strongswan` probably failed to build, its the only reason why it would be missing.

jonkerj, 2025-01-29:
That would be my guess as well, but, assuming rc6 does not get re-built continuously, why did it vanish from the rc6 repo last week? It was there on Jan 24, and is gone now.

robimarko, 2025-01-29:
Yeah, its failling:
https://downloads.openwrt.org/releases/faillogs-24.10/aarch64_cortex-a53/packages/strongswan/compile.txt

Packages are not versioned per RC

divinehawk, 2025-02-01:
https://github.com/strongswan/strongswan/commit/60336ceecbd1cda73aa26dd44cfdaf2e31a046e1 perhaps?

robimarko, 2025-02-06:
https://github.com/openwrt/packages/commit/56250817c47b040f0381bca3c9c3395f435e5054 will fix it

closed 2025-02-06 by robimarko