openwrt/openwrt #17770: ZyXEL EX5601-T0/T-56 Very slow wireless speeds in repeater/sta mode: vht-msc goes from 7 to 0 under load

wirl, 2025-01-29:
### Describe the bug

Accidentally closed my [other issue](https://github.com/openwrt/openwrt/issues/17750) about this, but it is ok and took the opportunity to install the latest rc instead of my self compiled snapshot and re-test and re-open an issue about it. I would assume knowing exactly what imagine I run will help in fixing it unlike self compiled images.

I am running this router in repeater/sta mode and getting very low speeds compared to my old routers. I don't know if it is related to the low tx power issue reported for this target too. The power doesn't seem to be an issue here., it's not stuck at a low value and changing it to a low value also didn't improve anything. What happens is that as soon as the connection is under load the speed drops to about 100mbit/s and stays there until the load is finished or cancelled. 

I have enabled/disabled things like hardware nat, setting packet steering to use all cores and irq balance. None of these improved or fixed the problem.

The main thing I noticed is that the link stats change as soon as there is load on the connection, tx goes from vht-msc 7 to 0 and instantly goes back up once the load is stopped. I also noticed the tx link rate sits at 6mbit/s and the number of retries is very high.

I did a lot of searching and found that many MediaTek chips suffer from similar/related issues. But nothing that really matches my issue. I am convinced that it is not user error. I have two other routers to test with and they're both fine and also run OpenWrt with pretty much the same settings. Speeds are as expected there. The issue must lie in the MediaTek driver somewhere.

I really hope this issue can be solved. I bought this router to serve as repeater and to run a few containers on.

Please let me know what logs I need to provide and what tests I can do. If this is a known issue that can't be fixed also please let me know. 

### OpenWrt version

r28417-daef29c75d

### OpenWrt release

24.10.0-rc7

### OpenWrt target/subtarget

mediatek/filogic

### Device

Zyxel EX5601-T0 ubootmod

### Image kind

Official downloaded image

### Steps to reproduce

- Install snapshot and setup a client mode connection
- Compare speeds to another router with the same client mode connection

### Actual behaviour

Low speeds, maxing out around 100mbit/s

### Expected behaviour

High speeds, minimally 600-700mbit/s

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

wirl, 2025-01-30:
I looked everywhere but couldn't find anything in the stock firmwere, which did have a lot of options so maybe I missed it. But I don't think ti has it, I did see things like mesh and such but no simple sta/wds modes.

```
root@OpenWrt:~# cat /sys/kernel/debug/ieee80211/phy1/mt76/txpower_sku

Phy1 Tx power table (channel 36)
                      1m     2m     5m    11m
CCK (TMAC)      :     34     34     34     34
                      6m     9m    12m    18m    24m    36m    48m    54m
OFDM (TMAC)     :     34     34     34     34     34     34     34     34
                    mcs0   mcs1   mcs2   mcs3   mcs4   mcs5   mcs6   mcs7
HT_BW20 (TMAC)  :     34     34     34     34     34     34     34     34
                    mcs0   mcs1   mcs2   mcs3   mcs4   mcs5   mcs6   mcs7  mcs32
HT_BW40 (TMAC)  :     34     34     34     34     34     34     34     34     34
                    mcs0   mcs1   mcs2   mcs3   mcs4   mcs5   mcs6   mcs7   mcs8   mcs9  mcs10  mcs11
VHT_BW20 (TMAC) :     34     34     34     34     34     34     34     34     34     34      0      0
VHT_BW40 (TMAC) :     34     34     34     34     34     34     34     34     34     34      0      0
VHT_BW80 (TMAC) :     34     34     34     34     34     34     34     34     34     34      0      0
VHT_BW160 (TMAC):     34     34     34     34     34     34     34     34     34     34      0      0
HE_RU26 (TMAC)  :     34     34     34     34     34     34     34     34     34     34     34     34
HE_RU52 (TMAC)  :     34     34     34     34     34     34     34     34     34     34     34     34
HE_RU106 (TMAC) :     34     34     34     34     34     34     34     34     34     34     34     34
HE_RU242 (TMAC) :     34     34     34     34     34     34     34     34     34     34     34     34
HE_RU484 (TMAC) :     34     34     34     34     34     34     34     34     34     34     34     34
HE_RU996 (TMAC) :     34     34     34     34     34     34     34     34     34     34     34     34
HE_RU2x996 (TMAC):     34     34     34     34     34     34     34     34     34     34     34     34

Tx power (bbp)  :     34
```

wirl, 2025-01-30:
I looked everywhere but couldn't find anything in the stock firmwere, which did have a lot of options so maybe I missed it. But I don't think it has it, I did see things like mesh and such but no simple sta/wds modes.

```
root@OpenWrt:~# cat /sys/kernel/debug/ieee80211/phy1/mt76/txpower_sku

Phy1 Tx power table (channel 36)
                      1m     2m     5m    11m
CCK (TMAC)      :     34     34     34     34
                      6m     9m    12m    18m    24m    36m    48m    54m
OFDM (TMAC)     :     34     34     34     34     34     34     34     34
                    mcs0   mcs1   mcs2   mcs3   mcs4   mcs5   mcs6   mcs7
HT_BW20 (TMAC)  :     34     34     34     34     34     34     34     34
                    mcs0   mcs1   mcs2   mcs3   mcs4   mcs5   mcs6   mcs7  mcs32
HT_BW40 (TMAC)  :     34     34     34     34     34     34     34     34     34
                    mcs0   mcs1   mcs2   mcs3   mcs4   mcs5   mcs6   mcs7   mcs8   mcs9  mcs10  mcs11
VHT_BW20 (TMAC) :     34     34     34     34     34     34     34     34     34     34      0      0
VHT_BW40 (TMAC) :     34     34     34     34     34     34     34     34     34     34      0      0
VHT_BW80 (TMAC) :     34     34     34     34     34     34     34     34     34     34      0      0
VHT_BW160 (TMAC):     34     34     34     34     34     34     34     34     34     34      0      0
HE_RU26 (TMAC)  :     34     34     34     34     34     34     34     34     34     34     34     34
HE_RU52 (TMAC)  :     34     34     34     34     34     34     34     34     34     34     34     34
HE_RU106 (TMAC) :     34     34     34     34     34     34     34     34     34     34     34     34
HE_RU242 (TMAC) :     34     34     34     34     34     34     34     34     34     34     34     34
HE_RU484 (TMAC) :     34     34     34     34     34     34     34     34     34     34     34     34
HE_RU996 (TMAC) :     34     34     34     34     34     34     34     34     34     34     34     34
HE_RU2x996 (TMAC):     34     34     34     34     34     34     34     34     34     34     34     34

Tx power (bbp)  :     34
```

wirl, 2025-01-30:
Stock firmware and OpenWrt work well as access point, I couldn't test the speeds properly on the stock firmware but iperf3 testing got good results on OpenWrt, maybe not as high as possible but still almost 10 times faster than in sta mode. It is mostly noticeable when downloading files, other than that it works well. Connections don't drop and the router is very stable.

I tested both bands and both seems to cap out around 100bit/s. It is so odd, I even found an older wall plugged repeater (DAP-1620) which also has a mt chip and that one reaches over 200mbit/s easily as well. Also running Openwrt of course! But the wireless module is different; MT7615DN the Zyxel has a MT7976.

I don't know enough to figure out what's actually happening thats causing the tx rate to drop so much, it's not cpu load or anything like that. Cpu and ram usage were the first thing I checked. I'm running some extra collectd stuff so see if anything stands out there but I can't see anything useful. As soon as I start a download, iperf3 or another speed test the tx rate instantly drops to 6mbit/s.

I experimented with some options like vender vht modes but didn't do anything, but I could've done it wrong. I'm a bit out of my depth here. Just trying things people suggested in the issues here and the forum for similar problems -- desperation haha.

wirl, 2025-01-31:
Did some more experimenting, disabling the by the default enable ```sr_scene_detect``` didn't do anything. I did notice that in addition to the link rate being 6mbit/s the channel width also goes back to 20mhz instead of 80. Something is going on in the driver but I don't know how to look or where to look. Both dmesg and logread show nothing in relation to wireless, no errors, no warnings.

There doesn't seem to be an option for the kernel module to log more verbosely, maybe this has to be done with the mt76 module but I can't find any parameters for it (/sys/module/mt76/parameters does not exist). I don't know how to get more logging out of the drivers or how to compile the mt76/mt7915e modules with more logging/debug.

I wonder if it might be better to open an issue in the mt75 repo? This seems very much related to the driver itself and not OpenWrt as a whole.

wirl, 2025-01-31:
Connection idle:
```
RX: 975.0 Mbit/s, 80 MHz, VHT-MCS 7, VHT-NSS 3, Short GI
TX: 702.0 Mbit/s, 80 MHz, VHT-MCS 5, VHT-NSS 3
```

Connection under load:
```
RX: 877.5 Mbit/s, 80 MHz, VHT-MCS 7, VHT-NSS 3
TX: 6.0 Mbit/s, 20 MHz
```

There are no modes listed at all. Tried forcing vht modes in the config, but it also did nothing.

wirl, 2025-02-01:
Another morning of fruitless experiments. I can't solve this bug on my own. Sadly the router is not very useful when it only does 100mbit/s as a wireless repeater. Hopefully this bug can be fixed but I'm out of my depth and have to give up trying to figure it out on my own.

wirl, 2025-02-01:
Wireless chips are cl242 (rebranded mt7592) + cl2330 (not sure, likely also rebranded mt7xxx).

https://wikidevi.wi-cat.ru/Celeno

It's a cable modem, using a Compal board:

https://wikidevi.wi-cat.ru/Compal_Broadband_Networks_CH7465LG-LC

The interface for the modem is very basic. It is provided by the isp and they own it, update the firmware and such too. I can not configure much other than the basics like which channel and widths. I also experimented with those settings. Setting it to only 80mhz, 40mhz or 20mhz. I've also tried firmwares posted [here](https://git01.mediatek.com/plugins/gitiles/openwrt/feeds/mtk-openwrt-feeds/+/refs/heads/master/autobuild/autobuild_5.4_mac80211_release/package/kernel/mt76/src/firmware): they load and work but also don't fix the issue 😅 

I am just baffled that the only router I own having this issue is the Zyxel. The others with (I think very similar) mt chipsets work fine. They are just no longer up to the tasks I need them to do and the Zyxel ticks all the boxes.

Thank you for helping me, very much appreciated!

wirl, 2025-02-02:
Here is the vht info:
```
root@OpenWrt:~# cat /sys/kernel/debug/ieee80211/phy1/netdev:phy1-sta0/stations/*
/vht_capa
VHT supported
cap: 0x33827930
		MAX-MPDU-3895
		80Mhz
		RXLDPC
		SHORT-GI-80
		RXSTBC_1
		SU-BEAMFORMER-CAPABLE
		SU-BEAMFORMEE-CAPABLE
		BEAMFORMEE-STS: 0x3
		SOUNDING-DIMENSIONS: 0x2
		MPDU-LENGTH-EXPONENT: 0x7
		LINK-ADAPTATION-VHT-MRQ-MFB: 0x0
		RX-ANTENNA-PATTERN
		TX-ANTENNA-PATTERN
RX MCS: ffea
TX MCS: ffea
```

wirl, 2025-02-02:
Recompiled with above commit but no differences unfortunately. If it is helpful to recompile the driver with extra logging or debugging please let me know. 

Also hope I edited the right file, I made the changes in ```/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mt76-2025.01.22~a22d59e4/mt7915/mcu.c``` and recompiled.

wirl, 2025-02-02:
I'm not sure myself so to be certain I edited all of the mcu.c files and recompiled again 😅 

```
/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-6.6.74/drivers/net/wireless/mediatek/mt76/mt7915/mcu.c
/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mac80211-regular/backports-6.12.6/drivers/net/wireless/mediatek/mt76/mt7915/mcu.c
/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/mt76-2025.01.22~a22d59e4/mt7915/mcu.c
/build_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/linux-6.6.74/drivers/net/wireless/mediatek/mt76/mt7915/mcu.c
```

I'm sure the commit is applied now but still same issue.

zearp, 2025-02-06:
Can anyone reproduce this on their model?,

zearp, 2025-02-06:
Having the same issue here in station mode with the latest 24.10.0 stable image. I'm using a 8821au usb stick for the time being and it has been working well as temporary solution.

wirl, 2025-02-06:
Thank you for confirming!

Unfortunately I am using the usb port with an external disk as external root overlay. I'm just stumped how to solve this issue. I am not a developer and don;t know nought about the inner workings of wireless and this driver. So I don't even know what's going wrong other than it being related to vht modes but I could be wrong in that assumption.

I have signed up and posted about this on forum and maybe also open an issue in the mt76 driver repo as the driver seems to have a bug and this issue tracker is for OpenWrt in general. I'm worried opening another issue about this will get me trouble. For now I try to patiently wait and hope for a solution.

As a temporary solution here I've connected my old router to the new one using an ethernet cable and have the old router be my "usb" stick. Not ideal but better than having 100mbit/s max. Just hope and pray this bug can actually be solved.

Underworld6, 2025-02-06:
I see similar issue described here: [https://github.com/openwrt/openwrt/issues/17530](url).
and I am experiencing the same with my RT3200.

wirl, 2025-02-06:
Thank you! That seems very similar indeed and likely related in some way! Just hope we get the right eyes to look at this. Maybe it is easy to solve for someone who knows the ins and outs of this driver.

Underworld6, 2025-02-06:
Something which may be relevant: [https://forum.openwrt.org/t/mesh-and-ap-on-the-same-radio-stopped-working-on-24-10-0/223821/14?u=underworld](url)
I will test that on my Belkin Rt3200, but probably during the weekend.

wirl, 2025-02-06:
I had tried older OpenWrt versions but not different versions of wpad. I did now including the wolf ssl ones, mini and full and also does not change the behaviour. I don't think wpad is involved in sta mode, but I could be wrong.

wirl, 2025-02-07:
Good idea and done, it is getting a bit messy/information dense but I don't know how to write things with more brevity as I just don't understand enough about the underlaying systems at play here. But I do my best haha!

wirl, 2025-03-09:
https://github.com/openwrt/mt76/issues/956#issuecomment-2708727020

closed 2025-03-09 by wirl