openwrt/openwrt #17776: WAX206 MAC address is the same for all wired ports including WAN

sur1v, 2025-01-29:
### Describe the bug

MAC adresses for all wired interfaces are the same, tested on rc6 and rc7

### OpenWrt version

r28417-daef29c75d

### OpenWrt release

24.10.0-rc7

### OpenWrt target/subtarget

mediatek/mt7622

### Device

Netgear WAX206

### Image kind

Official downloaded image

### Steps to reproduce

After a fresh factory install, listed MAC adresses for all wired interfaces are the same

### Actual behaviour

WAN port is unusable until you manually change MAC address

### Expected behaviour

MAC address for WAN por should be different to avoid MAC collision

### Additional info

Tested on rc6 and rc7

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

dangowrt, 2025-01-29:
This is indeed a problem we should fix. As you got the device at hand, it would be very cool if you can revert to the factory firmware and investigate the MAC address assignment there. The device tree lists two different offsets for the MAC addresses which are being used for LAN and WAN, so the first step would be to check if the addresses are actually identical in the `factory` partition, and if so, what the vendor firmware does in that case.

1. Please dump the content of the `Factory` MTD partition (I believe it should be `/dev/mtd4`, you can check that via ` cat /proc/mtd`), ie. `hexdump -C /dev/mtd4`
2. If the addresses at offsets `0x7fff4` and `0x7fffa` in the `Factory` partition are indeed identical: Revert to the stock firmware and see which MAC addresses are used.

sur1v, 2025-01-29:
Hello @dangowrt, thanks for this quick answer and instructions, much appreciated, I'm very happy to help. I wasn't able to find the offsets, but bellow you will find the logs. Let me know how to proceed, if needed.

```
root@auror4:~# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00080000 00020000 "Preloader"
mtd1: 00040000 00020000 "ATF"
mtd2: 00080000 00020000 "Bootloader"
mtd3: 00080000 00020000 "Config"
mtd4: 00100000 00020000 "Factory"
mtd5: 02600000 00020000 "firmware"
mtd6: 00600000 00020000 "kernel"
mtd7: 02000000 00020000 "ubi"
mtd8: 02600000 00020000 "firmware_backup"
mtd9: 00800000 00020000 "CFG"
mtd10: 00400000 00020000 "RAE"
mtd11: 00100000 00020000 "POT"
mtd12: 00400000 00020000 "Language"
mtd13: 00200000 00020000 "Traffic"
mtd14: 00100000 00020000 "Cert"
mtd15: 00100000 00020000 "NTGRcryptK"
mtd16: 00500000 00020000 "NTGRcryptD"
mtd17: 00100000 00020000 "LOG"
mtd18: 00640000 00020000 "User_data"
root@auror4:~# hexdump -C /dev/mtd4
00000000  22 76 02 00 00 0c 43 26  60 e0 00 00 00 00 00 00  |"v....C&`.......|
00000010  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000030  00 00 00 00 44 00 00 20  00 00 00 00 10 00 20 00  |....D.. ...... .|
00000040  00 00 44 00 04 00 00 00  00 00 00 00 00 00 00 00  |..D.............|
00000050  20 00 00 bb 40 b2 c3 c3  26 81 81 81 41 c4 26 80  | ...@...&...A.&.|
00000060  80 81 41 c4 26 c1 80 80  c0 c5 26 81 81 81 00 00  |..A.&.....&.....|
00000070  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000000b0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 c6 c6  |................|
000000c0  c5 c5 c2 c1 00 c4 00 c4  c3 c1 c1 81 83 c2 c2 c1  |................|
000000d0  81 81 83 83 82 00 00 00  00 00 00 00 00 00 00 00  |................|
000000e0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000000f0  83 00 2e 00 9e 00 8f 87  8b 00 00 00 00 00 00 00  |................|
00000100  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000120  0b 00 00 09 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000130  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000150  00 00 00 00 00 00 00 00  00 00 00 00 00 00 77 07  |..............w.|
00000160  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000220  fc 00 f7 00 fc 00 f7 00  fc 00 f7 00 fd 00 fa a5  |................|
00000230  90 64 42 5d 3e 1e 93 60  2f 00 00 00 00 00 00 00  |.dB]>..`/.......|
00000240  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000400  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00005000  15 79 00 00 00 0c 43 26  46 a0 00 0c 43 26 59 97  |.y....C&F...C&Y.|
00005010  15 79 c3 14 00 80 02 00  15 79 c3 14 ec a0 01 00  |.y.......y......|
00005020  16 79 c3 14 00 80 02 00  16 79 c3 14 00 00 01 00  |.y.......y......|
00005030  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00005040  00 00 00 00 00 00 00 00  00 00 80 00 00 00 00 00  |................|
00005050  01 00 91 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00005060  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00005070  30 1c 05 00 00 00 06 ad  c0 00 00 00 00 00 00 00  |0...............|
00005080  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000050a0  00 00 00 00 00 00 00 00  90 ff 45 77 7e 4a 5f d5  |..........Ew~J_.|
000050b0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00005190  24 24 08 00 2b 00 00 15  00 00 00 00 00 00 00 00  |$$..+...........|
000051a0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000051d0  01 00 fe fd fd fd f8 f9  01 00 fe fd fd fd f8 f9  |................|
[... 137 lines cut ...]
00005c90  f6 f5 ee ec 03 00 fb f8  f7 f4 ed ee 03 00 fb f8  |................|
00005ca0  f7 f4 ed ee 03 00 fb f8  f7 f4 ee ef 03 00 fb f8  |................|
00005cb0  f7 f4 ee ef 03 00 fb f8  f7 f6 ee ed 03 00 fb f8  |................|
00005cc0  f7 f6 ee ed 03 00 fb f8  f8 f5 ef ee 03 00 fb f8  |................|
00005cd0  f8 f5 ef ee 03 00 fc f9  f8 f6 f1 f1 03 00 fc f9  |................|
00005ce0  f8 f6 f1 f1 02 00 fd fa  fa f8 f2 f3 02 00 fd fa  |................|
00005cf0  fa f8 f2 f3 01 00 fd fc  fd fd f7 f6 01 00 fd fc  |................|
00005d00  fd fd f7 f6 02 00 fc f9  f8 f7 f0 ef 02 00 fc f9  |................|
00005d10  f8 f7 f0 ef 02 00 fd fb  f9 f9 f1 ef 02 00 fd fb  |................|
00005d20  f9 f9 f1 ef 02 00 fc fb  f9 fa f4 f3 02 00 fc fb  |................|
00005d30  f9 fa f4 f3 01 00 fd fb  fa fa f2 f2 01 00 fd fb  |................|
00005d40  fa fa f2 f2 00 00 fd fb  fb fa f3 f4 00 00 fd fb  |................|
00005d50  fb fa f3 f4 01 00 fd fc  fb fb f6 f5 01 00 fd fc  |................|
00005d60  fb fb f6 f5 00 00 fd fc  fc fc f7 f5 00 00 fd fc  |................|
00005d70  fc fc f7 f5 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00005d80  de e8 f4 00 0c d9 e1 ed  f5 01 09 11 00 00 00 00  |................|
00005d90  01 00 fc f9 f9 f8 f2 f3  01 00 fc f9 f9 f8 f2 f3  |................|
00005da0  02 00 fd fa fa f8 f2 f3  02 00 fd fa fa f8 f2 f3  |................|
00005db0  00 00 fd fc fc fc f7 f5  00 00 fd fc fc fc f7 f5  |................|
00005dc0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00005e00  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00100000
```

dangowrt, 2025-01-29:
Oh, so that's a bit odd. There are only the WiFi EEPROMs in the `Factory` partition, no MAC addresses. Maybe Netgear has changed that in a later revision of the mass production process.
I suggest to also check look into `Config` and maybe `POT`, watching out for the expected MAC addresses.
If it isn't obvious we won't get around taking a closer look at the vendor firmware...

sur1v, 2025-01-30:
Requested dumps bellow, I think there are some MACs on `Config`
```
root@auror4:~# hexdump -C /dev/mtd3
00000000  e0 e3 58 4c 61 72 63 68  3d 61 72 6d 00 61 74 66  |..XLarch=arm.atf|
00000010  5f 66 69 6c 65 6e 61 6d  65 3d 74 72 75 73 74 7a  |_filename=trustz|
00000020  6f 6e 65 2e 62 69 6e 00  62 61 63 6b 75 70 5f 6f  |one.bin.backup_o|
00000030  73 5f 6e 65 65 64 5f 63  68 65 63 6b 3d 31 00 62  |s_need_check=1.b|
00000040  61 75 64 72 61 74 65 3d  31 31 35 32 30 30 00 62  |audrate=115200.b|
00000050  6f 61 72 64 3d 6d 74 37  36 32 32 5f 65 76 62 00  |oard=mt7622_evb.|
00000060  62 6f 61 72 64 5f 6e 61  6d 65 3d 6d 74 37 36 32  |board_name=mt762|
00000070  32 5f 65 76 62 00 62 6f  6f 74 30 3d 64 6f 77 6e  |2_evb.boot0=down|
00000080  6c 6f 61 64 5f 73 65 74  74 69 6e 67 20 6b 65 72  |load_setting ker|
00000090  6e 65 6c 3b 74 66 74 70  62 6f 6f 74 20 24 7b 6c  |nel;tftpboot ${l|
000000a0  6f 61 64 61 64 64 72 7d  20 24 7b 6b 65 72 6e 65  |oadaddr} ${kerne|
000000b0  6c 5f 66 69 6c 65 6e 61  6d 65 7d 3b 20 62 6f 6f  |l_filename}; boo|
000000c0  74 6d 00 62 6f 6f 74 31  3d 64 6f 77 6e 6c 6f 61  |tm.boot1=downloa|
000000d0  64 5f 73 65 74 74 69 6e  67 20 6b 65 72 6e 65 6c  |d_setting kernel|
000000e0  3b 74 66 74 70 62 6f 6f  74 20 24 7b 6c 6f 61 64  |;tftpboot ${load|
000000f0  61 64 64 72 7d 20 24 7b  6b 65 72 6e 65 6c 5f 66  |addr} ${kernel_f|
00000100  69 6c 65 6e 61 6d 65 7d  3b 72 75 6e 20 62 6f 6f  |ilename};run boo|
00000110  74 5f 77 72 5f 69 6d 67  3b 72 75 6e 20 62 6f 6f  |t_wr_img;run boo|
00000120  74 5f 72 64 5f 69 6d 67  3b 62 6f 6f 74 6d 00 62  |t_rd_img;bootm.b|
00000130  6f 6f 74 31 30 3d 69 66  20 64 69 61 6c 6f 67 20  |oot10=if dialog |
00000140  22 57 41 52 4e 49 4e 47  2c 20 74 68 69 73 20 6f  |"WARNING, this o|
00000150  70 65 72 61 74 69 6f 6e  20 77 69 6c 6c 20 66 6c  |peration will fl|
00000160  61 73 68 20 61 6c 6c 20  70 61 72 74 69 74 69 6f  |ash all partitio|
00000170  6e 73 20 28 70 72 65 6c  6f 61 64 65 72 20 2b 20  |ns (preloader + |
00000180  61 74 66 20 2b 20 75 62  6f 6f 74 20 2b 20 6c 69  |atf + uboot + li|
00000190  6e 75 78 29 22 3b 74 68  65 6e 20 64 6f 77 6e 6c  |nux)";then downl|
000001a0  6f 61 64 5f 73 65 74 74  69 6e 67 20 66 6c 61 73  |oad_setting flas|
000001b0  68 69 6d 61 67 65 3b 74  66 74 70 62 6f 6f 74 20  |himage;tftpboot |
000001c0  24 7b 6c 6f 61 64 61 64  64 72 7d 20 24 7b 66 6c  |${loadaddr} ${fl|
000001d0  61 73 68 69 6d 61 67 65  5f 66 69 6c 65 6e 61 6d  |ashimage_filenam|
000001e0  65 7d 3b 72 75 6e 20 77  72 5f 66 6c 61 73 68 69  |e};run wr_flashi|
000001f0  6d 61 67 65 3b 69 6e 76  61 69 6c 64 5f 65 6e 76  |mage;invaild_env|
00000200  3b 65 6c 73 65 20 65 63  68 6f 20 22 6f 70 65 72  |;else echo "oper|
00000210  61 74 69 6f 6e 20 61 62  6f 72 74 65 64 20 62 79  |ation aborted by|
00000220  20 75 73 65 72 22 3b 66  69 3b 00 62 6f 6f 74 32  | user";fi;.boot2|
00000230  3d 72 75 6e 20 62 6f 6f  74 5f 63 68 65 63 6b 5f  |=run boot_check_|
00000240  62 61 63 6b 75 70 5f 6f  73 3b 72 75 6e 20 62 6f  |backup_os;run bo|
00000250  6f 74 5f 72 64 5f 69 6d  67 3b 62 6f 6f 74 6d 00  |ot_rd_img;bootm.|
00000260  62 6f 6f 74 33 3d 64 6f  77 6e 6c 6f 61 64 5f 73  |boot3=download_s|
00000270  65 74 74 69 6e 67 20 75  62 6f 6f 74 3b 74 66 74  |etting uboot;tft|
00000280  70 62 6f 6f 74 20 24 7b  6c 6f 61 64 61 64 64 72  |pboot ${loadaddr|
00000290  7d 20 24 7b 75 62 6f 6f  74 5f 66 69 6c 65 6e 61  |} ${uboot_filena|
000002a0  6d 65 7d 3b 72 75 6e 20  77 72 5f 75 62 6f 6f 74  |me};run wr_uboot|
000002b0  3b 69 6e 76 61 69 6c 64  5f 65 6e 76 00 62 6f 6f  |;invaild_env.boo|
000002c0  74 34 3d 6c 6f 61 64 62  3b 72 75 6e 20 77 72 5f  |t4=loadb;run wr_|
000002d0  75 62 6f 6f 74 3b 69 6e  76 61 69 6c 64 5f 65 6e  |uboot;invaild_en|
000002e0  76 00 62 6f 6f 74 35 3d  64 6f 77 6e 6c 6f 61 64  |v.boot5=download|
000002f0  5f 73 65 74 74 69 6e 67  20 61 74 66 3b 74 66 74  |_setting atf;tft|
00000300  70 62 6f 6f 74 20 24 7b  6c 6f 61 64 61 64 64 72  |pboot ${loadaddr|
00000310  7d 20 24 7b 61 74 66 5f  66 69 6c 65 6e 61 6d 65  |} ${atf_filename|
00000320  7d 3b 72 75 6e 20 77 72  5f 61 74 66 00 62 6f 6f  |};run wr_atf.boo|
00000330  74 36 3d 64 6f 77 6e 6c  6f 61 64 5f 73 65 74 74  |t6=download_sett|
00000340  69 6e 67 20 70 72 65 6c  6f 61 64 65 72 3b 74 66  |ing preloader;tf|
00000350  74 70 62 6f 6f 74 20 24  7b 6c 6f 61 64 61 64 64  |tpboot ${loadadd|
00000360  72 7d 20 24 7b 70 72 65  6c 6f 61 64 65 72 5f 66  |r} ${preloader_f|
00000370  69 6c 65 6e 61 6d 65 7d  3b 72 75 6e 20 77 72 5f  |ilename};run wr_|
00000380  70 6c 00 62 6f 6f 74 37  3d 64 6f 77 6e 6c 6f 61  |pl.boot7=downloa|
00000390  64 5f 73 65 74 74 69 6e  67 20 68 64 72 3b 74 66  |d_setting hdr;tf|
000003a0  74 70 62 6f 6f 74 20 24  7b 6c 6f 61 64 61 64 64  |tpboot ${loadadd|
000003b0  72 7d 20 24 7b 68 64 72  5f 66 69 6c 65 6e 61 6d  |r} ${hdr_filenam|
000003c0  65 7d 3b 72 75 6e 20 77  72 5f 72 6f 6d 5f 68 64  |e};run wr_rom_hd|
000003d0  72 00 62 6f 6f 74 38 3d  64 6f 77 6e 6c 6f 61 64  |r.boot8=download|
000003e0  5f 73 65 74 74 69 6e 67  20 63 74 70 3b 74 66 74  |_setting ctp;tft|
000003f0  70 62 6f 6f 74 20 24 7b  6c 6f 61 64 61 64 64 72  |pboot ${loadaddr|
00000400  7d 20 24 7b 63 74 70 5f  66 69 6c 65 6e 61 6d 65  |} ${ctp_filename|
00000410  7d 3b 72 75 6e 20 77 72  5f 63 74 70 00 62 6f 6f  |};run wr_ctp.boo|
00000420  74 39 3d 72 75 6e 20 62  6f 6f 74 5f 72 64 5f 63  |t9=run boot_rd_c|
[... 261 lines cut ...]
00020040  6e 3d 32 34 31 42 30 43  38 36 30 34 42 41 00 6d  |n=241B0C8604BA.m|
[... cut ...]
00028000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00080000
root@auror4:~# hexdump -C /dev/mtd11
00000000  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000210  00 00 00 00 00 00 00 00  00 00 00 00 ff ff ff ff  |................|
00000220  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00000800  f3 2d 28 64 00 11 22 33  44 55 ff ff ff ff ff ff  |.-(d.."3DU......|
00000810  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00100000
```