openwrt/openwrt #17787: AVM FritzBox 7360 SL: missing "Internet"-LED

misery, 2025-01-30:
### Describe the bug

I have the 7360 HWRev 183 here. It is missing the configuration for the "internet"-LED. As you can see on the image it has a LED for "Internet" but it is not possible to configure something to it.

https://www.amazon.de/AVM-Router-DECT-Basis-geeignet-Deutschland/dp/B008K2WQF2

### OpenWrt version

r28417-daef29c75d

### OpenWrt release

24.10.0-rc7

### OpenWrt target/subtarget

lantiq/xrx200

### Device

AVM FRITZ!Box 7360 SL

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

maurerle, 2025-02-01:
The device shares a device tree source interface (dtsi) with the 7360 and 7362sl:

https://github.com/openwrt/openwrt/blob/openwrt-24.10/target/linux/lantiq/files-5.4/arch/mips/boot/dts/lantiq/vr9_avm_fritz736x.dtsi

Here, the internet LED is missing.
Can you show what you this command returns for you:

`ls /sys/devices/platform/leds/leds/`

misery, 2025-02-01:
Here it is 🙂

$ ls /sys/devices/platform/leds/leds/
green:dect   green:info   green:power  green:wlan   red:info     red:power

nicefile, 2025-02-19:
it's not a bug is a lack of the feature. This LED was never declared in .dts   .Since I own and use  7362SL this LED doesn't work for me either

nicefile, 2025-03-05:
So I've tested this myself and finally got green glow under FON led.  I don't know how i could live without it for so long ;)