openwrt/openwrt #17789: Flashing recent OpenWRT Snapshot on WRT32X fails.

wriggerz, 2025-01-30:
### Describe the bug

https://forum.openwrt.org/t/flashing-wrt32x-failing/223182

As posted here, tried flashing router with latest snapshot build, as of 29/01, 11pm GMT.

Router fails to boot.

Cloned repo has fixes in place for previous similar issue.

ghoffman has also tested and reports that flashing to an earlier build works, but not to r28724.

Let me know what I can do to assist with logs etc. First time posting a bug so not sure what you'll need to investigate.

Thanks

### OpenWrt version

SNAPSHOT

### OpenWrt release

r28724 

### OpenWrt target/subtarget

mvebu / cortexa9 / venom

### Device

WRT32X

### Image kind

Self-built image

### Steps to reproduce

Attempt to update to a recent SNAPSHOT build on WRT32X.

### Actual behaviour

Router reboots and becomes inaccessible, not sure whats going on under the hood at that point.

3 power cycles to fail back to other image.

### Expected behaviour

Router reboots and allows me to connect to configure.

### Additional info

Compiling completes without issue, but after flashing the router it becomes inaccessible. I've tried SYSUPGRADE with and without keeping settings, and I tried the FACTORY image, but that doesn't work either.

Luckily these routers have a dual flash so I can fail back fairly easily.

I came across another post which sounded a lot like my issue: [Sysupgrade not flashing (WRT3200) - #12 by TheBeard](https://forum.openwrt.org/t/sysupgrade-not-flashing-wrt3200/222743/12)

But after checking both issues linked in that thread, I checked the corresponding files and can see that those changes are already present in my cloned repo.

r28676 -> r28695 - this succeeded
r28695 -> r28724 - this failed; device hung; power cycle X3 -> fall back ok

update:
r28695 -> r28676 (backwards) succeeds, so i think the issue is with recent builds after r28695

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

hnyman, 2025-01-30:
First guess: commit eaa8211 from @linusw  adds a needed new module for wrt3200acm package list, but not for wrt32x, although they should have practically identical hardware.

Wrt32x .dts refers to "armada-385-linksys.dtsi" which defined the affected switch.

Adding kmod-dsa-mv88e6xxx to wrt32x might be the correct solution?

Ps there might be other similar routers where the grey mentioned in the commit did not find the switch reference due to inclusion via .dtsi

wriggerz, 2025-01-30:
Just compiled a new image including kmod-dsa-mv88e6xxx and it's working.

ghoffman9, 2025-01-30:
i originally reportedd this on openwrt forum and confirm the above findings for main - snapshot wrt32x r28724.

ghoffman9, 2025-01-30:
i can confirm the bug on wrt32x, and the solution by including the kmod.   i am the one who reported the problem originating after r28695; firmware build from firmware-selector.

it would be better to include wrt32x in the  list of devices in the .dtsi to include the module rather than have to manually add it to the build via firmware-selector.

ghoffman9, 2025-01-31:
still broken in r28735

hnyman, 2025-01-31:
There is now PR #17793

hauke, 2025-02-02:
Fixed in https://github.com/openwrt/openwrt/pull/17793

closed 2025-02-02 by hauke