openwrt/openwrt #17795: redmi AX6000 /lib/upgrade/do_stage2: line 92: fit_do_upgrade: not found

ozbenh, 2025-01-31:
### Describe the bug

Hi !

I have a Xiaomi RedMi AX6000 rooted and converted to uboot flash layout (I did that with 23.05.5). I later successfully sysupgraded it to 24.10-rc2. However it won't upgrade to anything else after that.

Any attempt at sysupgrade (I tried a few, using either images generated by unattended sysupgrade or the plain https://downloads.openwrt.org/releases/24.10.0-rc7/targets/mediatek/filogic/openwrt-24.10.0-rc7-mediatek-filogic-xiaomi_redmi-router-ax6000-ubootmod-squashfs-sysupgrade.itb with the same result: the router reboots to do the upgrade and comes back with the same 24.10-rc2.

I eventually bit the bullet and soldered a header for the serial port and here it what it says:
```
Fri Jan 31 03:51:23 UTC 2025 upgrade: Switching to ramdisk...
<5>[  336.695844] UBIFS (ubi0:4): background thread "ubifs_bgt0_4" stops
<5>[  336.708788] UBIFS (ubi0:4): un-mount UBI device 0
Fri Jan 31 03:51:24 UTC 2025 upgrade: Performing system upgrade...
/lib/upgrade/do_stage2: line 92: fit_do_upgrade: not found
Fri Jan 31 03:51:24 UTC 2025 upgrade: Upgrade completed
Fri Jan 31 03:51:25 UTC 2025 upgrade: Rebooting system...
umoun<0>[  337.743538] reboot: Restarting system
```

Just in case I updated uboot preloader and bl31 to the latest from the 24.10-rc7 downloads, with the same results.

Any idea what's wrong ? Anything I can do to beat it back into shape ?

### OpenWrt version

r28161-ea17e958b9

### OpenWrt release

24.10.0-rc2

### OpenWrt target/subtarget

mediatek/filogic

### Device

Xiaomi Redmi Router AX6000 (OpenWrt U-Boot layout)

### Image kind

Official downloaded image

### Steps to reproduce

(see description)

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

ozbenh, 2025-01-31:
I managed to boot into recovery (using the reset button) which is a 23.05.5 based recovery. From there, I tried sysupgrad'ing to the -rc7 image. Now it just goes back into recovery. I tried using recovery to sysupgrade to any image, it just goes back to recovery on reboot. Not sure what's up... Unfortunately, the serial port RX doesn't seem to be working (it gets garbled), so I can see the output from the modem but not type in uboot

ozbenh, 2025-01-31:
Allright, I got it back ... pfiew ! From recovery, I reformatted the whole ubi (/dev/mtd5) and redid the partitions as per the wiki instructions (the ones for converting from stock to ubootmod layout). Then sysupgraded to install rc7 and that worked. A subsequent sysupgrade to get my "unattended sysupgrade" image with my extra packages worked too, along with restoring my settings.

It's back !

I'm going to leave this open in case there's still a bug lurking there that you guys might find from reading this, but feel free to close if not.

remynicol, 2025-03-13:
Same bug reproducted and same solution.

Details here : #18231

ozbenh, 2025-05-29:
It's broken again, I can't update from 24.10-rc7 to 24.10.1 ...

ozbenh, 2025-05-29:
So following https://github.com/openwrt/openwrt/issues/12625 I verified that installing `fitblk` first then doing the attended sysupgrade resulted in a successful upgrade to 24.10.1 :-)