openwrt/openwrt #17800: ssh stuck at expecting SSH2_MSG_KEX_ECDH_REPLY

getreu, 2025-01-31:
### Describe the bug

Starting from rc6 and rc7 (I did not observe this with rc5), I could not connect to remote ssh servers any more. See also #17759.
The ssh client got stuck at expecting SSH2_MSG_KEX_ECDH_REPLY. 

I found [a post describing my problem](https://unix.stackexchange.com/questions/722954/ssh-stuck-at-expecting-ssh2-msg-kex-ecdh-reply) and the workaround as well as the solutions described worked for me:

**1. Workaround**

Add `-o KexAlgorithms=curve25519-sha256`:

```sh 
ssh -l root <server> -o KexAlgorithms=curve25519-sha256
```
    
**2. Solution**

Modify firewall rules 

Add the following message types to the "Allow-ICMPv6-Input"
and "Allo-ICMPv6-Forward" rules:

* `fragmentation-needed`, `certification-path-solicitation-message`, `certification-path-advertisement-message`

1. Delete the "Allow-ICMPv6-Input" and "Allo-ICMPv6-Forward" rules.
 
2. Recreate "Allow-ICMPv6-Input": 

   ```
   uci add firewall rule
   uci set firewall.@rule[-1].name='Allow-ICMPv6-Input'
   uci set firewall.@rule[-1].src='wan'
   uci set firewall.@rule[-1].proto='icmp'
   uci set firewall.@rule[-1].limit='1000/sec'
   uci set firewall.@rule[-1].family='ipv6'
   uci set firewall.@rule[-1].target='ACCEPT'
   uci add_list firewall.@rule[-1].icmp_type='bad-header'
   uci add_list firewall.@rule[-1].icmp_type='certification-path-solicitation-message'
   uci add_list firewall.@rule[-1].icmp_type='certification-path-advertisement-message'
   uci add_list firewall.@rule[-1].icmp_type='destination-unreachable'
   uci add_list firewall.@rule[-1].icmp_type='echo-reply'
   uci add_list firewall.@rule[-1].icmp_type='echo-request'
   uci add_list firewall.@rule[-1].icmp_type='fragmentation-needed'
   uci add_list firewall.@rule[-1].icmp_type='neighbour-advertisement'
   uci add_list firewall.@rule[-1].icmp_type='neighbour-solicitation'
   uci add_list firewall.@rule[-1].icmp_type='packet-too-big'
   uci add_list firewall.@rule[-1].icmp_type='router-advertisement'
   uci add_list firewall.@rule[-1].icmp_type='router-solicitation'
   uci add_list firewall.@rule[-1].icmp_type='time-exceeded'
   uci add_list firewall.@rule[-1].icmp_type='unknown-header-type'
   ```
   

3. Recreate "Allow-ICMPv6-Forward": 
   
   ```
   uci add firewall rule
   uci set firewall.@rule[-1].name='Allow-ICMPv6-Forward'
   uci set firewall.@rule[-1].src='wan'
   uci set firewall.@rule[-1].dest='*'
   uci set firewall.@rule[-1].proto='icmp'
   uci set firewall.@rule[-1].limit='1000/sec'
   uci set firewall.@rule[-1].family='ipv6'
   uci set firewall.@rule[-1].target='ACCEPT'
   uci add_list firewall.@rule[-1].icmp_type='bad-header'
   uci add_list firewall.@rule[-1].icmp_type='certification-path-solicitation-message'
   uci add_list firewall.@rule[-1].icmp_type='certification-path-advertisement-message'
   uci add_list firewall.@rule[-1].icmp_type='destination-unreachable'
   uci add_list firewall.@rule[-1].icmp_type='echo-reply'
   uci add_list firewall.@rule[-1].icmp_type='echo-request'
   uci add_list firewall.@rule[-1].icmp_type='fragmentation-needed'
   uci add_list firewall.@rule[-1].icmp_type='packet-too-big'
   uci add_list firewall.@rule[-1].icmp_type='time-exceeded'
   uci add_list firewall.@rule[-1].icmp_type='unknown-header-type'
   ```

**Question/Suggestion**

As `ssh` is a common use case, the firewall rules should not hinder reliable connection. Maybe some more ICMP type should be allowed by default?

### OpenWrt version

r28417-daef29c75d

### OpenWrt release

24.10.0-rc7

### OpenWrt target/subtarget

mediatek/filogic

### Device

Xiaomi Mi Router AX3000T (OpenWrt U-Boot layout)

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-01-31:
Wanna make an enhancement PR to both firewalls?

brada4, 2025-01-31:
Default file nowadays comes via ../firewall4 , (?} me add it there then ? carry over here. Spotted another, rule falls through to reject spilling over rationed rate

brada4, 2025-01-31:
The problem is a bit deeper, you get tcp rst whereas if default target was drop youd get just pmtu blackhole recovery ie connect afyer a hickup.

getreu, 2025-02-07:
If you are reading this, please have a look at PR #17895 also. It did not resolve the problem, but revealed some interesting aspects of it.

brada4, 2025-02-07:
Are you saved by getting back to 23.05? Also if there check the 2 pedantic rules in other linked pr.

getreu, 2025-02-07:
Thank you for asking. As far as I can tell, this bug affects only one of my projects: [Setting up an IPv6 only home network with OpenWrt](https://blog.getreu.net/20250121-IPv6-home-network/)
It is a very fun project and nice to share. I guess I should mention this bug to my readers. A bit frustrating is, that the problems manifest so differently, e.g. ssh connection, Firefox and Chromium can not connect via IPv6 etc. If I hadn't started this with rc5, I would have doubted the stability of Jol and Unbound. Rc5 works like a clockwork. As always, it is a good opportunity to learn.

brada4, 2025-02-07:
I understood that, I will try to replicate strange logged packets, but not today.

getreu, 2025-02-09:
I have good news. I restarted from zero and set up everything from the start with 24.10.0. Everything works and my problem is (hopefully) solved. Thank you for your help!
I leave this open, because there is still the problem with the strangely logged packages.

brada4, 2025-02-10:
Reported upstream
https://bugzilla.netfilter.org/show_bug.cgi?id=1790

brada4, 2025-02-12:
You can permit jool traffic with absent input interface name by adding following to forward chain via /etc/nftables.d/*.nft :
oifname vmap { "jool" : accept , "openwrt" : accept }

getreu, 2025-02-12:
Thank you for your tip! At the moment all works fine, but I keep this in mind in case the connection problems reappear.

getreu, 2025-02-16:
I found the file: 

    /etc/nftables.d# vi 10-custom-filter-chains.nft

It has 2 forward entries:

```
# chain user_pre_forward {
#    type filter hook forward priority -1; policy accept;
#}

#chain user_post_forward {
#    type filter hook forward priority 1; policy accept;
#    ct state new log prefix "Firewall4 accepted forward: "
#}
```

Do you suggest the following?

```
chain user_pre_forward {
   type filter hook forward priority -1; policy accept;
   oifname vmap { "jool" : accept , "openwrt" : accept }
}

#chain user_post_forward {
#    type filter hook forward priority 1; policy accept;
#    ct state new log prefix "Firewall4 accepted forward: "
#}
```

getreu, 2025-02-17:
Like this?

```
chain user_pre_forward {
   type filter hook forward priority -1;  oifname vmap { "jool" : accept , "openwrt" : accept }
   type filter hook forward priority -1; policy accept;
}

#chain user_post_forward {
#    type filter hook forward priority 1; policy accept;
#    ct state new log prefix "Firewall4 accepted forward: "
#}
```

PS: I have no idea what I am doing. I am not familiar with semantics and syntax.  Could you please correct the draft?

brada4, 2025-02-17:
Then dont touch it and remove jool namespace support.

getreu, 2025-02-17:
> Then dont touch it and remove jool namespace support.

???

getreu, 2025-02-20:
@brada4 If I understand right 

     oifname vmap { "jool" : accept , "openwrt" : accept }

is a workaround and the underlying problem is that the `IN=` tag is missing. Who is the perpetrator? Jool? Is there a chance that the problem gets fixed (any time soon)?

brada4, 2025-02-20:
netfilter blames other thing, so it is jool via namespace veth link. I have no elegant idea how to isolate one at a time.

getreu, 2025-02-21:
Shall we file a bug report in Jool?

brada4, 2025-02-21:
Try jool without netns, if you get input prerouting forward with empty source iifname - it is jool.
Also for upstream reports try to reconstruct issue in some desktop linux like debian, good luck.

getreu, 2025-02-24:
According to your experience, do you see anything missing in the interface setup? BTW, I noticed the missing litteral `name` in `ip link add jool type veth peer openwrt`. I do not know if this matters.

```sh
#!/bin/sh
ip link add jool type veth peer openwrt
ip netns add jool
ip link set dev openwrt netns jool
ip netns exec jool sh <<EOF
    sysctl -w net.ipv4.conf.all.forwarding=1
    sysctl -w net.ipv6.conf.all.forwarding=1
    sysctl -w net.ipv6.conf.openwrt.accept_ra=2
    sysctl -w net.ipv4.ip_local_port_range="32768 32999"
    ip link set dev lo up
    ip link set dev openwrt up
    ip addr add dev openwrt 192.168.164.2/24
    ip addr add dev openwrt fe80::64
    ip route add default via 192.168.164.1
    modprobe jool
    jool instance add --netfilter --pool6 64:ff9b::/96
    jool global update lowest-ipv6-mtu 1500
    jool pool4 add 192.168.164.2 33000-65535 --tcp
    jool pool4 add 192.168.164.2 33000-65535 --udp
    jool pool4 add 192.168.164.2 33000-65535 --icmp
EOF
```

brada4, 2025-02-24:
netns alone does not create packets with broken iifname, checked with nfs and wireguard.

getreu, 2025-04-05:
I installed the packages  `swconfig` and  `kmod-swconfig` and I can `ssh <server>` without `-o KexAlgorithms=curve25519-sha256`.

brada4, 2025-04-29:
Try the following:

First (in both sides of namespace link)
sysctl net.ipv4.icmp_errors_use_inbound_ifaddr=1

Then try to add fragmentation-needed to icmp6 accepted types (it is not seen on the internet, it is 46 mis-translation by jool)

Add something.nft to /etc/nftables.d/
```
chain forward {
  type filter hook forward priority filter; policy drop;
  iifname "potato" counter accept
}
```
simply null if name passed iifname lo match but not iif lo. Since packets match any iifname you can also use broccoli if you like that more.

Dont blame me if it gets fixed one day.

getreu, 2025-04-29:
This did pass:

```sh
# sysctl net.ipv4.icmp_errors_use_inbound_ifaddr=1
net.ipv4.icmp_errors_use_inbound_ifaddr = 1
#
```

and your rule was inserted too:

```nft
table inet fw4 {
	chain forward {
		type filter hook forward priority filter; policy drop;
		iifname "potatos" counter packets 0 bytes 0 accept
		ct state vmap { established : accept, related : accept } comment "!fw4: Handle forwarded flows"
		iifname "br-lan" jump forward_lan comment "!fw4: Handle lan IPv4/IPv6 forward traffic"
		iifname { "wan", "pppoe-wan" } jump forward_wan comment "!fw4: Handle wan IPv4/IPv6 forward traffic"
		iifname "jool" jump forward_jool comment "!fw4: Handle jool IPv4/IPv6 forward traffic"
		meta nfproto ipv6 iifname "br-lan64" jump forward_lan64 comment "!fw4: Handle lan64 IPv6 forward traffic"
		jump handle_reject
	}
```

I added `fragmentation-needed` to:

    Firewall -> Traffic Rules -> Allow-ICMPv6-Input
    Firewall -> Traffic Rules -> Allow-ICMPv6-Forward

Unfortunately I could not observe any improvement.

BTW: what helps is to reduce the MTU on `lan` and `lan64` to `1450`.

brada4, 2025-04-29:
There is a bug in jool-namespace-veth-nftables chain to be determined, keep potato until it is fixed for good.

getreu, 2025-04-30:
@brada4 : I added your workaround at the end of the  [Configure Joo sectionl](https://blog.getreu.net/20250121-IPv6-home-network/#configure-jool) in the article [Setting up an IPv6 only home network with OpenWrt](https://blog.getreu.net/20250121-IPv6-home-network/). In the text I call it _the empty IN-tag workaround_. Thank you for finding and suggesting this workaround!

brada4, 2025-04-30:
Well, the workaround abuses what I think is a bug, this issue can be closed for now as it is not clear how empty iifname should actually work. Come back if something changes (ie potato rule matches no packet in new kernel/jool)

getreu, 2025-05-01:
Are you aware of an upstream follow up of this bug? Then we can leave a reference here.  I told that one attempt was rejected.

brada4, 2025-05-01:
Here
https://github.com/openwrt/openwrt/issues/17800#issuecomment-2647272051

getreu, 2025-05-01:
Oh, the status of this does not look promising: 

> <https://bugzilla.netfilter.org/show_bug.cgi?id=1790>
> WORKSFORME
> All attempts at reproducing this bug were futile, and reading the code produces no clues as to why the described behavior would occur. If more information appears later, the bug can be reopened. 

I think that's all that can be done at the moment. The issue is not really fixed, but we have a good workaround easy to implement and which should not have any security impact.

closed 2025-05-01 by getreu

brada4, 2025-05-01:
Kind of i feel 1c at fault by cutting away half of bug pair that "works"
https://github.com/openwrt/firewall4/commit/a5553dae70439c7e4fa910490fcf12a1ffff5bd2