openwrt/openwrt #17802: ../src/mod_wolfssl.c.924) SSL: couldn't read X509 certificates

mic-user, 2025-01-31:
### Describe the bug

wolfssl block to load lighttpd as below
`# service lighttpd start
2025-01-31 21:35:07: (../src/mod_wolfssl.c.924) SSL: couldn't read X509 certificates from '/etc/fullchain.pem'
Illegal instruction
lighttpd.conf validation failed
`

### OpenWrt version

r28417-daef29c75d

### OpenWrt release

24.10.0-rc7

### OpenWrt target/subtarget

mvebu/cortexa9

### Device

Linksys WRT32X

### Image kind

Self-built image

### Steps to reproduce

install lighttpd wolfssl mod
service lighttpd start

### Actual behaviour

2025-01-31 21:35:07: (../src/mod_wolfssl.c.924) SSL: couldn't read X509 certificates from '/etc/fullchain.pem'
Illegal instruction
lighttpd.conf validation failed

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

gstrauss, 2025-02-01:
Between r6 and r7, wolfssl was upgraded from wolfssl 5.7.2 to 5.7.6, and lighttpd was upgraded from 1.4.76 to 1.4.77.

`2025-01-31 21:35:07: (../src/mod_wolfssl.c.924) SSL: couldn't read X509 certificates from '/etc/fullchain.pem'`
The code in lighttpd mod_wolfssl has not changed for loading certificates, so please look at your `/etc/fullchain.pem`, and perhaps attach it here (as long as it does not contain any private keys).  If you can, please test with wolfssl 5.7.2.

`Illegal instruction`
This appears to me to be a separate issue.

(I am a lighttpd developer, but am sorry that I will not have further time for this at the moment.  I'll follow this and may have time in another week or so.)

In the meantime, I would recommend https://wiki.lighttpd.net/Docs_SSL since your lighttpd.conf would probably benefit from specifying `ssl.pemfile = "/etc/fullchain.pem"` and avoiding the use of `ssl.ca-file` unless you are using it to verify client certificates, in which case you should use `ssl.verifyclient.ca-file`.

mic-user, 2025-02-02:
> Between r6 and r7, wolfssl was upgraded from wolfssl 5.7.2 to 5.7.6, and lighttpd was upgraded from 1.4.76 to 1.4.77.
>
> `2025-01-31 21:35:07: (../src/mod_wolfssl.c.924) SSL: couldn't read X509 [...]

Hi. Quick reply here.
- The pem include private key(public+private cat file) which came from 'Let's Encrypt' by the Dehydrated.
- At this moment, using r6 wolfssl which nornally operating with below combination.
`Package name	Version
libwolfssl5.7.2.e624513f	 == 5.7.2-stable-r1
libustream-wolfssl20201210	== 2024.07.28~99bd3d2b-r1
lighttpd-mod-wolfssl	== 1.4.76-r2
wpad-basic-wolfssl	== 2024.09.15~5ace39b0-r2
`
- As your comment, I have changed the `ssl.ca-file(which still working)` to `ssl.verifyclient.ca-file` on 1.4.67(in rc6) not 1.4.77(on rc7).
- seems working. But very slow to be loaded(very slow/delay `service lighttpd start`)
(for now can't test it on rc7 chased that life matters, but later)

gstrauss, 2025-02-02:
Thank you for confirming the details.
      
>     * As your comment, I have changed the `ssl.ca-file(which still working)` to `ssl.verifyclient.ca-file` on 1.4.67(in rc6)  [Ed: 1.4.76 typo] not 1.4.77(on rc7).
>       seems working.

FYI: I renamed `ssl.ca-file` to `ssl.verifyclient.ca-file` in lighttpd 1.4.60 to reflect its actual use, and documented the openssl misfeature of using the CA list for client certificate verification to build the server certficate chain in https://wiki.lighttpd.net/Docs_SSL.  Yes, both lighttpd.conf directives still work (and one aliases the other), but only for backwards compatibility.

However, if you are not using client certificate verification (``ssl.verifyclient.activate = "enabled"`), you should use lighttpd.conf `ssl.pemfile = "/etc/fullchain.pem"` and omit `ssl.verifyclient.ca-file`.

>       But very slow to be loaded(very slow/delay `service lighttpd start`)

The slowdown is very likely the random number generation initialization in wolfssl.  lighttpd takes pains to defer random number generation initialization in lighttpd so that lighttpd startup is faster, but the underlying TLS libraries have their own RNG initialization procedures.  => You won't see this startup delay in lighttpd if you use lighttpd mod_mbedtls instead of lighttpd mod_wolfssl.

FYI: modern versions of openwrt default to lighttpd mod_mbedtls for TLS support.  If you have been updating from older versions of openwrt, you might consider reconfiguring lighttpd to use lighttpd mod_mbedtls.

mic-user, 2025-02-02:
@gstrauss Thanks for the tips.
And...need your advice that is the 'mod_mbedtls' support mod_wolfssl 's full features?

mic-user, 2025-02-02:
@gstrauss 
Since you told me that mbedtls working, 
I have rebuild a rc7 image quickly and upload it.
the lighttpd still not working(whole backup restored/uploaded)

# /usr/sbin/lighttpd -D -f /etc/lighttpd/lighttpd.conf && cat /var/log/lighttpd/error.log
2025-02-02 13:01:14: (../src/plugin.c.221) dlopen() failed for: /usr/lib/lighttpd/mod_wolfssl.so Error loading shared library /usr/lib/lighttpd/mod_wolfssl.so: No such file or directory
2025-02-02 13:01:14: (../src/server.c.1649) loading plugins finally failed

There is no wolfssl mod installed
and there is no wolfssl conf  file. where should I check?

gstrauss, 2025-02-02:
> And...need your advice that is the 'mod_mbedtls' support mod_wolfssl 's full features?

lighttpd mod_mbedtls and lighttpd mod_wolfssl have similar TLS support and features.

The only real difference between the TLS support *was* that mbedtls did not support TLSv1.3 well before mbedtls 3.6.1; mbedtls 3.6.2 is in openwrt 24.10.

> There is no wolfssl mod installed and there is no wolfssl conf file. where should I check?

`grep -l wolfssl -r /etc/lighttpd`

/etc/lighttpd/lighttpd.conf includes other files under /etc/lighttpd/conf.d/*.conf and you'll probably find `server.modules += ("mod_wolfssl")` in there.  You need to change that to `"mod_mbedtls"`.

mic-user, 2025-02-02:
@gstrauss 
Thanks for the script. Found and rid the duplicate line so that zero wolfssl line.
But,
```
# /usr/sbin/lighttpd -tt -f /etc/lighttpd/lighttpd.conf
Segmentation fault
```

rc7 with mbedtls

mic-user, 2025-02-02:
I have to rollback to rc6 that include wolfssl.
Can't do this for now more.

```
Sun Feb  2 13:34:18 2025 mail.info msmtp: host=smtp.gmail.com tls=on auth=on user=npdmailing@gmail.com from=npdmailing@gmail.com recipients=npdmailing@gmail.com mailsize=373 smtpstatus=250 smtpmsg='250 2.0.0 OK  1738470858 98e67ed59e1d1-2f83bc97da0sm8614699a91.3 - gsmtp' exitcode=EX_OK
Sun Feb  2 13:34:20 2025 daemon.err smbd[7572]: [2025/02/02 13:34:20.209849,  0] ../../source3/smbd/smb2_service.c:746(make_connection_snum)
Sun Feb  2 13:34:20 2025 daemon.err smbd[7572]:   make_connection_snum: canonicalize_connect_path failed for service MediaEndPoint, path /mnt/nas_root
Sun Feb  2 13:34:20 2025 daemon.err smbd[7572]: [2025/02/02 13:34:20.213078,  0] ../../source3/smbd/smb2_service.c:746(make_connection_snum)
Sun Feb  2 13:34:20 2025 daemon.err smbd[7572]:   make_connection_snum: canonicalize_connect_path failed for service MediaEndPoint, path /mnt/nas_root
Sun Feb  2 13:34:20 2025 daemon.err smbd[7572]: [2025/02/02 13:34:20.215939,  0] ../../source3/smbd/smb2_service.c:746(make_connection_snum)
Sun Feb  2 13:34:20 2025 daemon.err smbd[7572]:   make_connection_snum: canonicalize_connect_path failed for service MediaEndPoint, path /mnt/nas_root
Sun Feb  2 13:34:20 2025 daemon.err smbd[7572]: [2025/02/02 13:34:20.218715,  0] ../../source3/smbd/smb2_service.c:746(make_connection_snum)
Sun Feb  2 13:34:20 2025 daemon.err smbd[7572]:   make_connection_snum: canonicalize_connect_path failed for service MediaEndPoint, path /mnt/nas_root
Sun Feb  2 13:34:20 2025 daemon.err smbd[7572]: [2025/02/02 13:34:20.221721,  0] ../../source3/smbd/smb2_service.c:746(make_connection_snum)
Sun Feb  2 13:34:20 2025 daemon.err smbd[7572]:   make_connection_snum: canonicalize_connect_path failed for service MediaEndPoint, path /mnt/nas_root
Sun Feb  2 13:34:20 2025 daemon.err smbd[7572]: [2025/02/02 13:34:20.224518,  0] ../../source3/smbd/smb2_service.c:746(make_connection_snum)
Sun Feb  2 13:34:20 2025 daemon.err smbd[7572]:   make_connection_snum: canonicalize_connect_path failed for service MediaEndPoint, path /mnt/nas_root
Sun Feb  2 13:34:20 2025 daemon.err smbd[7572]: [2025/02/02 13:34:20.227317,  0] ../../source3/smbd/smb2_service.c:746(make_connection_snum)
Sun Feb  2 13:34:20 2025 daemon.err smbd[7572]:   make_connection_snum: canonicalize_connect_path failed for service MediaEndPoint, path /mnt/nas_root
Sun Feb  2 13:34:20 2025 daemon.err smbd[7572]: [2025/02/02 13:34:20.230124,  0] ../../source3/smbd/smb2_service.c:746(make_connection_snum)
Sun Feb  2 13:34:20 2025 daemon.err smbd[7572]:   make_connection_snum: canonicalize_connect_path failed for service MediaEndPoint, path /mnt/nas_root
Sun Feb  2 13:34:20 2025 daemon.err smbd[7572]: [2025/02/02 13:34:20.233452,  0] ../../source3/smbd/smb2_service.c:746(make_connection_snum)
Sun Feb  2 13:34:20 2025 daemon.err smbd[7572]:   make_connection_snum: canonicalize_connect_path failed for service MediaEndPoint, path /mnt/nas_root
Sun Feb  2 13:34:20 2025 daemon.err smbd[7572]: [2025/02/02 13:34:20.236261,  0] ../../source3/smbd/smb2_service.c:746(make_connection_snum)
Sun Feb  2 13:34:20 2025 daemon.err smbd[7572]:   make_connection_snum: canonicalize_connect_path failed for service MediaEndPoint, path /mnt/nas_root
Sun Feb  2 13:34:20 2025 daemon.err smbd[7572]: [2025/02/02 13:34:20.239084,  0] ../../source3/smbd/smb2_service.c:746(make_connection_snum)
Sun Feb  2 13:34:20 2025 daemon.err smbd[7572]:   make_connection_snum: canonicalize_connect_path failed for service MediaEndPoint, path /mnt/nas_root
Sun Feb  2 13:34:20 2025 daemon.err smbd[7572]: [2025/02/02 13:34:20.241843,  0] ../../source3/smbd/smb2_service.c:746(make_connection_snum)
Sun Feb  2 13:34:20 2025 daemon.err smbd[7572]:   make_connection_snum: canonicalize_connect_path failed for service MediaEndPoint, path /mnt/nas_root
Sun Feb  2 13:34:24 2025 daemon.info procd: Instance fail2ban::instance1 s in a crash loop 6 crashes, 1 seconds since last crash
Sun Feb  2 13:35:10 2025 daemon.notice hostapd: phy2-ap0: DFS-CAC-COMPLETED success=1 freq=5260 ht_enabled=1 chan_offset=1 chan_width=2 cf1=5270 cf2=0 radar_detected=0
Sun Feb  2 13:35:10 2025 kern.info kernel: [  115.693946] mwifiex_sdio mmc0:0001:1: CAC timer finished; No radar detected
Sun Feb  2 13:35:10 2025 kern.warn kernel: [  115.713293] ------------[ cut here ]------------
Sun Feb  2 13:35:10 2025 kern.warn kernel: [  115.717951] WARNING: CPU: 1 PID: 2640 at backports-6.12.6/drivers/net/wireless/marvell/mwifiex/sta_cmd.c:1048 0xbf385b3c [mwifiex@79df89e0+0x2c000]
Sun Feb  2 13:35:10 2025 kern.warn kernel: [  115.731268] memcpy: detected field-spanning write (size 84) of single field "domain->triplet" at backports-6.12.6/drivers/net/wireless/marvell/mwifiex/sta_cmd.c:1048 (size 3)
Sun Feb  2 13:35:10 2025 kern.warn kernel: [  115.746911] Modules linked in: nft_fib_inet nf_flow_table_inet nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack_netlink nf_conntrack iptable_mangle iptable_filter ipt_REJECT ipt_ECN ip_tables xt_time xt_tcpudp xt_tcpmss xt_statistic xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY x_tables sch_cake rfcomm nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_ [... cut ...]
Sun Feb  2 13:35:10 2025 kern.warn kernel: [  115.975550] Exception stack(0xdab5bfa8 to 0xdab5bff0)
Sun Feb  2 13:35:10 2025 kern.warn kernel: [  115.980623] bfa0:                   00000000 00000000 00000019 befee4e0 00000000 00000000
Sun Feb  2 13:35:10 2025 kern.warn kernel: [  115.988838] bfc0: 00000000 00000000 b6d2bcb0 00000128 00000004 befee530 00000001 00000000
Sun Feb  2 13:35:10 2025 kern.warn kernel: [  115.997052] bfe0: befee490 befee480 b6efa9cc b6ef9cec
Sun Feb  2 13:35:10 2025 kern.warn kernel: [  116.002177] ---[ end trace 0000000000000000 ]---

```

gstrauss, 2025-02-02:
```
# /usr/sbin/lighttpd -tt -f /etc/lighttpd/lighttpd.conf
Segmentation fault
```
Well, that's not good.  If you can get an `strace /usr/sbin/lighttpd -tt -f /etc/lighttpd/lighttpd.conf` that might be helpful.  I won't have access to some test hardware for another week, and will try to repro then.

mic-user, 2025-02-02:
@gstrauss (Is that a way to payback contribution to opensource community, I need to do.)

```
execve("/usr/sbin/lighttpd", ["/usr/sbin/lighttpd", "-tt", "-f", "/etc/lighttpd/lighttpd.conf"], 0xbed7fdac /* 13 vars */) = 0
set_tls(0xb6f88fac)                     = 0
set_tid_address(0xb6f87854)             = 9097
brk(NULL)                               = 0x1821000
brk(0x1823000)                          = 0x1823000
mmap2(0x1821000, 4096, PROT_NONE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x1821000
open("/etc/ld-musl-armhf.path", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/libmbedcrypto.so.16", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/local/lib/libmbedcrypto.so.16", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/libmbedcrypto.so.16", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
statx(3, "", AT_STATX_SYNC_AS_STAT|AT_NO_AUTOMOUNT|AT_EMPTY_PATH, STATX_BASIC_STATS, {stx_mask=STATX_BASIC_STATS|STATX_MNT_ID, stx_attributes=0, stx_mode=S_IFREG|0644, stx_size=327847, ...}) = 0
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\0\0\0\0004\0\0\0"..., 936) = 936
mmap2(NULL, 344064, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0xb6eb7000
mmap2(0xb6f06000, 20480, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x4e000) = 0xb6f06000
mmap2(0xb6f09000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0xb6f09000
close(3)                                = 0
open("/lib/libpcre2-8.so.0", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/local/lib/libpcre2-8.so.0", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/libpcre2-8.so.0", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
statx(3, "", AT_STATX_SYNC_AS_STAT|AT_NO_AUTOMOUNT|AT_EMPTY_PATH, STATX_BASIC_STATS, {stx_mask=STATX_BASIC_STATS|STATX_MNT_ID, stx_attributes=0, stx_mode=S_IFREG|0644, stx_size=471307, ...}) = 0
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\0\0\0\0004\0\0\0"..., 936) = 936
mmap2(NULL, 475136, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0xb6e43000
mmap2(0xb6eb5000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x72000) = 0xb6eb5000
close(3)                                = 0
open("/lib/libgcc_s.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
statx(3, "", AT_STATX_SYNC_AS_STAT|AT_NO_AUTOMOUNT|AT_EMPTY_PATH, STATX_BASIC_STATS, {stx_mask=STATX_BASIC_STATS|STATX_MNT_ID, stx_attributes=0, stx_mode=S_IFREG|0644, stx_size=45058, ...}) = 0
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\0\0\0\0004\0\0\0"..., 936) = 936
mmap2(NULL, 49152, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0xb6e37000
mmap2(0xb6e41000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0xa000) = 0xb6e41000
close(3)                                = 0
mprotect(0xb6f06000, 8192, PROT_READ)   = 0
mprotect(0xb6eb5000, 4096, PROT_READ)   = 0
mprotect(0xb6e41000, 4096, PROT_READ)   = 0
mprotect(0x51000, 4096, PROT_READ)      = 0
getuid32()                              = 0
open("/etc/TZ", O_RDONLY|O_NONBLOCK|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/etc/localtime", O_RDONLY|O_NONBLOCK|O_LARGEFILE|O_CLOEXEC) = 3
statx(3, "", AT_STATX_SYNC_AS_STAT|AT_NO_AUTOMOUNT|AT_EMPTY_PATH, STATX_BASIC_STATS, {stx_mask=STATX_BASIC_STATS|STATX_MNT_ID, stx_attributes=0, stx_mode=S_IFREG|0644, stx_size=617, ...}) = 0
mmap2(NULL, 617, PROT_READ, MAP_SHARED, 3, 0) = 0xb6e36000
close(3)                                = 0
clock_gettime64(CLOCK_REALTIME, {tv_sec=1738471822, tv_nsec=797594876}) = 0
clock_gettime64(CLOCK_MONOTONIC_COARSE, {tv_sec=150, tv_nsec=117975396}) = 0
clock_gettime64(CLOCK_MONOTONIC_COARSE, {tv_sec=150, tv_nsec=117975396}) = 0
getuid32()                              = 0
getcwd("/root", 4096)                   = 6
getpid()                                = 9097
open("/etc/lighttpd/lighttpd.conf", O_RDONLY|O_NOCTTY|O_NONBLOCK|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
statx(3, "", AT_STATX_SYNC_AS_STAT|AT_NO_AUTOMOUNT|AT_EMPTY_PATH, STATX_BASIC_STATS, {stx_mask=STATX_TYPE|STATX_MODE|STATX_NLINK|STATX_UID|STATX_GID|STATX_MTIME|STATX_CTIME|STATX_INO|STATX_SIZE|STATX_BLOCKS|STATX_MNT_ID, stx_attributes=STATX_ATTR_COMPRESSED, stx_mode=S_IFREG|0644, stx_size=2332, ...}) = 0
mmap2(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6e34000
read(3, "### Documentation\n# https://wiki"..., 2332) = 2332
close(3)                                = 0
statx(AT_FDCWD, "/etc/lighttpd/mime.conf", AT_STATX_SYNC_AS_STAT|AT_SYMLINK_NOFOLLOW|AT_NO_AUTOMOUNT, STATX_BASIC_STATS, {stx_mask=STATX_BASIC_STATS|STATX_MNT_ID, stx_attributes=0, stx_mode=S_IFREG|0644, stx_size=56745, ...}) = 0
open("/etc/lighttpd/mime.conf", O_RDONLY|O_NOCTTY|O_NONBLOCK|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
statx(3, "", AT_STATX_SYNC_AS_STAT|AT_NO_AUTOMOUNT|AT_EMPTY_PATH, STATX_BASIC_STATS, {stx_mask=STATX_BASIC_STATS|STATX_MNT_ID, stx_attributes=0, stx_mode=S_IFREG|0644, stx_size=56745, ...}) = 0
mmap2(NULL, 57344, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6e26000
read(3, "# created by create-mime.conf.pl"..., 56745) = 56745
close(3)                                = 0
mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6e25000
[... cut ...]
munmap(0xb6df1000, 4096)                = 0
mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6df1000
munmap(0xb6df1000, 4096)                = 0
mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6df1000
munmap(0xb6df1000, 4096)                = 0
mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6df1000
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=NULL} ---
+++ killed by SIGSEGV +++
Segmentation fault

```

gstrauss, 2025-02-02:
> (If that a way to payback contribution to opensource community, I need to do.)

Thank you for you help so far in tracking this down.

```
open("/etc/dehydrated/certs/ha.FILENAMEEX/combined.pem", O_RDONLY|O_NOCTTY|O_NONBLOCK|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
statx(3, "", AT_STATX_SYNC_AS_STAT|AT_NO_AUTOMOUNT|AT_EMPTY_PATH, STATX_BASIC_STATS, {stx_mask=STATX_TYPE|STATX_MODE|STATX_NLINK|STATX_UID|STATX_GID|STATX_MTIME|STATX_CTIME|STATX_INO|STATX_SIZE|STATX_BLOCKS|STATX_MNT_ID, stx_attributes=STATX_ATTR_COMPRESSED, stx_mode=S_IFREG|0644, stx_size=3177, ...}) = 0
mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6df1000
read(3, "-----BEGIN CERTIFICATE-----\nMIID"..., 3177) = 3177
close(3)
```
I can see that lighttpd is loading the file into memory.  The next thing called is `mbedtls_x509_crt_parse()`.  If you copy the combined.pem and remove the private key, does it still fail?  If so, can you share the file (without the private key) at https://paste.lighttpd.net/

I do not think this is an issue with lighttpd, but am willing to dig to help track this down.

Do you have gdb installed on the system?
```
gdb --args /usr/sbin/lighttpd -tt -f /etc/lighttpd/lighttpd.conf
(gdb) run
SEGV
(gdb) bt full
```

gstrauss, 2025-02-02:
Instead of using the combined.pem, please use `ssl.privkey = "/etc/dehydrated/certs/ha.FILENAMEEX/privkey.pem"` with the above.

mic-user, 2025-02-02:
@gstrauss 
- I have remove private key for ssl.pem
- I have added the line /etc/dehydrated/certs/ha.FILENAMEEX/privkey.pem
and gdb log below 

```
# gdb --args /usr/sbin/lighttpd -tt -f /etc/lighttpd/lighttpd.conf
GNU gdb (GDB) 15.2
Copyright (C) 2024 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "arm-openwrt-linux".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /usr/sbin/lighttpd...
(No debugging symbols found in /usr/sbin/lighttpd)
(gdb) run
Starting program: /usr/sbin/lighttpd -tt -f /etc/lighttpd/lighttpd.conf

Program received signal SIGSEGV, Segmentation fault.
0x00000000 in ?? ()
(gdb) bt full
#0  0x00000000 in ?? ()
No symbol table info available.
#1  0xb6f42008 in ?? ()
No symbol table info available.
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb)

```

mic-user, 2025-02-02:
@gstrauss 
FYI,
if you build a rc6 image include lighttpd-mbedtls from the image building portal,
you going to get the same problem.
But the rc6 image previously I have built is okay.

mic-user, 2025-02-07:
Any fix?

gstrauss, 2025-02-08:
@mic-user wrote
> Any fix?

I already posted the following above, *twice*:

Please see https://github.com/openwrt/openwrt/issues/17802#issuecomment-2629037831
>> (I am a lighttpd developer, but am sorry that I will not have further time for this at the moment. I'll follow this and may have time in another week or so.)

Please see https://github.com/openwrt/openwrt/issues/17802#issuecomment-2629240861
>> I won't have access to some test hardware for another week, and will try to repro then.

mic-user, 2025-02-08:
@gstrauss 
Then, the official OpenWRT version has been released without your contributed version.(wow...holy)
which means not the time to upgrade into the just released official version.
(personally, I will wiat for the issue close notification)
Cheers.

gstrauss, 2025-02-08:
I'm disappointed, too.  I'll be requesting backports of lighttpd 1.4.77, probably next week.

gstrauss, 2025-02-12:
I used `dehydrated` and have no issues on an OpenWRT derivative using lighttpd 1.4.76 and lighttpd mod_mbedtls.

Would you be able to share your `/etc/dehydrated/certs/ha.FILENAMEEX/fullchain.pem` ?
Please do not include the private key.

mic-user, 2025-02-12:
@gstrauss 
Hi, I have sent the evidence PEM trough your email that opened on your GitHub profile.
When examined, let me know.
Cheers.

mic-user, 2025-02-12:
@gstrauss And please check with wolfssl too.

gstrauss, 2025-02-12:
A quick check with your provided .pem loads the .pem fine with both mbedtls and wolfssl.  Both predictably error out with a mismatched private key, since I do not have your private key (and do not want it).  On my system, `lighttpd -D -f /etc/lighttpd/lighttpd.conf -tt` runs in 0.81 seconds with lighttpd mod_mbedtls, and 2.92 seconds with lighttpd mod_wolfssl.  I believe the slower startup with lighttpd mod_wolfssl is due to the underlying wolfssl library RNG initialization, but did not re-verify that.

On my system:
```
# opkg list-installed | grep -E "lighttpd|mbedtls|wolfssl"
libmbedtls - 2.28.8-1
libwolfssl - 5.7.2-stable-1
lighttpd - 1.4.76-2
...
lighttpd-mod-mbedtls - 1.4.76-2
lighttpd-mod-wolfssl - 1.4.76-2
```
(and I had to modify `/etc/lighttpd/conf.d/30-{mbedtls,wolfssl}.conf` to enable only one at a time for mbedtls or wolfssl.)

mic-user, 2025-02-12:
@gstrauss Yeah. no problem. Your contribution is important not others.
By the way, there is another chance that the error occured from...
kind of(donno how to call it) 'flash memory image's boundary over-ran' which large image? or Hardware failure?
I also suspect two of both (not only the lighttpd matters).
(Fyi, 'sysupgrade' size is 61,666,782 bytes. If you want, I'll send you the built image link later  with new build)
So, you can give me any another aspect while you proceeding. 
Cheers

mic-user, 2025-02-12:
@gstrauss Do you have the WRT32X device that to run image?

mic-user, 2025-02-12:
@gstrauss I have sent a thing. But not sure that is right you expect.

gstrauss, 2025-02-13:
> Then, the official OpenWRT version has been released without your contributed version.(wow...holy)

OpenWRT 24.10.0 includes lighttpd 1.4.77, so I do not understand to what you are referring.

Is anyone else having a similar issue?  This may be an issue specific to @mic-user

@mic-user: please backup your system and perform a fresh install of 24.10.0 rather than a sysupgrade.

mic-user, 2025-02-13:
@gstrauss 
Regarding re-installing, 
openwrt guys not suggest reinstalling as [article](https://forum.openwrt.org/t/need-advice-for-an-warning-message-with-24-10-0-rc4-factory-rom/220349/15). My feeling was they so freaked out seem that the guys also have no confidence or whatever reason(brick or flash damages) about re-fresh-flashing. Will figure it out myself.

gstrauss, 2025-03-23:
#25985 was never picked up by openwrt commiters and after 5 weeks has now been superceded by #26202, which has been merged to openwrt packages master branch.

In the interim, I have received no other reports of issues, so @mic-user issue here appears to be isolated to the hardware or the, less likely, choices made in the image builder.   I suggest using a stock sysupgrade image.

mic-user, 2025-04-18:
Since similar decade issue from the package
- https://community.ui.com/questions/network-c-543-SSL-couldnt-read-X509-certificate-from-etc-lighttpd-server-pem/1a6ba835-7146-4173-bc9a-9f91c2c15d8e
- https://serverfault.com/questions/723651/ssl-couldnt-read-x509-certificate-from-pem-file-lighttpd

For the records for all the people,
Solution would be,

Do this first if you use let'sencrypt,
# awk 'NR==1,/END CERTIFICATE/{print}' fullchain.pem > ca.pem

And set isolated pems each
        ssl.verifyclient.ca-file = "/xxxxxxxxxxxxxxxxxxxxxx/ca.pem"
        ssl.pemfile = "/xxxxxxxxxxxxxxxxxxxxxx/fullchain.pem"
        ssl.privkey = "/xxxxxxxxxxxxxxxxxxxxxx/privkey.pem"

gstrauss, 2025-04-19:
> ### Describe the bug
> 
> wolfssl block to load lighttpd as below `# service lighttpd start 2025-01-31 21:35:07: (../src/mod_wolfssl.c.924) SSL: couldn't read X509 certificates from '/etc/fullchain.pem'
> 
> Illegal instruction
> 
> lighttpd.conf validation failed `

@mic-user: are you able to reproduce the issue and do your suggested changes fix the issue?

If not, I suggest that other readers not waste their time here, as the advice given does not seem to have been methodically tested.

[lighttpd TLS docs](https://wiki.lighttpd.net/Docs_SSL) recommend against setting `ssl.verifyclient.ca-file` or `ssl.ca-file` if not also configuring client certificate verification, so @mic-user advice is against the recommendations of the lighttpd developer (me) who wrote lighttpd `src/mod_wolfssl.c`.

As I posted last month, I was unable to reproduce the issue reported by @mic-user, and have not heard of any similar issues reported since.

mic-user, 2025-04-19:
The solution verified.
Don't need to be shame that you didn't suggest or understand what the problem is. No one expecting professionalism from open source. Just rest.

gstrauss, 2025-04-19:
Your original post had the error
> 2025-01-31 21:35:07: (../src/mod_wolfssl.c.924) SSL: couldn't read X509 certificates from '/etc/fullchain.pem'
and you were told not to use `ssl.ca-file` since you were misusing it.  That's about all any reader here needs to know, especially after seeing that you are still using it incorrectly.

I am glad your solution "works" for you, but I also do not expect to be relevant to others.  It sounds to me as if you had corrupted .pem files, and whatever you did trying to solve it had the side effect of replacing the corrupted files.

gstrauss, 2025-04-19:
>> awk 'NR==1,/END CERTIFICATE/{print}' fullchain.pem > ca.pem

That *truncates* the .pem file after the first certificate.  That is most certainly *wrong* if there is more than one certificate in the chain, which is the intention of fullchain.pem.  For Let's Encrypt certificates, fullchain.pem contains the certificate followed by an intermediate certificate.  Truncating the Let's Encrypt fullchain.pem file is *wrong*.

This is sufficient evidence that @mic-user does not know what they are doing with certificates, and @mic-user advice about certificates should be considered suspect, and likely ignored.

mic-user, 2025-04-19:
You don't even know what the option 'ssl.verifyclient.ca-file' really works on openwrt. dont you?
You don't even know what the format of let's encrypt pem format, don't you?
If you want to stay as a proud lighttpd maintainer,
must verify the result yourself before you spit out talk on the public archive as being a proud maintainer
not just chasing people to say that 'ah, is not confirmed me' type talk which whining. 
And you don't even suggest any of solution or workaround a bit yet. Zero solution from you by all the long time spending. did you? No. nothing from you. only but negative talks with no solutions. 
You don't even have no control the plug-in library release and changes. Out of your boundary kid what you are doing now.
So, you are the nothing different type of an open source guy who have zero experience of consulting not a bit. just a socket deamon engineer.

So, 
the solution on the openwrt 24.10.1 for the error "../src/mod_wolfssl.c.924) SSL: couldn't read X509 certificates"
is 
```
awk 'NR==1,/END CERTIFICATE/{print}' fullchain.pem > ca.pem

And set isolated pems each
ssl.verifyclient.ca-file = "/xxxxxxxxxxxxxxxxxxxxxx/ca.pem"
ssl.pemfile = "/xxxxxxxxxxxxxxxxxxxxxx/fullchain.pem"
ssl.privkey = "/xxxxxxxxxxxxxxxxxxxxxx/privkey.pem"
```
And if you have nothing solution, keeping silence for staying for lighttpd maintainer
unless you will be look down on from the public and the package either. Every one watching the archive and remains for ever.

closed 2025-04-19 by mic-user

gstrauss, 2025-04-19:
> You don't even know what the format of let's encrypt pem format, don't you?

In fact, I do know.  `while openssl x509 -noout -text; do :; done </path/to/fullchain.pem`
I am not going to continue to prove myself to you.

Your "solution" is *wrong*.  It is a good thing that lighttpd mod_wolfssl.c uses the chain from fullchain.pem.  If you have not configured `ssl.verifyclient.activate`, then lighttpd will subsequently ignore your `ssl.verifyclient.ca-file`, assuming you have not truncated or otherwise corrupted your fullchain.pem from Let's Encrypt since the fullchain.pem from Let's Encrpyt includes intermediate certificates.

You have not established any credibility here, and have made things up.
I have proven as such in my posts above and have provided evidence that others can verify.

Stop asking questions and I'll let you have the last word.

As an aside, your anger makes your English very difficult to understand.