openwrt/openwrt #17804: wifi_schedule  error S99wifi_schedule: uci: Parse error (invalid command) at line 9, byte 0

przemekp1, 2025-01-31:
Po aktualizacji do wersji 21.10 rc7, mam taki komunikat w logach

Fri Jan 31 13:43:30 2025 daemon.notice procd: /etc/rc.d/S99wifi_schedule: uci: Parse error (invalid command) at line 9, byte 0

Kernel Version：6.6.73
Model：ASUS RT-AC65P
Architecture：MediaTek MT7621 ver:1 eco:3
Firmware Version：OpenWrt 24.10.0-rc7 r28417-daef29c75d

brada4, 2025-01-31:
Helps if you report the issue in correct place (../packages/), here it is mandatory to fill issue template down to last letter of build tag. English please.

namiltd, 2025-01-31:
https://openwrt.org/releases/24.10/notes-24.10.0-rc7:
```
Upgrading to 24.10

Sysupgrade can be used to upgrade a device from 23.05 to 24.10, and configuration will be preserved in most cases.

:!: Sysupgrade from 22.03 to 24.10 is not officially supported. 
```
And you are updating from an even older version.

brada4, 2025-01-31:
This one can be  closed...