openwrt/openwrt #17807: BananaPi R4: sfp ports still named "eth1" and "eth2"

kusmi, 2025-01-31:
### Describe the bug

In December there was a commit: https://github.com/openwrt/openwrt/commit/cd8dcfef378044a1687adfa3738f01f9a9622baf which finally changed the named sfp-ports from **eth1** and **eth2** to **sfp-lan** and **sfp-wan**.

But this commit was not picked up in the 24.10.0-rc releases yet.

I would propose to include this rename right into the final 24.10.0 release, as it probably would confuse users, it the rename would happen on a small upgrade release like 24.10.1.

What do you think?

### OpenWrt version

r28417-daef29c75d

### OpenWrt release

24.10.0-rc7

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R4

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

KFOq, 2025-02-01:
@hauke

yes this issue still exist on 24.10.0-rc releases ----->eth1 and eth2 for sfp ports
[image]

On OpenWRT Snapshots their names are SFP wan and  SFP lan

SimonIremonger, 2025-02-06:
Please check with 24.10.0 still present?
A similar/related interface naming issue hopefully to be included in 24.10 too:  https://github.com/openwrt/openwrt/pull/17757

kusmi, 2025-02-19:
it is still present in final release 24.10.0: they are still called eth2 and eth1.