openwrt/openwrt #17808: [MikroTik RouterBOARD M33G mt7621] Inconsistent eth0 mac address across reflashes/upgrades

nemesifier, 2025-01-31:
### Describe the bug

The mac address of eth0 is not consistent across reflashing / upgrades.

### OpenWrt version

OpenWrt 24.10

### OpenWrt release

r28157-c981096416

### OpenWrt target/subtarget

mt7621

### Device

MikroTik RouterBOARD M33G

### Image kind

Official downloaded image

### Steps to reproduce

1. Flash the image on a device, take note of the mac address of eth0.
2. Now reset to factory defaults (eg: `firstboot -y; reboot`). Once the device boots, take note of the mac address of eth0 again.

### Actual behaviour

The mac address of step1 and step2 are different.

### Expected behaviour

The mac address should not change after reflashing/restoring to factory defaults.

### Additional info

The eth0 interface doesn't seem to point to any physical interface, the interfaces pointing to physical interfaces are lan1, lan2 and wan.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

nemesifier, 2025-01-31:
I tested this both from a self-compiled image or from the official openwrt image, let me update the description accordingly.

brada4, 2025-02-01:
And which release gave you nonextant version tags?

nemesifier, 2025-02-01:
@brada4 I am not 100% sure I understood your question.

I tested both the OpenWrt 23.05.5 official image, after noticing the issue on a custom build based on the openwrt-24.10 branch.

In both cases, flashing the image or upgrading causes the mac address of eth0 to change. Let me know if I misunderstood you and I'll provide more info.