openwrt/openwrt #17810: Cannot boot RT3883 board on 24.10

nariox, 2025-02-01:
### Describe the bug

I have an old TEW-692GR that I haven't used in a while, but with the excitement of 24.10, I decided to try it out. Unfortunately, I cannot get it to boot. I am attaching the bootlog.txt. On 23.05.5, I can boot normally.

Perhaps it is related to this patch, where an internal clock driver was introduced: [LINK](https://lore.kernel.org/lkml/CAMhs-H-0Upz--k0tkm7BFCTd0b0Gso-c_uPyzeAjOigZowbK1Q@mail.gmail.com/) or this [commit](https://git.openwrt.org/?p=openwrt/openwrt.git;a=commitdiff;h=4b8b7438f78346ccc3fc7393a3d026f63df0cdcd).

I don't really use this board anymore and I wouldn't want this to block the release of 24.10, but it might be helpful to put a warning that these images are broken for the TEW-692GR (and perhaps others).

### OpenWrt version

24.10

### OpenWrt release

rc1-rc7

### OpenWrt target/subtarget

ramips/r3883

### Device

TrendNET TEW-692gr

### Image kind

Official downloaded image

### Steps to reproduce

- Flash 24.10-rc7 to the TEW-692.
- Board does not boot

### Actual behaviour

- Kernel panic during boot (bootlog attached [bootlog.txt](https://github.com/user-attachments/files/18626857/bootlog.txt))
- States: Couldn't register fixed clock 3

### Expected behaviour

- Normal boot

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2025-02-01:
I suspect the probem came from https://github.com/openwrt/openwrt/commit/7bb99bca3dfd878e6ad950b218c0cb96d36d14f4#diff-315cf6d64cb39d26ff6ff40eab31a5e471417bc8fee34190bbea3a9e070c1b24
Ping @DragonBluep

namiltd, 2025-02-01:
@nariox Please upload the [999-rt3883_clock_init.patch](https://github.com/user-attachments/files/18628704/999-rt3883_clock_init.patch) file to target/linux/generic/hack-6.6 generate images and try to run.

nariox, 2025-02-01:
Wow, that was fast. I'll give it a try. My build machine is busy right now, but I'll build it as soon as I can.

nariox, 2025-02-01:
Partial success!
Now it goes past the first part. It still gets stuck after loading some uart driver (I think, not sure). Attached is the new serial boot log.
[bootlog_afterpatch.txt](https://github.com/user-attachments/files/18630377/bootlog_afterpatch.txt)

nariox, 2025-02-01:
Actually, I have just looked at the RT-N56U issue. I believe these are duplicate. I am going to close as a duplicate of #17677

closed 2025-02-01 by nariox