openwrt/openwrt #17812: 'luci-proto-wireguard'  WireGuard Status page changing connected 'client name' to 'Untitled peer'

Eurodocsis, 2025-02-01:
### Describe the bug

Describe the bug
When visiting Wireguard status page to see connected clients you see your list of connected clients/peers
However after the first connect the name change from "yourclientname" to Untitled Peer'

[image]

### OpenWrt version

r24106-10cc5fcd00

### OpenWrt release

23.05.5 r24106

### OpenWrt target/subtarget

mediatek/filogic

### Device

Zyxel EX5601-T0 ubootmod

### Image kind

Official downloaded image

### Steps to reproduce

Reboot your router , go to list http://yourip/ cgi-bin/luci/admin/status/wireguard or WireGuard Status
You will see all devices listed by name and details When you connect a device the name change to "Untitled peer"

i have all possible ipv6 settings disabled on devices and interfaces ( my isp not have any ipv6 support )

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

agevlakh, 2025-02-01:
This is not a bug. You need go to interfces -> wg -> peers and delete strings endpoint host and endpoint port in each peer and save, reboot and try again.

Eurodocsis, 2025-02-01:
> This is not a bug. You need go to interfces -> wg -> peers and delete strings endpoint host and endpoint port in each peer and save, reboot and try again.

Thank you for your reply but why would i need to remove the endpoint and port ? Do you mean you need to re-add them ? Just removing then don't make sense

agevlakh, 2025-02-01:
no, you need to go to each peer and delete these lines. as I understand it, if you leave them, then the peer tries to connect from these IPs and ports, and when you try to connect from your device via wireguard, then a double connection occurs, which causes the effect you described.

Eurodocsis, 2025-02-02:
> no, you need to go to each peer and delete these lines. as I understand it, if you leave them, then the peer tries to connect from these IPs and ports, and when you try to connect from your device via wireguard, then a double connection occurs, which causes the effect you described.

Hi So basically you are saying We are adding the lines ip:port to get a config exported and delete them after?

Let me test it I keep you posted

Eurodocsis, 2025-02-02:
@agevlakh Even if it sounds odd to me removing that part and seeing its 'optional'  It got the issue fixed. I want to thank you taking assisting me this .

cheers

Eurodocsis, 2025-02-02:
resolved here https://github.com/openwrt/openwrt/issues/17812#issuecomment-2629096226  

Thanks!

closed 2025-02-02 by Eurodocsis