openwrt/openwrt #17816: Cudy TR3000 v1 2.5G port  Start cannot start automatically

NNET-ELi, 2025-02-01:
### Describe the bug

Device: Cudy TR3000 v1
Version: OpenWrt 24.10.0-rc7 r28417-daef29c75d / LuCI openwrt-24.10 branch 25.027.83426~170375e
Every time a gigabit rate network cable is plugged in, eth0(2.5G port)， It won't start by itself when starting.
You need to use "ifup wan" to start.

[    0.000000] psci: probing for conduit method from DT.
[    1.809156] mtk_soc_eth 15100000.ethernet eth0: mediatek frame engine at 0xffffffc081700000, irq 75
[    1.819071] mtk_soc_eth 15100000.ethernet eth1: mediatek frame engine at 0xffffffc081700000, irq 75
[    3.036350] mtk_soc_eth 15100000.ethernet eth1: PHY [mdio-bus:00] driver [MediaTek MT7981 PHY] (irq=POLL)
[    3.049086] mtk_soc_eth 15100000.ethernet eth1: configuring for phy/gmii link mode
[    7.196036] mtk_soc_eth 15100000.ethernet eth1: Link is Up - 1Gbps/Full - flow control rx/tx
[    7.531860] mtk_soc_eth 15100000.ethernet eth1: Link is Down
[   12.818174] mtk_soc_eth 15100000.ethernet eth1: PHY [mdio-bus:00] driver [MediaTek MT7981 PHY] (irq=POLL)
[   12.831141] mtk_soc_eth 15100000.ethernet eth1: configuring for phy/gmii link mode
[   12.839933] br-lan: port 1(eth1) entered blocking state
[   12.845168] br-lan: port 1(eth1) entered disabled state
[   12.850490] mtk_soc_eth 15100000.ethernet eth1: entered allmulticast mode
[   12.857484] mtk_soc_eth 15100000.ethernet eth1: entered promiscuous mode
[   13.295824] mtk_soc_eth 15100000.ethernet eth0: mtk_open: could not attach PHY: -110
[   19.116032] mtk_soc_eth 15100000.ethernet eth1: Link is Up - 1Gbps/Full - flow control rx/tx
[   19.116061] br-lan: port 1(eth1) entered blocking state
[   19.129694] br-lan: port 1(eth1) entered forwarding state

root@XXXX:~# ifconfig
br-lan    Link encap:Ethernet  HWaddr 80:AF:CA:88:01:C7  
          inet addr:100.100.1.1  Bcast:100.100.1.255  Mask:255.255.255.0
          inet6 addr: fe80::82af:caff:fe88:1c7/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1540 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1390 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:236991 (231.4 KiB)  TX bytes:1494886 (1.4 MiB)

eth1      Link encap:Ethernet  HWaddr 80:AF:CA:88:01:C7  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1579 errors:0 dropped:4 overruns:0 frame:0
          TX packets:1714 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:261556 (255.4 KiB)  TX bytes:1521298 (1.4 MiB)
          Interrupt:75 

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:120 errors:0 dropped:0 overruns:0 frame:0
          TX packets:120 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:8832 (8.6 KiB)  TX bytes:8832 (8.6 KiB)

### OpenWrt version

24.10.0-rc7

### OpenWrt release

r28417-daef29c75d

### OpenWrt target/subtarget

mediatek/filogic

### Device

Cudy TR3000 v1

### Image kind

Official downloaded image

### Steps to reproduce

ETH0 plugs in the gigabit network cable, and then turns on device power. 
It's always like this

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

jfox52, 2025-02-02:
> ### Describe the bug
> 
> Device: Cudy TR3000 v1 Version: OpenWrt 24.10.0-rc7 r28417-daef29c75d / LuCI openwrt-24.10 branch 25.027.83426~170375e Every time a gigabit rate network cable is plugged in, eth0(2.5G port)， It won't start by itself when starting. You need to use "ifup wan" to start.
> 
> [ 0.000000] psci: probing for conduit method from DT. [ 1.809156] mtk_soc_eth 15100000.ethernet eth0: mediatek frame engine at 0xffffffc081700000, irq 75 [ 1.819071] mtk_soc_eth 15100000.ethernet eth1: mediatek frame engine at 0xffffffc081700000, irq 75 [ 3.036350] mtk_soc_eth 15100000.ethernet eth1: PHY [mdio-bus:00] driver [MediaTek MT7981 PHY] (irq=POLL) [ 3.049086] mtk_soc_eth 15100000.ethernet eth1: configuring for phy/gmii link mode [ 7.196036] mtk_soc_eth 15100000.ethernet eth1: Link is Up - 1Gbps/Full - flow control rx/tx [ 7.531860] mtk_soc_eth 15100000.ethernet eth1: Link is Down [ 12.818174] mtk_soc_eth 15100000.ethernet eth1: PHY [mdio-bus:00] driver [MediaTek MT7981 PHY] (irq=POLL) [ 12.831141] mtk_soc_eth 15100000.ethernet eth1: configuring for phy/gmii link mode [ 12.839933] br-lan: port 1(eth1) entered blocking state [ 12.845168] br-lan: port 1(eth1) entered disabled state [ 12.850490] mtk_soc_eth 15100000.ethernet eth1: entered allmulticast mode [ 12.857484] mtk_soc_eth 15100000.ethernet eth1: entered promiscuous mode [ 13.295824] mtk_soc_eth 15100000.ethernet eth0: mtk_open: could not attach PHY: -110 [ 19.116032] mtk_soc_eth 15100000.ethernet eth1: Link is Up - 1Gbps/Full - flow control rx/tx [ 19.116061] br-lan: port 1(eth1) entered blocking state [ 19.129694] br-lan: port 1(eth1) entered forwarding state
> 
> root@XXXX:~# ifconfig br-lan Link encap:Ethernet HWaddr 80:AF:CA:88:01:C7 inet addr:100.100.1.1 Bcast:100.100.1.255 Mask:255.255.255.0 inet6 addr: fe80::82af:caff:fe88:1c7/64 Scope:Link UP BROADCAST RUNNING MULTICAST MTU:1500 Metric:1 RX packets:1540 errors:0 dropped:0 overruns:0 frame:0 TX packets:1390 errors:0 dropped:0 overruns:0 carrier:0 collisions:0 txqueuelen:1000 RX bytes:236991 (231.4 KiB) TX bytes:1494886 (1.4 MiB)
> 
> eth1 Link encap:Ethernet HWaddr 80:AF:CA:88:01:C7 UP BROADCAST RUNNING MULTICAST MTU:1500 Metric:1 RX packets:1579 errors:0 dropped:4 overruns:0 frame:0 TX packets:1714 errors:0 dropped:0 overruns:0 carrier:0 collisions:0 txqueuelen:1000 RX bytes:261556 (255.4 KiB) TX bytes:1521298 (1.4 MiB) Interrupt:75
> 
> lo Link encap:Local Loopback inet addr:127.0.0.1 Mask:255.0.0.0 inet6 addr: ::1/128 Scope:Host UP LOOPBACK RUNNING MTU:65536 Metric:1 RX packets:120 errors:0 dropped:0 overruns:0 frame:0 TX packets:120 errors:0 dropped:0 overruns:0 carrier:0 collisions:0 txqueuelen:1000 RX bytes:8832 (8.6 KiB) TX bytes:8832 (8.6 KiB)
> 
> ### OpenWrt version
> 
> 24.10.0-rc7
> 
> ### OpenWrt release
> 
> r28417-daef29c75d
> 
> ### OpenWrt target/subtarget
> 
> mediatek/filogic
> 
> ### Device
> 
> Cudy TR3000 v1
> 
> ### Image kind
> 
> Official downloaded image
> 
> ### Steps to reproduce
> 
> ETH0 plugs in the gigabit network cable, and then turns on device power. It's always like this
> 
> ### Actual behaviour
> 
> _No response_
> 
> ### Expected behaviour
> 
> _No response_
> 
> ### Additional info
> 
> _No response_
> 
> ### Diffconfig
> 
> ```
> 
> ```
> 
> ### Terms
> 
> * [x]  I am reporting an issue for OpenWrt, not an unsupported fork.

I encountered the same issue as you. If I plug the Ethernet cable into the WAN port after the device has booted, the interface works normally.

NNET-ELi, 2025-02-02:
> I encountered the same issue as you. If I plug the Ethernet cable into the WAN port after the device has booted, the interface works normally.

Yes, so my current temporary solution is to add an "ifup wan" in the startup script to temporarily solve the problem that the startup eth0 does not automatically up

JulioMC81, 2025-02-02:
Hi. This issue should have been solved in the latest master and 24.10 snapshot with this commit:

Master snapshot:
https://github.com/openwrt/openwrt/commit/c87a767801ef375feadb0a5c41d5a674ad3a7d2c

24.10 snapshot:
https://github.com/openwrt/openwrt/commit/81db3077483388ec8fa7be441e34756438678f78

I'm using r28424 24.10 snapshot and the WAN port works perfectly:
https://downloads.openwrt.org/releases/24.10-SNAPSHOT/targets/mediatek/filogic/

Greetings.

NNET-ELi, 2025-02-03:
@JulioMC81 Hi JulioMC81, I tested this feature according to this version and it worked fine, thanks!

closed 2025-02-04 by Noltari