openwrt/openwrt #17818: After updating to rc7, WiFi does not work

splayboyDWR, 2025-02-01:
### Describe the bug

kern.info kernel: [   27.920353] ieee80211 phy0: rt2x00_set_rt: Info - RT chipset 6352, rev 0500 detected
kern.info kernel: [   27.935890] ieee80211 phy0: rt2x00lib_read_eeprom_file: Info - Loading EEPROM data from 'soc_wmac.eeprom'.
kern.err kernel: [   28.046567] ieee80211 phy0: rt2800_init_eeprom: Error - Invalid RF chipset 0x40ab detected
kern.err kernel: [   28.063145] ieee80211 phy0: rt2x00lib_probe_dev: Error - Failed to allocate device
kern.info kernel: [   28.553172] jitterentropy: Initialization failed with host not compliant with requirements: 9
.
.
daemon.notice netifd: radio0 (2543): WARNING: Variable 'data' does not exist or is not an array/object
daemon.notice netifd: radio0 (2543): Phy not found
daemon.notice netifd: radio0 (2543): Could not find PHY for device 'radio0'
daemon.notice netifd: Wireless device 'radio0' set retry=0
daemon.crit netifd: Wireless device 'radio0' setup failed, retry=0
daemon.notice netifd: Wireless device 'radio0' is now down

[log.txt](https://github.com/user-attachments/files/18629728/log.txt)

### OpenWrt version

r28417-daef29c75d

### OpenWrt release

24.10.0-rc7

### OpenWrt target/subtarget

ramips/mt7620

### Device

D-Link DWR-921 C1

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

aimacintyre, 2025-02-02:
What version did you upgrade from where it was working?

splayboyDWR, 2025-02-02:
From 23.05 to 24.10.0-rc0/1/2/3/4/5/6 it was still working. 
Since 24/10-rc.. I have already been using OWUT.

DragonBluep, 2025-02-02:
Please attach the "jboot" and "config" partition backup here.

splayboyDWR, 2025-02-02:
I have attached it, thanks for your help.

hauke, 2025-02-02:
This is probably related to this change 150aa9aeee509dba5255885618d8df372670e8f3

hauke, 2025-02-02:
I see the log file, but I do not see the content of the jboot partition. Please share that content.

closed 2025-02-03 by openwrt-bot

589userr589, 2025-02-06:
Same story on version 24.10.0.

When a password is set on the wifi network, it offers to enter the password when logging in, when a wifi device wants to connect, the router refuses the connection. The device acts as a wifi jammer, only networks that are on the same channel as the problematic device are visible.

I tested on a Linksys E1700 device.

https://github.com/openwrt/openwrt/issues/7959#issuecomment-2614464075

commit f628467d (2025-02-03): Revert "ramips: mt7620: fix patching mac address in caldata"

This reverts commit 3295f6f1c254cd7e5e5285a05581bf6abbde8999.

It looks like the eeprom gets broken after this change.
I think this change was not tested on a real device before it was
merged.
The MAC addresses will be broken again after this revert.

Fixes: #17818
commit 6df0e3d0 (2025-02-03): Revert "ramips: mt7620: fix patching mac address in caldata"

This reverts commit 3295f6f1c254cd7e5e5285a05581bf6abbde8999.

It looks like the eeprom gets broken after this change.
I think this change was not tested on a real device before it was
merged.
The MAC addresses will be broken again after this revert.

Fixes: #17818

(cherry picked from commit f628467dfd6b32ead172d33a2593f04901395343)